cyoung/stratux #7: Can't build

felixk, 2015-08-18:
I'm having trouble building the go code...I get the following on raspian:

jpoirier, 2015-08-18:
GOARCH would need to be one of the following: 386|amd64|arm, but you don't need to set any build variables if you're building for the host system, i.e. targeting the system the compiler is running on.

Check where the other go compiler is 
    $ which go

Check the compiler versions 
    $ go version
    $ go-linux-arm version

You want to use whichever is the latest, which is probably go-linux-arm .

As far as the build errors for the "go" version of the compiler, I'm guessing it's a very old version (maybe 1.0.2) and not compatible with newer code in serial embd.

felixk, 2015-08-19:
Thanks! I built GO from source and installed version 1.4.2 and it built fine. Thanks for this great project. I can't wait to try it in the air.

cyoung, 2015-08-19:
felixk, are you testing AHRS and/or GPS?

felixk, 2015-08-19:
Not yet, i don't have the mpu6050 or the ry835ai. I was thinking about buying them. Why do you need some help testing?

cyoung, 2015-08-19:
The MPU6050 is part of the RY835AI, along with the BMP180.

Just asking since you were compiling from source. You won't gain any extra features by compiling the current code over the v0.1r2 image, the only things in there since then are GPS/AHRS.

felixk, 2015-08-19:
Got it. Thanks. I definitely want the AHRS and GPS capability. I'm going to pick up the RY835AI and give it a try.

closed 2015-08-19 by cyoung