cyoung/stratux #9: Significant jitter in heartbeats

BrantKyser, 2015-08-20:
I ran Wireshark on my desktop to capture the IP traffic from stratux running on a Raspberry Pi 2. I was expecting to see a heartbeat packet every second; however, it appears that there is a lot of jitter in the heartbeats. It is not uncommon for multiple seconds to elapse between heartbeats.

I've attached the pcap file from Wireshark that shows this.

I don't know much about go; however, looking at the code I'm guessing that either there is some sort of locking issue on outConn on line 273 or the other functions called in the heartbeat loop (lines 275 - 279) are taking up significant time.

jpoirier, 2015-08-21:
Probably a good idea to start doing some profiling. I'll run pprof on the code later tonight.

cyoung, 2015-08-21:
It should be more consistent now with changes today. There was a time.Sleep() in the heartbeat loop, changed it to a ticker. Also network.go replaced the "outConn.Write()" stuff that was in there. I kind of just sidestepped what seemed to be issues with broadcast packets in network.go and didn't really diagnose it completely. It was causing issues with Android devices at least, and it seemed to be a contributing factor in this also.

cyoung, 2015-08-24:
Brant, would you mind seeing if you can reproducing it using the latest?

BrantKyser, 2015-08-24:
I was playing with the latest last night, and at a cursory glance it looked much better. I'll look at it closer and report back.

kdknigga, 2015-08-24:
I took this screenshot with the code that was newest the evening of the
21st.

http://imgur.com/xH79huR

It's no pcap, but it looks pretty decent.

BrantKyser, 2015-08-24:
The latest version looks good!

Over a 5 min packet capture, I'm getting an average time between heartbeats of 1.003 seconds with a standard deviation of .078.

closed 2015-08-24 by BrantKyser

cyoung, 2015-08-24:
Thanks for Brant and Kris!

BrantKyser, 2015-08-25:
No problem. This is a neat project; let me know if there are any tasks you would like some help with.

cyoung, 2015-08-25:
Could you send me an email? cyoung@gmail