cyoung/stratux #18: Faulty traffic name display

mbliu, 2015-08-24:
For the latest master branch (as of 8/23 0630Z) targets in WingX appear with a nonsense name "uuuuu" above it. What is interesting is each target has a different number of u's in the name as seen in the screenshot. (This screenshot was with the dump978 sample data, but I've observed this with real signals too. Don't have Foreflight so couldn't test with it.) With release v0.1r2, there was nothing at all displayed above the traffic. Using an original Raspberry Pi and compiling everything from source (ARM6). 
[image]

cyoung, 2015-08-24:
This is an actual feature, not a bug :)

It's a hack that is used mostly for ForeFlight - which shows all traffic (at least the way we're sending it) as 978 UAT traffic.

'u' is for TIS-B + UAT traffic. 'e' is for 1090ES traffic.

It'll fill in a 'u' or an 'e' for each packet it receives, up to 8 characters. If the tail number is available, it will place it in the front up of the tail up to 8 characters. So you'll see flights like eDAL1234, eeeUA221.

closed 2015-08-24 by cyoung

mbliu, 2015-08-24:
Ah interesting hack. May I ask why it is useful in flight to know how many packets were received (aside from development)? It might be nice to make this an option in the stratux.conf, as at least on WingX it clutters the screen when zoomed out. (Where is that conf file by the way? Since I installed from source, it always complains it can't find /etc/stratux.conf, and I can't seem to find a template in the source.) Thanks for all the amazing work so far!

cyoung, 2015-08-24:
Don't worry about it complaining about stratux.conf.

It's not that useful to use in flight, more useful for testing as this version that displays UAT and 1090ES traffic at the same time is still being tested.

If you want to remove it, just edit the source and remove these lines:

https://github.com/cyoung/stratux/blob/master/traffic.go#L465-L469

https://github.com/cyoung/stratux/blob/master/traffic.go#L326-L329