cyoung/stratux #20: shut down ports on ICMP unreachable

cyoung, 2015-08-28:
07:42:44.279269 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:44.478652 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:44.722414 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:44.884646 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:44.969908 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:45.078377 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:45.278579 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:45.478433 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:45.678527 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:45.879260 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:45.969995 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36
	07:42:46.078459 IP 192.168.10.10 > 192.168.10.1: ICMP 192.168.10.10 udp port 43211 unreachable, length 36

FF control messages:

wake
	07:40:46.828028 IP (tos 0x0, ttl 64, id 20545, offset 0, flags [none], proto UDP (17), length 88)
	    192.168.10.10.50113 > 192.168.10.255.50113: [udp sum ok] UDP, length 60
	        0x0000:  4500 0058 5041 0000 4011 93fa c0a8 0a0a  E..XPA..@.......
	        0x0010:  c0a8 0aff c3c1 c3c1 0044 5f80 fffe 6900  .........D_...i.
	        0x0020:  2d00 6300 6100 6e00 2d00 7000 6c00 6100  -.c.a.n.-.p.l.a.
	        0x0030:  7900 2d00 6600 6600 6d00 2d00 7500 6400  y.-.f.f.m.-.u.d.
	        0x0040:  7000 3a00 2b00 3a00 2d00 3a00 3d00 3a00  p.:.+.:.-.:.=.:.
	        0x0050:  6900 5000 6100 6400                      i.P.a.d.

sleep
	07:40:43.118753 IP (tos 0x0, ttl 64, id 29141, offset 0, flags [none], proto UDP (17), length 72)
	    192.168.10.10.50113 > 192.168.10.255.50113: [udp sum ok] UDP, length 44
	        0x0000:  4500 0048 71d5 0000 4011 7276 c0a8 0a0a  E..Hq...@.rv....
	        0x0010:  c0a8 0aff c3c1 c3c1 0034 09a2 fffe 6900  .........4....i.
	        0x0020:  2d00 6300 6100 6e00 6e00 6f00 7400 2d00  -.c.a.n.n.o.t.-.
	        0x0030:  7000 6c00 6100 7900 2d00 6600 6600 6d00  p.l.a.y.-.f.f.m.
	        0x0040:  2d00 7500 6400 7000                      -.u.d.p.

active (every 10 sec)
	07:40:46.827974 IP (tos 0x0, ttl 64, id 16702, offset 0, flags [none], proto UDP (17), length 74)
	    192.168.10.10.50113 > 192.168.10.255.50113: [udp sum ok] UDP, length 46
	        0x0000:  4500 004a 413e 0000 4011 a30b c0a8 0a0a  E..JA>..@.......
	        0x0010:  c0a8 0aff c3c1 c3c1 0036 c29d fffe 6900  .........6....i.
	        0x0020:  2d00 7700 6100 6e00 7400 2d00 7400 6f00  -.w.a.n.t.-.t.o.
	        0x0030:  2d00 7000 6c00 6100 7900 2d00 6600 6600  -.p.l.a.y.-.f.f.
	        0x0040:  6d00 2d00 7500 6400 7000                 m.-.u.d.p.

cyoung, 2015-09-01:
Is anyone able to do a study on the way apps act on the network in sleep/background states?

* ICMP unreachable on any apps, Android or iPad, while in the background?
* Any apps sending on/off type messages like FF above?

gms5002, 2015-09-07:
This is probably not helpful, but you never know....WingX Pro 7 seems to repeatedly spew out messages like

```
23:43:03.714986 IP (tos 0x0, ttl 255, id 28864, offset 0, flags [none], proto UDP (17), length 331)
    192.168.10.11.5353 > 224.0.0.251.5353: 0*- [0q] 5/0/3 iPad.local. (Cache flush) AAAA fe80::c21:922:8b2a:24f3, iPad.local. (Cache flush) A 192.168.10.11, 3.f.4.2.a.2.b.8.2.2.9.0.1.2.c.0.0.0.0.0.
0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. PTR iPad.local., iPad._device-info._tcp.local. (Cache flush) TXT "model=J86mAP", 11.10.168.192.in-addr.arpa. PTR iPad.local. (303)
        0x0000:  4500 014b 70c0 0000 ff11 9e32 c0a8 0a0b  E..Kp......2....
        0x0010:  e000 00fb 14e9 14e9 0137 0950 0000 8400  .........7.P....
        0x0020:  0000 0005 0000 0003 0469 5061 6405 6c6f  .........iPad.lo
        0x0030:  6361 6c00 001c 8001 0000 0078 0010 fe80  cal........x....
        0x0040:  0000 0000 0000 0c21 0922 8b2a 24f3 c00c  .......!.".*$...
        0x0050:  0001 8001 0000 0078 0004 c0a8 0a0b 0133  .......x.......3
        0x0060:  0166 0134 0132 0161 0132 0162 0138 0132  .f.4.2.a.2.b.8.2
        0x0070:  0132 0139 0130 0131 0132 0163 0130 0130  .2.9.0.1.2.c.0.0
        0x0080:  0130 0130 0130 0130 0130 0130 0130 0130  .0.0.0.0.0.0.0.0
        0x0090:  0130 0130 0130 0130 0138 0165 0166 0369  .0.0.0.0.8.e.f.i
        0x00a0:  7036 0461 7270 6100 000c 0001 0000 0078  p6.arpa........x
        0x00b0:  0002 c00c 0469 5061 640c 5f64 6576 6963  .....iPad._devic
        0x00c0:  652d 696e 666f 045f 7463 70c0 1100 1080  e-info._tcp.....
        0x00d0:  0100 0011 9400 0d0c 6d6f 6465 6c3d 4a38  ........model=J8
        0x00e0:  366d 4150 0231 3102 3130 0331 3638 0331  6mAP.11.10.168.1
        0x00f0:  3932 0769 6e2d 6164 6472 c086 000c 0001  92.in-addr......
        0x0100:  0000 0078 0002 c00c c00c 0001 8001 0000  ...x............
        0x0110:  0078 0004 c0a8 0a0b c00c 001c 8001 0000  .x..............
        0x0120:  0078 0010 fe80 0000 0000 0000 0c21 0922  .x...........!."
        0x0130:  8b2a 24f3 0000 2905 a000 0000 0000 0cfd  .*$...).........
        0x0140:  ea00 0826 182b ff5f 5e4f 75              ...&.+._^Ou
23:43:03.715461 IP6 (flowlabel 0xfc999, hlim 255, next-header UDP (17) payload length: 311) fe80::c21:922:8b2a:24f3.5353 > ff02::fb.5353: [udp sum ok] 0*- [0q] 5/0/3 iPad.local. (Cache flush) AAAA
fe80::c21:922:8b2a:24f3, iPad.local. (Cache flush) A 192.168.10.11, 3.f.4.2.a.2.b.8.2.2.9.0.1.2.c.0.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. PTR iPad.local., iPad._device-info._tcp.local. (Cache f
lush) TXT "model=J86mAP", 11.10.168.192.in-addr.arpa. PTR iPad.local. (303)
        0x0000:  600f c999 0137 11ff fe80 0000 0000 0000  `....7..........
        0x0010:  0c21 0922 8b2a 24f3 ff02 0000 0000 0000  .!.".*$.........
        0x0020:  0000 0000 0000 00fb 14e9 14e9 0137 f11f  .............7..
        0x0030:  0000 8400 0000 0005 0000 0003 0469 5061  .............iPa
        0x0040:  6405 6c6f 6361 6c00 001c 8001 0000 0078  d.local........x
        0x0050:  0010 fe80 0000 0000 0000 0c21 0922 8b2a  ...........!.".*
        0x0060:  24f3 c00c 0001 8001 0000 0078 0004 c0a8  $..........x....
        0x0070:  0a0b 0133 0166 0134 0132 0161 0132 0162  ...3.f.4.2.a.2.b
        0x0080:  0138 0132 0132 0139 0130 0131 0132 0163  .8.2.2.9.0.1.2.c
        0x0090:  0130 0130 0130 0130 0130 0130 0130 0130  .0.0.0.0.0.0.0.0
        0x00a0:  0130 0130 0130 0130 0130 0130 0138 0165  .0.0.0.0.0.0.8.e
        0x00b0:  0166 0369 7036 0461 7270 6100 000c 0001  .f.ip6.arpa.....
        0x00c0:  0000 0078 0002 c00c 0469 5061 640c 5f64  ...x.....iPad._d
        0x00d0:  6576 6963 652d 696e 666f 045f 7463 70c0  evice-info._tcp.
        0x00e0:  1100 1080 0100 0011 9400 0d0c 6d6f 6465  ............mode
        0x00f0:  6c3d 4a38 366d 4150 0231 3102 3130 0331  l=J86mAP.11.10.1
        0x0100:  3638 0331 3932 0769 6e2d 6164 6472 c086  68.192.in-addr..
        0x0110:  000c 0001 0000 0078 0002 c00c c00c 0001  .......x........
        0x0120:  8001 0000 0078 0004 c0a8 0a0b c00c 001c  .....x..........
        0x0130:  8001 0000 0078 0010 fe80 0000 0000 0000  .....x..........
        0x0140:  0c21 0922 8b2a 24f3 0000 2905 a000 0000  .!.".*$...).....
        0x0150:  0000 0cfd ea00 0826 182b ff5f 5e4f 75    .......&.+._^Ou
```
While open.  I did not observe any 'going to sleep' or 'waking up' type messages.

cyoung, 2015-09-11:
Looks like an iPad thing versus a WingX thing.

But that brings up a good point. Something like this could be used to detect "sleep mode" based on tablet OS info and reduce the number of cases to just two. Were you able to find anything else on that?

gms5002, 2015-09-14:
I can check WingX with regards to the ICMP unreachable thing, just not sure what command you were using to check that?  I can also post on the reddit group and get people to check other apps/platforms.  I don't have any android stuff.

cyoung, 2015-09-14:
tcpdump -i wlan0 -n port not 22

Standard image has tcpdump installed now on it

gms5002, 2015-09-14:
Not really seeing anything in common between ForeFlight and WingX but take a look...

This is the iPad without anything open:

```
18:28:19.457849 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
18:28:20.409235 ARP, Request who-has 192.168.10.10 tell 192.168.10.1, length 28
18:28:20.410296 IP 192.168.10.1.54021 > 192.168.10.11.43211: UDP, length 32
18:28:20.464916 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
18:28:21.409412 ARP, Request who-has 192.168.10.10 tell 192.168.10.1, length 28
18:28:21.410651 IP 192.168.10.1.54021 > 192.168.10.11.43211: UDP, length 32
18:28:21.505658 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
18:28:22.409221 ARP, Request who-has 192.168.10.10 tell 192.168.10.1, length 28
```

Upon opening ForeFlight, it changes to something like this:

```
18:28:22.475407 IP 192.168.10.11.50113 > 192.168.10.255.50113: UDP, length 46
18:28:22.475886 IP 192.168.10.11.50113 > 192.168.10.255.50113: UDP, length 60
18:28:22.477286 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
18:28:22.506609 ARP, Request who-has 4.2.2.2 tell 192.168.10.11, length 28
18:28:23.409217 ARP, Request who-has 192.168.10.10 tell 192.168.10.1, length 28
18:28:23.409789 IP 192.168.10.1.34017 > 192.168.10.11.4000: UDP, length 32
18:28:23.410057 IP 192.168.10.1.54021 > 192.168.10.11.43211: UDP, length 32
18:28:23.435103 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
18:28:24.250551 ARP, Request who-has 4.2.2.2 tell 192.168.10.11, length 28
18:28:24.409286 IP 192.168.10.1.34017 > 192.168.10.11.4000: UDP, length 11
18:28:24.409740 ARP, Request who-has 192.168.10.10 tell 192.168.10.1, length 28
18:28:24.410126 IP 192.168.10.1.34017 > 192.168.10.11.4000: UDP, length 32
18:28:24.410346 IP 192.168.10.1.54021 > 192.168.10.11.43211: UDP, length 32
18:28:24.433068 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
```

Putting ForeFlight to sleep does this:

```
18:28:32.479336 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 4000 unreachable, length 36
18:28:32.479767 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 4000 unreachable, length 36
18:28:32.479791 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
18:28:33.409541 ARP, Request who-has 192.168.10.10 tell 192.168.10.1, length 28
18:28:33.410563 IP 192.168.10.1.34017 > 192.168.10.11.4000: UDP, length 32
18:28:33.411030 IP 192.168.10.1.54021 > 192.168.10.11.43211: UDP, length 32
18:28:33.433712 IP 192.168.10.11.50113 > 192.168.10.255.50113: UDP, length 44
18:28:33.435151 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 4000 unreachable, length 36
18:28:33.435545 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
```

Opening WingX starts this repeatedly (as mentioned before):

```
18:32:36.958787 IP 192.168.10.11.5353 > 224.0.0.251.5353: 0 PTR (QU)? _nslogger-ssl._tcp.local. (42)
18:32:36.959659 IP6 fe80::1416:76ac:d16c:3daf.5353 > ff02::fb.5353: 0 PTR (QU)? _nslogger-ssl._tcp.local. (42)
18:32:36.960140 IP 192.168.10.11.5353 > 224.0.0.251.5353: 0 [1n] [1au] ANY (QM)? 11.10.168.192.in-addr.arpa. (91)
18:32:36.961895 IP6 fe80::1416:76ac:d16c:3daf.5353 > ff02::fb.5353: 0 [1n] [1au] ANY (QM)? 11.10.168.192.in-addr.arpa. (91)
18:32:37.409106 IP 192.168.10.1.54021 > 192.168.10.11.43211: UDP, length 11
18:32:37.409289 ARP, Request who-has 192.168.10.10 tell 192.168.10.1, length 28
18:32:37.435954 IP 192.168.10.11.5353 > 224.0.0.251.5353: 0 [6q] [9n] [1au] ANY (QM)? iPad.local. ANY (QM)? B12ED1BA-DDA3-4D9F-9933-D7C5D15ABAD7_pblipc._pblipc._tcp.local. ANY (QM)? f.a.d.3.c.6.1.d.c.a.6.7.6.1.4.1.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. ANY (QM)? iPad._device-info._tcp.local. ANY (QM)? 11.10.168.192.in-addr.arpa. ANY (QM)? iPad.local. (424)
18:32:37.436336 IP6 fe80::1416:76ac:d16c:3daf.5353 > ff02::fb.5353: 0 [6q] [9n] [1au] ANY (QM)? iPad.local. ANY (QM)? B12ED1BA-DDA3-4D9F-9933-D7C5D15ABAD7_pblipc._pblipc._tcp.local. ANY (QM)? f.a.d.3.c.6.1.d.c.a.6.7.6.1.4.1.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. ANY (QM)? iPad._device-info._tcp.local. ANY (QM)? 11.10.168.192.in-addr.arpa. ANY (QM)? iPad.local. (424)
18:32:37.438916 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
18:32:37.816880 IP 192.168.10.11.5353 > 224.0.0.251.5353: 0 PTR (QM)? _nslogger-ssl._tcp.local. (42)
18:32:37.818403 IP6 fe80::1416:76ac:d16c:3daf.5353 > ff02::fb.5353: 0 PTR (QM)? _nslogger-ssl._tcp.local. (42)
18:32:37.819241 IP 192.168.10.11.5353 > 224.0.0.251.5353: 0 [5q] [7n] [1au] ANY (QM)? B12ED1BA-DDA3-4D9F-9933-D7C5D15ABAD7_pblipc._pblipc._tcp.local. ANY (QM)? f.a.d.3.c.6.1.d.c.a.6.7.6.1.4.1.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. ANY (QM)? iPad._device-info._tcp.local. ANY (QM)? 11.10.168.192.in-addr.arpa. ANY (QM)? iPad.local. (374)
18:32:37.830847 IP6 fe80::1416:76ac:d16c:3daf.5353 > ff02::fb.5353: 0 [5q] [7n] [1au] ANY (QM)? B12ED1BA-DDA3-4D9F-9933-D7C5D15ABAD7_pblipc._pblipc._tcp.local. ANY (QM)? f.a.d.3.c.6.1.d.c.a.6.7.6.1.4.1.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. ANY (QM)? iPad._device-info._tcp.local. ANY (QM)? 11.10.168.192.in-addr.arpa. ANY (QM)? iPad.local. (374)
18:32:37.834538 IP 192.168.10.11.5353 > 224.0.0.251.5353: 0*- [0q] 1/0/1 PTR iPad.local. (85)
[... cut ...]
18:37:06.409420 IP 192.168.10.1.54021 > 192.168.10.11.43211: UDP, length 11
18:37:06.511894 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
18:37:07.409232 ARP, Request who-has 192.168.10.10 tell 192.168.10.1, length 28
18:37:08.415339 ARP, Request who-has 192.168.10.10 tell 192.168.10.1, length 28
18:37:08.415848 IP 192.168.10.1.54021 > 192.168.10.11.43211: UDP, length 11
18:37:08.457106 IP 192.168.10.11 > 192.168.10.1: ICMP 192.168.10.11 udp port 43211 unreachable, length 36
```

cyoung, 2015-09-15:
Good work, I think what is best is to add something to "probe" periodically. It's moving up in priority because the packet rate causing locks is exacerbated by sending out data on every port all the time.

cyoung, 2015-09-15:
@gms5002 - thanks for your help. Would you be able to test https://github.com/cyoung/stratux/commit/5dda5d62202d21ef968ca33963831aef086d0e58 and see if you have any compatibility issues? It starts buffering messages when the iOSAndroid device goes to sleep or is disconnected, and should send them when the receiving app becomes available again.

gms5002, 2015-09-15:
Sure, I'll give it a shot this afternoon.

gms5002, 2015-09-15:
Seems to work alright with ForeFlight and WingX.  I tried varying combinations of restarting iPad, apps, and the Pi, and also switching between apps.  There was one instance where I could not get it to send any data the iPad...resolved by restarting the Pi but I can't reproduce it.  Interestingly, this also improves #22

cyoung, 2015-09-17:
https://github.com/cyoung/stratux/commit/5dda5d62202d21ef968ca33963831aef086d0e58

closed 2015-09-17 by cyoung