cyoung/stratux #21: Add RY835AI GPS serial port config to stratux.conf

rslitchfield, 2015-08-29:
The GPS serial port for the RY835AI can be connected via USB or GPIO UART.
Add GPS serial port config to stratux.conf to allow switching between the two ports without recompile of RY835AI.go.
Default to /dev/ttyACM0 (USB).
Suggested code updates:

gen_gdl90.go
   line 374 add
      GPS_Input    serial.Config
   line 448 add
      globalSettings.GPS_Input = serial.Config{Name: "/dev/ttyACM0", Baud: 9600}

RY835AI.go line 54
    serialConfig = &globalSettings.GPS_Input

cyoung, 2015-08-29:
It should always show up as the same device on the recommended hardware configuration. What are you seeing?

Will be switching over to use the I2C on the NEO-M8N shortly in any case.

rslitchfield, 2015-08-29:
Currently seeing /dev/ttyAMA0

cyoung, 2015-08-29:
Interesting, do you have anything else hooked up to USB? What's your full hardware config?

rslitchfield, 2015-08-29:
I'm using the r0.3 image on SD, RY835AI GPS to GPIO (RX, TX, VPP, GND), WiFi.
I don't currently have any SDRs plugged in.
I can see the GPS data with: $ sudo screen /dev/ttyAMA0 9600
Power from AC adapter
stratux startup disabled in /etc/rc2.d S01 -> K01
Attempting to get GPS data into and out of gen_gdl90
Neither I2C of RY835AI wired to Pi GPIO yet

rslitchfield, 2015-08-29:
[image]
I'm using the WiFi to ssh into the Pi from my laptop.
I have both SDRs working with 987 and 1090 data independently, but am having USB power issues. Powered USB hub arriving tomorrow.
Should I just abandon the RS232, wire up both I2C ports, and wait for the next release?
Which GPIO I2C port should I tie to the GPS and which to the AHRS?  RY835AI I2C 1 -> Pi GPIO I2C 1 and RY835AI I2C 2 -> Pi GPIO I2C 2?

cyoung, 2015-09-15:
Should be working with the latest, hook up TXD (RY835AI) to Pin 10.

closed 2015-09-15 by cyoung

rslitchfield, 2015-09-19:
Confirmed working. Thank you.

JohnPond, 2015-12-11:
Does anyone have the wiring diagram for RY835AI to Raspberry Pi 2 GPIO for GPS, AHRS, etc. running Stratux and ForeFlight?

Thanks,
John Pond

rwilmoth, 2015-12-11:
https://www.reddit.com/r/stratux/comments/3mf8xg/ry835ai_gps_connections_nonusb/?

The top level post is the pinout that I used.