cyoung/stratux #22: 0 relative altitude showing in ForeFlight intermittently

cyoung, 2015-08-30:
Not sure if it is happening in other apps.

Restarting the app works for me, but some users are reporting that restarting the app doesn't fix the issue.

haykinson, 2015-08-30:
confirming that restarting the app doesn't fix this. restarted the app, reconnected wifi, restarted stratux services -- still showing all zeros, despite (when tapped) showing the correct altitude. (this is all from 1090 data).

cyoung, 2015-08-30:
Do you any UAT signals on the ground? I need some dump978 captures to reproduce it.

haykinson, 2015-08-31:
Not locally; I'll take a drive to where I seem to get them from time to time.

Why dump978 captures, if this is 1090 data? Do you mean you want some dump1090 captures? From its --net port, or from the stuff it dumps to console?

cyoung, 2015-08-31:
Most people are having the issues while seeing UAT traffic

dmurray14, 2015-08-31:
On my last trip I found out my home base airport actually has a tower on premises. If you'd like dumps of anything, let me know - I'm heading back over there tomorrow.

gms5002, 2015-09-05:
I am seeing this as well.  Happy to capture some data for you if you can help me out with how to do that.  Is it as simple as /usr/bin/dump1090 > capture.txt?

cyoung, 2015-09-05:
https://github.com/cyoung/stratux/commit/77b86a534f04171fa6cc752eef43dc90ac1aa856

     globalSettings.ReplayLog = false

Change this to true. Give it a run, try to get the bug to reproduce, then send over /var/log/stratux*.log.

gms5002, 2015-09-05:
Alright, thanks I'll give it a shot later tonight.

gms5002, 2015-09-06:
So I might need a little help here as I'm not familiar with GO and having trouble compiling.  I downloaded Google GO from here http://dave.cheney.net/unofficial-arm-tarballs.  I tried to run 'go build' on the source code but it's giving me this error:

```
pi@raspberrypi ~/stratux $ go build
# _/home/pi/stratux
/usr/bin/ld: cannot find -ldump978
collect2: ld returned 1 exit status
```

gms5002, 2015-09-06:
I must be doing something wrong.  I ran make and saw gen_gdl90 was created.  So I copied that version over the one in /usr/bin.  I restarted the Pi and fired up Foreflight.  I did not see any traffic in Foreflight, however the web ui indicated about 1500 1090ES messages.  Looking in /var/log/stratux.log I see the following hundreds of times, but nothing else of any value:

```
2015/09/06 00:43:14 err parsing lat (): strconv.ParseFloat: parsing "": invalid syntax
2015/09/06 00:43:14 err parsing lng (): strconv.ParseFloat: parsing "": invalid syntax
2015/09/06 00:43:18 err parsing lat (): strconv.ParseFloat: parsing "": invalid syntax
2015/09/06 00:43:18 err parsing lng (): strconv.ParseFloat: parsing "": invalid syntax
```

gms5002, 2015-09-07:
I noticed today while flying with 77b86a5 that this was happening.  I restarted the Pi without success, then I closed and opened Foreflight which also didn't work.  What did work was killing Foreflight (double tap home button and flick Foreflight off the screen).  After that it was fine for the rest of the flight.  This leads me to believe it's something to do with the link to Foreflight and not the actual ADS-B message processing.  Any ideas on how to further debug this? Does the 'logging' setting under the Foreflight devices menu do anything?

JohnOCFII, 2015-09-07:
I had the same experience as @gms5002 when flying yesterday.  It operated normally, showing altitude as expected, for about 30 minutes.  After that, we noticed the 0 for altitude difference.  Force-quitting ForeFlight resolved the issue for the remainder of the flight (about another 30 minutes).

cyoung, 2015-09-07:
@gms5002 

"err parsing lat (): strconv.ParseFloat: parsing "": invalid syntax" is normal.

I'm building an image now to send back debug info, will post it shortly with instructions.

cyoung, 2015-09-07:
https://www.dropbox.com/s/t0w7dng01jxbz9n/stratux-debug-77b86a534f-09062015.img.zip?dl=0

Send over http://192.168.10.1/logs/stratux-es.log
and
http://192.168.10.1/logs/stratux-uat.log

with a description of the issue. Can attach the files here.

AvSquirrel, 2015-09-07:
I was flying with @JohnOCFII when we saw the bug. We were testing with two iPads. One flipped to showing "0" relative altitude, while the other continued to indicate the correct relative altitude.

With a Stratus 2, I've occasionally seen the relative altitude go *up* to the MSL altitude of the target (i.e. it believes its own altitude is 0' MSL). Force quit likewise forces normal behavior. Wonder to what extent this isn't just a Stratux bug, but also an interaction / bug in ForeFlight .

gms5002, 2015-09-07:
I will do some testing today with the build from @cyoung and try to nail down steps to recreate the issue.

gms5002, 2015-09-07:
Alright, I believe this will procedure recreate the issue every time:

* iOS 8.4.1
* Foreflight 7.2 (1730)

1. Start up Foreflight and Stratux and pair them.
2. Go to the maps page in Foreflight and verify you are seeing some traffic
   * Remain on the maps page (not sure if this matters but it's what I did in the test)
3. Force quit Foreflight (double tap home button, swipe Foreflight off the top of the screen)
4. Restart Stratux via SSH (sudo shutdown -r now)
5. Wait for stratux wifi network to appear and connect to it
6. Open Foreflight
   * All traffic will show 0 relative altitude
7. Force quit Foreflight
8. Open Foreflight
   * All traffic will now show correct relative altitude 

The logs cover one run through steps 4-8 over the course of a couple minutes. It seems I can only attach images to the issue so I've uploaded here: https://www.dropbox.com/sh/zfsngdt6kmmpwb5/AACsA6YIqdHNXihjPu5-ZK4Ia?dl=0

cyoung, 2015-09-07:
@gms5002, thanks for the details and for uploading the logs!

It has to do with receiving traffic before acquiring the current GPS altitude, or if the GPS lock is lost momentarily. Your procedure works because traffic is being sent (and FF apparently receives it before) the first GPS lock is acquired on startup. It should be remedied when using external GPS on the stratux. If not, who wants to submit a bug report to FF?...

cyoung, 2015-09-07:
If you disable Location Services for FF in your iPad settings it will do it also, you have to restart, though.

gms5002, 2015-09-07:
Hmm, interesting.  I haven't pulled the trigger on the RY835AI yet, so I can't verify whether the external GPS fixes it.  However, it appears as though if you start Foreflight, let it find the GPS position, and then turn on the Pi it works as expected.  Oddly enough, I can even switch into my checklist app for a few minutes (simulating a run up).  When I switch back to Foreflight, it finds its position again but the traffic altitude is still showing up correctly.

Is it possible to add a small delay (perhaps 5-10 seconds) before the stratux starts sending data to Foreflight after pairing?

cyoung, 2015-09-07:
Uncomment https://github.com/cyoung/stratux/blob/master/main/network.go#L124-L127

See https://github.com/cyoung/stratux/blob/master/main/network.go#L148-L152

Yeah, but I don't think it'll fix it. Can play around with those parameters above

gms5002, 2015-09-07:
I uncommented https://github.com/cyoung/stratux/blob/master/main/network.go#L124-L127 and that seems to have resolved the startup issue I described above.

If I disable location services, force quit and reopen Foreflight I see all zero altitudes as you mentioned before.  However, now when I turn location services back on and switch into Foreflight the altitudes start displaying (without force quit and restart).

I'm not sure how to simulate a loss of GPS in flight though.  I tried going into my basement, putting the ipad under a metal pot, and wrapping said pot in aluminum foil...Foreflight just mocks me and remains on 10m accuracy.  I want to try to find out what happens if Foreflight remains open during the entire process of losing and regaining GPS position.  Can anyone get Foreflight to lose GPS position? Perhaps if you put the iPad under some bricks or cinder blocks in might work.

@cyoung Do you want a pull request for the changes? I'm not sure if there are any side effects from the change.

cyoung, 2015-09-07:
No. The side effects of the change are that port 4000 will not receive data for non-FF users. Will either need to add something to work around that or just wait for an FF bug fix.

cyoung, 2015-09-07:
Optionally, we could make a "reset" button on the interface. It just needs to stop sending all traffic reports for a period of time to give the iPad time to get a GPS altitude.

gms5002, 2015-09-07:
And is it correct that stratux finds clients by checking connected dhcp addresses and starts broadcasting over udp port 4000 to everyone...there is no formal handshake?

cyoung, 2015-09-08:
Not for anything other than FF that I know of, but I'm sure there's a reliable way to detect it for anything -- just needs some research. #20 as it seems you've seen already.

cyoung, 2015-09-14:
Just tested FF 7.3. Looks good. Anyone want to go flying with it that had the problem showing up quite often?

gms5002, 2015-09-14:
I'm going to be safety piloting for a friend sometime this week, I'll check it out with 0.3b1 or whatever is current at the time.

cyoung, 2015-09-14:
You were able to reproduce it on the ground, right? Can you try to do the same as you did before

gms5002, 2015-09-14:
Just tested from the ground and it doesn't seem like there was any change for me.  It's still reproducible with the steps above.  Both workarounds (Starting ForeFlight before connecting to stratux network and restarting ForeFlight when the issues occurs) still work.

cyoung, 2015-09-14:
But does it recover after a moment or two?

gms5002, 2015-09-14:
Negative.  I did steps 1-6 above, opened ForeFlight and saw all 0 altitudes.  Then I waited about 5 minutes, still showing all 0s.  I then went further away from the house to acquire new traffic, and even the new targets that popped up had 0 altitude.  I'm using 0.3b1 and ForeFlight 7.3 (1777).

egid, 2015-09-15:
I had this bug intermittently as well. I'll have a trip report this weekend. Did some rough testing at work today and saw a ton of KSEA traffic, no instances of 0 altitude (0.3b1 + 7.3).

gms5002, 2015-09-15:
Commit 5dda5d6 seems to greatly improve the behavior of this with regards to to the start up issue I listed above.  That one is now resolved.  However, I did see it occur twice while testing for #20 when I switched between apps, but I haven't been able to nail it down to reproducible steps.

cyoung, 2015-09-17:
FF 7.3.

closed 2015-09-17 by cyoung