cyoung/stratux #33: Find clock source for reasonably accurate time

cyoung, 2015-09-07:
UAT uplink messages perhaps?

kdknigga, 2015-09-07:
GPS?

224XS, 2015-09-08:
Most devices that will be used to display data from Stratux will have access to one or more of these
1) GPS
2) Baseband cell network
3) ntd
plus
UAT downlink messages when available should be no more than a second off their GPS timebase
UAT uplink "should" also be no more than a second off, probably much less.

How accurate do you need?

224XS, 2015-09-16:
I rechecked the UAT uplink time spec - the header timestamp it is only good for resolving an 
80 micros slot in a single second (GDL 90 Spec section 3.3.1, p15)

I assume you want the time to set the rasp pi clock and a connection to www or baseband is not available inside the rasp pi.

users with a gps module will have a good time source of course. The display device (iPad, phone, etc) should also have a quality internal source that can be harvested and relayed back to the pi on running the web UI or a separate one time process.

Anything I can do?

cyoung, 2015-09-16:
https://github.com/cyoung/stratux/blob/master/main/ry835ai.go#L118-L125

Seconds since midnight UAT - well, it's a timestamp, but not enough to set the Pi's clock. Is a full date/time sent on any GNSS actually?

Seems like a lot of trouble to do this with signal decoding. Yeah, if you're able to - let's just go with the iPad/Android clock.

bradanlane, 2015-09-16:
I'm not sure you will have a transparent method for sending a date-time stamp *back* from the mobile device to the Stratux. Is there something that can be harvested from the network handshake? Or are you assuming something gets added to the WebUI?

GPS is the best choice when one is installed in the Stratux.

kdknigga, 2015-09-16:
Yes, GPS has a full date and time stamp.

In fact, GPS time is pretty much the best time source one can have aside
from a caesium fountain.

ZDA is the NMEA message.

https://github.com/cyoung/stratux/blob/master/main/ry835ai.go#L118-L125

Seconds since midnight UAT - well, it's a timestamp, but not enough to set
the Pi's clock. Is a full date/time sent on any GNSS actually?

Seems like a lot of trouble to do this with signal decoding. Yeah, if
you're able to - let's just go with the iPad/Android clock.

224XS, 2015-09-16:
Hi,

Several NMEA sentences include full time info:
RMC, ZDA

Although I have not used it, ublox seems to have iTOW timestamps which would serve.

I’ll start looking into using the display device clock

Ira
N224XS

cyoung, 2015-09-16:
I take back what I said previously then, looks like GPS is ideal -

      $GNRMC,141228.00,A,x,N,y,W,0.289,,160915,,,A*7C

Would you be interested in implementing this @224XS ?

224XS, 2015-09-16:
I have not got the module, but I can take a crack at the rasp pi code for it.

Do you have a preference for the architecture you want to use for pushing stuff from the web UI back into the pi/stratux?

if there is no GPS module, the pi should try to pull from the display device don’t you think?

Ira

AvSquirrel, 2015-09-16:
@cyoung / @224XS - I do have the module, and would be able to build any code that you need tested.

bradanlane, 2015-09-16:
@224xs - is there a solution which does not require the user to launch the WebUI (or any other "extra" UI/App) on the mobile device?

224XS, 2015-09-16:
Excellent!

224XS, 2015-09-16:
I would think a rasp pi startup process would grab accurate time from a GPS module. Only if there's no GPS on the Pi would you'll need to shoot the time from the webUI

haykinson, 2015-09-16:
I don't think you should do this at startup time, at least not synchronously. The GPS may take some time to acquire a time source, so it's best done as a daemon or in code.

Also, http://www.satsignal.eu/ntp/Raspberry-Pi-NTP.html and http://www.bufferbloat.net/projects/cerowrt/wiki/Getting_time_from_gpsd can serve as good inputs into the discussion.

kdknigga, 2015-09-16:
I've got XGPS160+gpsd+ntpd working on my rig, but I haven't yet made
gen_gdl90 work with gpsd.

My home computer is currently not working, so I need to get that fixed
before I can tinker more.

(Pro tip: don't let HP try to fix your Linux laptop.  You'll get it back
running Windows even though it was only the motherboard that was broken.)

heelix, 2015-09-16:
Is it pulling time from the XGPS160?  I've got that same GPS hardware.

kdknigga, 2015-09-16:
As I recall, yes.  It's been a while since I've messed with it, though.

I'll confirm and report back when I get my laptop back in working order.

Also, I got a Reyax module not too long ago.  I'll see if that GPS plays
nice with gpsd.

224XS, 2015-09-16:
True!  A startup process that tests for the presence of the GPS, then waits 60 s for a cold start since there is no keep alive batt.

cyoung, 2015-09-16:
I don't think you'll even get a valid RMC message unless there is a GPS lock.

kdknigga, 2015-09-18:
RPI + ry835ai + gpsd + ntpd = http://imgur.com/a/1XYsN

closed 2015-09-23 by cyoung

cyoung, 2015-10-19:
@224XS - your code is integrated now.

Haven't looked into it really but the NEO-M8N seems to have a decent clock even without a GPS fix, definitely better than the Pi in any case.

224XS, 2015-10-19:
No worries Chris. Happy to help!
I just got a basket case of a 1953 PA-22/20, so my attention is going to be a bit divided now 
Ira

commit edcccbda (2015-09-23): Fixes #33.

Set system time from RMC sentence.