cyoung/stratux #34: Duplicate targets in FLTPLANGO

PM-ME-YOUR-EMPENNAGE, 2015-09-07:
Howdy. Here's a video that best exhibits the issue I'm having: https://youtu.be/Cqjrd-AVHYw

The screen grab is from my Galaxy S4 but the same exact glitch happens in my Ipad 2 3g , but I don't know if that's sufficient evidence to say the stratux is culprit.

Basically, if I'm not in range of a GBT and picking up raw 978 from airborne aircraft, I'll have 2 targets for each plane following/right on top of each other, one will show whatever the plane is squawking (ie 1200) and the other the actual tailnumber for the aircraft. However, when in range of a GBT this seems to correct itself. Haven't tested with 1090es yet (trying to figure that out still).

224XS, 2015-09-08:
I don't recognize the software on the S4, is the same package as on your iOS?  It could be a fault
of the display software since stratus just passes the messages it receives through wifi to the tablet.
There is a known glitch in ADSB in systems where if you do not have ADSB out, you may see two 
targets for yourself.  Also, if you have both 978 and 1090 running, each actual target should have the
same ICAO identifier and thus the display software should merge the two data streams to a single 
target

PM-ME-YOUR-EMPENNAGE, 2015-09-08:
Yes. It's fltplango, just like in the Ipad, same glitch.

cyoung, 2015-09-09:
Just tested it out with your data @PM-ME-YOUR-EMPENNAGE.

When I tap on the targets (there were 10 in your file) and click "clear breadcrumbs", it actually clears the trail. Their traffic icons are way too big, so most of them overlap. In the settings menu there is "Toggle breadcumbs - Turns the breadcrumb trail on/off" which doesn't specify if checked is on or off, but in either position it displays the breadcrumbs - so that is obviously broken.

Their "Clear/remove breadcrumbs" doesn't work either. Not sure the difference between clearing and removing breadcrumbs, but there are breadcrumbs still displayed when using that feature from the menu.

Looks like some unimplemented features in FltPlan Go.

If you look on Google Play, their changelog is "•added ADS-B traffic" as of Sept 7, looks like they're just working out the bugs. I'll send a bug report.

cyoung, 2015-09-15:
"Hi Chris,

Thanks for letting us know.

The traffic on our Android Go app is a work in progress.
There are many features that need to be added and many bug fixes to be made.
Look for continuing enhancements over the next weeks and months.

Hope this explains."

cyoung, 2015-09-24:
Looks like fltplango pushed an update a couple days ago. Appears that the breadcrumbs are not turned on by default. The traffic icons are still way too large to be useful and sometimes when there are too many traffic targets, the app crashes.

closed 2015-09-24 by cyoung