cyoung/stratux #37: UAT-only pi lockup in an area with fairly heavy traffic

egid, 2015-09-12:
[UAT log here](https://gist.github.com/egid/04e77cf1f9260aaebcd2) - seems to have missed most of the flight, though (09:40-10:20, log starts at 10:20).

Went up today with v0.3b1, a single NooElec dongle, and powered off an Anker. No battery issues the whole flight. Most of the time was just the KPAE hi ADS-B tower, it picked up one other towards the tail end.

We had two iPad Minis in the plane - a retina, and my original - both running ForeFlight latest. My original Mini kept locking up with the traffic view enabled; shortly after that, the pi would crash (or hang) with solid power and no activity LED. The Retina mini seemed more stable, slightly.

cyoung, 2015-09-12:
test/replay logs_stratux-uat.log | ./gen_gdl90

Trying now

egid, 2015-09-12:
Let me know if there's a way to extract the earlier logs - I'd love to get the crashy bits to you. This may not exhibit any issues at all. :(

cyoung, 2015-09-12:
Why would it not log the whole flight, btw? Timestamps are not correct on the Pis, they're just there for some frame of reference.

egid, 2015-09-12:
Oh - I totally didn't realize that. Now I remember seeing mention of finding a way to sync timestamps, so that makes sense.

cyoung, 2015-09-12:
Playing it back now at 4x speed - this is great stuff. You're getting a lot of ground traffic at SEA, besides that a ton in the air.

CPU and memory usage is a little high - going to let this play through and see if it crashes. Then again with pprof.

egid, 2015-09-12:
Awesome! Glad it's useful. I've got a long trip next weekend that may pay dividends as well, probably ~5h of use each way.

cyoung, 2015-09-12:
How often do you turn off the screens on your iPads? Do you keep the screen on all the time or turn it off quite a bit?

egid, 2015-09-12:
I tend to leave the iPad running - I've got panel power. The other iPad is _usually_ on but occasionally gets locked.

cyoung, 2015-09-12:
Have you had the Pi crash without using the old Mini? You mentioned that they happened together, is this always the case?

egid, 2015-09-12:
It didn't crash when I was just connected from my iPhone 6. Haven't tried it in the air, yet, though - had to get home, so it was a short flight.

cyoung, 2015-09-12:
I haven't been able to reproduce the crash - but still working on it. Are you able to build replay.go on the same Pi that crashed and reproduce it with the replay system?

egid, 2015-09-12:
Probably! I won't be able to get to it until tonight, but I'll give that a shot. Do you guys have some instructions for the replay stuff yet? I have a rough idea of what's required, but I haven't dug into the actual Stratux configs / build steps yet.

cyoung, 2015-09-12:
https://github.com/ssokol/stratux/wiki/Compiling-Stratux-on-the-Raspberry-Pi-2

     go build replay.go

Make sure gen_gdl90 isn't running on the screen (screen -x, ctrl +c)

Then run test/replay yourUATfile.log | /usr/bin/gen_gdl90

cyoung, 2015-09-12:
https://youtu.be/mevmmYu9wFw

cyoung, 2015-09-12:
When you're able, send over

     free -m
     cat /proc/cpuinfo

in idle state

egid, 2015-09-12:
Wow, that is a ton of aircraft on the ground at SEA. I'm gonna give the replay a shot now, thanks for the links & info.

cyoung, 2015-09-12:
They look stacked up because this was a replay at 4x speed

egid, 2015-09-12:
[free -m && etc](https://gist.github.com/egid/3e3a41676040571b1bcd)

Went through the build steps + tried the replay, but it looks like I missed something somewhere.

```pi@raspberrypi ~/stratux/test $ ./replay /var/logs/stratux-uat.log | /usr/bin/gen_gdl90
2015/09/12 21:45:57 can't read settings /etc/stratux.conf: open /etc/stratux.conf: no such file or directory
2015/09/12 21:45:57 managementInterface ListenAndServe: listen tcp :80: bind: permission denied
2015/09/12 21:45:57 client connected: 192.168.10.12:4000 ().
2015/09/12 21:45:57 client connected: 192.168.10.12:43211 ().
2015/09/12 21:45:57 client connected: 192.168.10.12:49002 ().
2015/09/12 21:45:57 client connected: 192.168.10.13:4000 ().
2015/09/12 21:45:57 client connected: 192.168.10.13:43211 ().
2015/09/12 21:45:57 client connected: 192.168.10.13:49002 ().
2015/09/12 21:45:57 client connected: 192.168.10.10:4000 ().
2015/09/12 21:45:57 client connected: 192.168.10.10:43211 ().
2015/09/12 21:45:57 client connected: 192.168.10.10:49002 ().
2015/09/12 21:45:57 lost stdin.```

cyoung, 2015-09-12:
Did you just get the latest? I changed the syntax in that time, it's ./replay /var/logs/stratux-uat.log /var/logs/stratux-es.log now

AvSquirrel, 2015-09-12:
I was able to replicate a "crash" -- or more properly a connection hiccup -- on @egid 's data. This was on my production dual-receiver Stratux, powered by a 2A wall wart. I used SSH to interact with the console, and monitored the status page over the LAN. ForeFlight was connected via iPhone 6.

I killed gen_gdl90 to stop the 978 receiver, but kept dump1090 running in the background when I started replay. Since I hadn't built my own executable for gen_gdl90, I piped the replay into /usr/bin/gen_gdl90:

    sudo kill <pid>
    sudo sh -c './replay logs_stratux-uat.log | /usr/bin/gen_gdl90'

ForeFlight immediately began displaying local 1090 traffic, as well as SEA-area traffic from the log file. After 00:04:45 of uptime, the status page stopped updating, and ForeFlight dropped the connection. This is similar behavior to what I had previously seen in flight.

However, the LAN connection remained alive: it responsed to pings, the SSH sessions stayed open and interactive, and I could still access logs. Top showed that gen_gdl90 was still running, and consuming 35-50% CPU. However, the WLAN connection was not responsive; SSH sessions could not be opened. "Forgetting" and reconnecting to the stratux WLAN seemed to reset things.

On my second attempt, the system stayed up for about 22 minutes, and failed at a time when it had seen 1782 UAT messages in the previous minute. Again, similar behavior: LAN connection stayed up; WLAN went down; forgetting and reconnecting brought it back.

AvSquirrel, 2015-09-13:
I ran several accelerated replays (5X and 10X realtime) of the @egid dataset, as well as [airborne data](https://github.com/cyoung/stratux/issues/39) from @JohnOCFII. On both data sets, I was able to repeatedly replicate freezes of the WiFi adapter and the management page. What's interesting is that I was also able to unfreeze it by disconnecting / reconnecting iOS devices from the stratux SSID, or by restarting hostapd over a wired SSH session.

I was not, however, able to pinpoint exactly which conditions caused lockups. It was not at a repeatable point in each log file, nor did it always occur at peak signal rate. In some cases, gen_gdl90 would happily digest 9000+ messages per minute, then choke at 2400 messages/min.

I've ruled out both power and temperature. Freezes (and normal operation) occurred at CPU temperatures ranging from 60°C and 30°C, and with two, one, or zero RTL-SDRs connected. 

Restarting hostapd with iOS devices connected immediately caused the management page to refresh, and in most cases caused data to begin flowing into ForeFlight again:

    cd /etc/init.d
    sudo ./hostapd restart

In addition to the Edimax adapter, I have a [TP-Link TL-WN722N adapter](http://www.tp-link.com/en/products/details/cat-11_TL-WN722N.html) sitting on a shelf. If I can find a working driver, it might be useful to see if it responds the same way.

egid, 2015-09-14:
I'm still working on getting replays working. :/ Not sure what's up with my setup; it works fine when run off the disk image, but the built-from-source bits don't know where the conf file is, which is weird.

I did manage to get a bunch more KSEA traffic today from the SW corner of my office building at work. No crashes or wifi drops on my iPhone 6.

cyoung, 2015-09-15:
@egid - which conf files? stratux.conf isn't needed for it to run.

@AvSquirrel - https://github.com/cyoung/stratux/commit/5dda5d62202d21ef968ca33963831aef086d0e58 could you try the latest? It does not fix the cause of the problem but it will greatly increase the threshold which may cause issues with the adapter. Before this, there is a ton of network chatter - for every message to any given app, there is at least two packets of waste (sent to another port, then an ICMP unreachable response sent back to the Pi).

AvSquirrel, 2015-09-16:
@cyoung - I downloaded, compiled, and bench tested this evening on a clean 0.3b1 image. For data sources, I looped @JohnOCFII 's UAT dump and @n4ohj 's [second ES dump](https://github.com/cyoung/stratux/issues/38) through replay.go at 10x real speed.

I experienced no lockups, even after running for three hours. With the 10X acceleration of the data sources, gen_gdl90 was reading a peak of about 9,500 UAT messages and 55,000 ES messages per minute.

The stratux.log doesn't show anything unusual, apart a 100 MB file size after three hours (872,144 "message queue overflow" messages and 1,040,186 "sleepy" status messages.) I'd suppress those messages for a release build.

[image]

[image]

[image]

cyoung, 2015-09-16:
Great work. I'll wait for another confirmation / flight test before closing it.

Will have this in mind in the future. At some point we may reach a top end on network traffic with this wifi adapter.

cyoung, 2015-09-16:
Any other testers? Changes now integrated into the latest beta image: https://github.com/cyoung/stratux/releases/tag/v0.3b2

egid, 2015-09-16:
I'll pull 0.3b2 down tonight and see if I can give it a shot. If nothing else, I'll get about 10 hours in over the next few days, I'll get you logs after that.

JohnOCFII, 2015-09-16:
> Any other testers?

I can pull down the fresh image and give it a shot from the ground/car.  I won't see the inside of an airplane until the weekend.

closed 2015-09-24 by cyoung