cyoung/stratux #38: Also experienced lockups using 0.3b1

n4ohj, 2015-09-12:
Was in a car within a few miles of DFW airport.  Unfortunately I did have both 978 and 1090 enabled.  When the unit gets in this state, the web page no longer responds, and the unit stops responding to Foreflight (however it is still showing connected via wifi).  

I power cycled the unit twice in the same location, and in both cases within two minutes the system had locked up again.  I have only seen this occur in very high traffic areas.

I also noticed before the system locked it had received about 800 Unknown UAT traffic type messages (I believe it was type 9999).

[Here](https://gist.github.com/n4ohj/8315b4edebb2039f410c) is the UAT file

I have the ES logs as well, but they are 192MB.

cyoung, 2015-09-12:
Could you zip up the ES log and send it?

n4ohj, 2015-09-12:
Sure, [here](https://www.dropbox.com/s/edxvculkhcc1d2t/stratux-es.zip?dl=0) it is on dropbox

cyoung, 2015-09-12:
When you're able, send over

     free -m
     cat /proc/cpuinfo

in idle state

n4ohj, 2015-09-12:
root@raspberrypi:~# free -m
             total       used       free     shared    buffers     cached
Mem:           927        279        647          0         35        180
-/+ buffers/cache:         63        863
Swap:           99          0         99
root@raspberrypi:~#  cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 57.60
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2709
Revision        : a01041
Serial          : 00000000df8bbf0f

cyoung, 2015-09-12:
How many clients connected reported in http://192.168.10.1?

What type of tablet/phone/computer? What app were you running? If FF, did you turn on/off the screen often?

n4ohj, 2015-09-12:
I only had my iPhone 5s and 4th gen iPad with me. Both devices were connected via wifi, but I only had FF running on the tablet (nothing on the phone).  I was not switching between FF and the web browser often- I only went to the Stratux web page once or twice to check the number of connections, and to verify the Stratux had locked up. 

I can run this test again if it would be helpful.

cyoung, 2015-09-12:
Could you check through the system logs? http://192.168.10.1/logs, message log and such. Maybe see around the time it crashed and you restarted it if anything was indicated.

n4ohj, 2015-09-12:
I've looked through the logs.  The only thing I see in the messages file between reboots that's remotely suspicious is a bunch of nulls just prior to the reboot.  I will see if I can do another run without the GPS and 1090 next time (and with reset logs).

n4ohj, 2015-09-13:
[Here](https://www.dropbox.com/s/lwpzb540ij9b87p/n4ohj_stratux_09132015.zip?dl=0) is a new dump from today.  This was taken at the same location as the dump yesterday, within close proximity to DFW airport.  Also, I used only a single dongle, no GPS/AHRS, and only the 978Mhz option enabled in the web UI.  The log files should be very clean, as I had truncated them prior to powering Stratux up near the airport.

There are three data collection runs.  In all three cases, the system locked up within about three minutes.  In each run, I powered up the unit, started FF on my ipad, then brought up the Stratux web UI on my iphone. 

In each case, the webui was showing about 2800 messages per minute just before the system locked up.  "Locked up" means foreflight no longer showed it was talking to "FreeFlight", and also the webUI wasn't responding.

I attempted to log in to the unit using Reflection on my ipad (via SSH).  It prompted me for the username, but never prompted for the password.

Please let me know if you have any questions.  I hope this helps.

cyoung, 2015-09-15:
Try https://github.com/cyoung/stratux/commit/5dda5d62202d21ef968ca33963831aef086d0e58 if able, if not will be in next beta image.

cyoung, 2015-09-16:
Try out https://github.com/cyoung/stratux/releases/tag/v0.3b2

n4ohj, 2015-09-17:
Two thumbs up on this release.  I drove to the same place we'd seen consistent crashes previously.  Everything worked perfectly.  I was able put the ipad to sleep, wake up and go back to FF without issues, and the traffic volumes were around 18,000.  [Here](https://www.dropbox.com/s/gblre999hsada7x/n4ohj_stratux_09162015.zip?dl=0) are the logfiles if you'd like to look at them.

cyoung, 2015-09-17:
Excellent test data set, @n4ohj :

[image]

cyoung, 2015-09-20:
Anyone else that was having this issue able to test?

closed 2015-09-24 by cyoung