cyoung/stratux #39: Log from UAT today - 0.3b1

JohnOCFII, 2015-09-12:
I'm hoping that the log file is persistent across shutdowns, as I flew for about 90 minutes, shutdown, drove home, powered up and grabbed this log.

No known crashes.  About halfway through the flight the ForeFlight display did show 'o' for aircraft altitudes.  Selecting the target did show the correct altitude.  I've attached a screen shot in addition to the UAT log.

Here is the zip'd up log file:  https://dl.dropboxusercontent.com/u/2506999/stratux-uat.log.zip

[image]

cyoung, 2015-09-12:
Thanks for this! Logs are persistent across reboots, but there is a logrotate process that may move them around and/or compress them. You can see all of the logs in the folder http://192.168.10.1/logs/. If it was rotated, it will still have the "stratus-uat.log" prefix.

cyoung, 2015-09-15:
[image]

cyoung, 2015-09-15:
Thanks @JohnOCFII.

closed 2015-09-15 by cyoung