cyoung/stratux #44: Other Wifi Adapters

Ylinen, 2015-09-17:
I have a Pi Kit that uses another wifi adapter that I would like to use for this project.  www.canakit.com.

Is it possible to add the drivers for other adapters to the image?

cyoung, 2015-09-17:
If it's a Pi 2, get this to replace the adapter: http://amzn.com/B003MTTJOY.
If it's not a Pi 2, get the recommended hardware kit.

It's not possible to add drivers for other adapters at this point.

closed 2015-09-17 by cyoung

bradanlane, 2015-09-17:
The easiest thing to do is to limit the supported hardware. 

If we want to support WiFi dongles which use RT5370 (a very common chip) we need to use an updated hostapd. I did a build of hostapd to include support for the nl80211 driver. The result works for both the original specified wifi and the ones using RT5370.

AvSquirrel, 2015-09-17:
@bradanlane - I've also been experimenting with a [hostapd build](https://github.com/lostincynicism/hostapd-rtl8188) that appears to use a newer, patched version of rtl871xdrv. Stability under high UAT loading on 0.3b1 and earlier seemed better than with the hostapd included in the release images.

It can also be compiled with support for nl80211 devices, meaning that rt2800usb (e.g. RT5370), ath9k-htc, and other kernel modules would be compatible out of box. I tested an older gen_gdl90 builds with an ath9k-htc compatible adapter has been pretty much bulletproof, even under heavy load.

Might be worth thinking about forking an image at some point after the project matures.

@Ylinen - If you're feeling adventurous, you could try installing a nl80211 compatible release of hostapd, and update /etc/hostapd/hostapd.conf to get the adapter working. Otherwise, pony up the $10 for the Edimax adapter.