cyoung/stratux #46: Stratux not seeing GPS data from GPIO TX/RX connection

JohnOCFII, 2015-09-18:
I've got Stratux v.03b2, and was attempting to use a UBLOX GPS.  I'm seeing traffic on /dev/ttyAMA0, and the stratux.log says it is getting GPS data, but neither the Stratux WebUI nor ForeFlight are seeing that I have GPS.

Log file here:  https://dl.dropboxusercontent.com/u/2506999/stratux09172015.log.zip

GPS tty output here: https://dl.dropboxusercontent.com/u/2506999/GPSTest09172015.txt.zip

Any ideas?

Thanks,

John

cyoung, 2015-09-18:
I see "$GPGGA" sentences here. That means that you changed the settings on the RY835AI to only use GPS. Temporary fix is to hook it up to u-center and set it to also use GLONASS. I'll add code now to support both general "GNSS" and "GPS" sentences.

cyoung, 2015-09-18:
Fixed by https://github.com/cyoung/stratux/commit/68d999f5e7b1b26855953c4f98e6543aadc068f9.

closed 2015-09-18 by cyoung

JohnOCFII, 2015-09-18:
Thanks!

I'm using a different GPS  (Ublox NEO-6M) unit that doesn't support GLONASS.  It's a unit I use with my RC quadcopters.

https://www.u-blox.com/en/product/neo-6-series

http://multirotormania.com/gps-modules/426-ublox-neo-6m-no-compass.html?search_query=gps&results=10

cyoung, 2015-09-18:
Sounds good, if you're able to recompile it should be working now. I tested it with the .txt file you sent and it was an easy mod.