cyoung/stratux #47: Magnetometer drift?

ssokol, 2015-09-18:
Received my RY835AI yesterday and have it connected and working. On the bench, in ForeFlight, the compass / DG indicator continuously drifts in a circle and the SynVis display appears to orbit around the PoV as if the aircraft was in a continuous right turn. It appears to be turning at about 2 - 3 degrees per second. In the absence of a track value from the GPS, is the heading being set by the magnetometer? If so, wow does it drift.

Please let me know what logs / data I can attach that may help answer this.

cyoung, 2015-09-18:
No, this is purely from the gyro. The AHRS code isn't ready to be used really and what you see there is mostly for testing out AHRS features in different apps.

closed 2015-09-18 by cyoung

fjwarren, 2015-09-18:
I'm seeing the same as well, and was thinking I may have had a defective module; thanks for info & I look forward to  further testing when it's released