cyoung/stratux #49: GPS fails silently after boot

ssokol, 2015-09-18:
Running v0.3b2 I seem to be having some kind of intermittent problem accessing the GPS. The process starts and other features (1090ES, AHRS, baro altitude) appear to work, but I never receive a GPS location and the web console shows 0 satellites. If I kill all processes and restart the stratux service, then the GPS works. The RY835AI is indicating a solid fix (red blinking GNSS LED).

Here's the log: https://gist.github.com/ssokol/0e417e9b0ded417d5fc5

cyoung, 2015-09-19:
Hmm, there are some i2c errors:

     2015/09/16 20:41:20 readBMP180(): open /dev/i2c-1: no such file or directory

Are you using the pre-built image or was that from something else?

Is this with the latest (https://github.com/cyoung/stratux/commit/10f21a651be84023a0a35d42d7d5f4d20ada94bc)? Just added some auto config for the RY835AI GPS.

ssokol, 2015-09-19:
That's the latest .img without any changes: https://github.com/cyoung/stratux/releases/download/v0.3b2/stratux-v0.3b2-5a568e27ce-09162015.img.zip

cyoung, 2015-09-19:
Were you able to identify the i2c error? Was the device hooked up to the GPIO pins correctly at that point or is it still throwing this /dev/i2c-1 error?

ssokol, 2015-09-19:
I had to run the enable function / script from raspi-config twice before it created the device. That was the source of that error. Once the device was enabled the i2c functions - AHRS, baro alt - worked consistently.

ssokol, 2015-09-19:
FWIW - I think this may be essentially the same issue as described here:

https://www.reddit.com/r/stratux/comments/3ldtaf/anyone_using_ry835ai_gps_via_gpio_with_v03b2/

cyoung, 2015-09-19:
Have you changed the settings on the NEO-M8N at all?

On that thread, the OP said he had a "defective jumper wire" (probably a bad solder joint). Another guy was using a GPS other than RY835AI and the NMEA sentences weren't understood by gen_gdl90. The last guy has an unknown issue so far.

Your log shows a lot of system setup issues like two gen_gdl90 running at the same time. Would you be able to try the v0.3b3 image with no mods and see how it works with your hardware setup? It has an auto-config now for the NEO-M8N which should reduce the number of factors we're working with here.

ssokol, 2015-09-20:
Will do. Downloading it now.

ssokol, 2015-09-20:
Ok. Just installed the new version. After installation I used the raspi-config utility to:

- Expand the filesystem
- Change the default password
- Enable i2c*
- Disable the serial console

I rebooted, connected from my iPad and enabled:

- 1090ES
- GPS
- AHRS

AHRS light went intermittently green / red. Satellite count remained at 0. 1090ES message count began incrementing. No UAT messages but that's normal - I'm not anywhere near a UAT tower. 

I checked the /dev directory and found that in spite of having run the i2c initiation script, there were no i2c devices. Did a reboot. After reboot (which required a hard power cycle to work properly) there were still no devices. I re-ran the script and rebooted again (again required power cycle to boot properly) the i2c devices appeared to be there. I tried ForeFlight and AHRS / barometric altitude were both working. No GPS.

I killed all Stratux processes (gen_gdl90, dump1090) and checked /dev/ttyAMA0 using screen (screen /dev/ttyAMA0 9600) and found that it was outputting GPS data properly, though the initial data dump was non-printable characters. After a second or two the terminal began outputting standard NMEA sentences with an occasional hiccup:

$GNRMC,155403.60,A,3915.09110,N,09426.26245,W,0.494,,200915,,,A*73
�b
  txbuf allocc5$GNRMC,155403.70,A,3915.09109,N,09426.26248,W,0.300,,200915,,,A*7D

I restarted the Stratux service and this time the GPS worked properly, showing 10 satellites in view in the admin page and 3m accuracy in ForeFlight.

It looks like there are several issues here:

The i2c activation script in raspi-config seems to fail to make the necessary changes. Perhaps this only happens if it is run in conjunction with other actions that require a reboot (i.e. expanding the filesystem or shutting off the serial console).

The GPS reader / parser is not activating properly during the initial startup sequence.

Something seems to have changed with the wifi implementation that causes it to hang on soft reboot, requiring a hard reboot (power cycle) to re-enable proper function. Specifically, after a soft reboot the network beacon is visible (the “stratux” SSID appears) but devices are unable to connect to it.

ssokol, 2015-09-20:
Oh, in regards to changes to the Ublox GPS - I followed the instructions in the link below to set the rate to 10 Hz and the baud rate at 9600. (I rather wonder if this is a safe combination - the data output at 10 Hz may be beyond what can be pushed across a 9600 bps connection.)

http://www.reyax.com/Module/GPS/RY835AI/RY835AI.pdf

cyoung, 2015-09-20:
Was it really necessary to do raspi-config? i2c and serial should have the correct settings from the stock image. I think when you did this the first time you actually turned them off when they were on, then had to turn them back on again.

It's changing the settings of the NEO-M8N to 115200 baud and settings to 10Hz update on each connection now.

cyoung, 2015-09-21:
Using only the web interface to enable GPS+AHRS after stock image install, is it working?

ssokol, 2015-09-21:
AHRS comes up and works through the groundspeed, altitude and vertical speed in ForeFlight intermittently switch between valid values and the "X" boxes.

GPS does not work. 0 satellites showing in the admin page. No fix showing in ForeFlight.

Here's the stratux.log output:

2015/09/18 20:47:24 Stratux v0.3b3 (c3e6d41ebe6f7dbdb842e99bf0c2a7b5a0025ed5) starting.
2015/09/18 20:47:24 can't read settings /etc/stratux.conf: open /etc/stratux.conf: no such file or directory
2015/09/18 20:47:25 ===== UAT Device name: Generic RTL2832U OEM =====
2015/09/18 20:47:25 	GetUsbStrings - Realtek                                                                                                                                                                                                                                                           RTL2838UHIDIR                                                                                                                                                                                                                                                     00000001                                                                                                                                                                                                                                                         
2015/09/18 20:47:25 	GetTunerType: RTLSDR_TUNER_R820T
2015/09/18 20:47:25 	SetTunerGainMode Successful
2015/09/18 20:47:26 	SetTunerGain Successful
2015/09/18 20:47:26 	SetSampleRate - rate: 2083334
2015/09/18 20:47:26 	GetSampleRate: 2083334
2015/09/18 20:47:26 	GetXtalFreq - Rtl: 28800000, Tuner: 28800000
2015/09/18 20:47:26 	SetXtalFreq - Center freq: 28800000, Tuner freq: 28800000
2015/09/18 20:47:26 	SetCenterFreq 978MHz Successful
2015/09/18 20:47:26 	GetCenterFreq: 978000000
2015/09/18 20:47:26 	Setting Bandwidth: 1000000
2015/09/18 20:47:26 	SetTunerBw 1000000 Successful
2015/09/18 20:47:26 	ResetBuffer Successful
2015/09/18 20:47:26 	GetFreqCorrection: 0
2015/09/18 20:47:26 	SetFreqCorrection 0 Failed, error: invalid parameter(s)
2015/09/18 20:47:54 client connected: 192.168.10.12:4000 (Stevens-MBP-2).
2015/09/18 20:47:54 client connected: 192.168.10.12:43211 (Stevens-MBP-2).
2015/09/18 20:47:54 client connected: 192.168.10.12:49002 (Stevens-MBP-2).
2015/09/18 20:48:24 client connected: 192.168.10.10:4000 (SMSi7).
2015/09/18 20:48:24 client connected: 192.168.10.10:43211 (SMSi7).
2015/09/18 20:48:24 client connected: 192.168.10.10:49002 (SMSi7).
2015/09/18 20:48:39 wrote settings.
2015/09/18 20:48:40 wrote settings.
2015/09/18 20:48:44 Using /dev/ttyAMA0 for GPS
2015/09/18 20:49:44 calibrate: pitch -6.065178, roll -2.594960
2015/09/18 20:49:54 client connected: 192.168.10.11:4000 (StevenSiPadMini).
2015/09/18 20:49:54 client connected: 192.168.10.11:43211 (StevenSiPadMini).
2015/09/18 20:49:54 client connected: 192.168.10.11:49002 (StevenSiPadMini).

ssokol, 2015-09-22:
I defaulted the GPS to factory settings and that seems to have fixed this. Still don't understand exactly what was happening. Closing. If anyone else runs into a similar issue please re-open.

closed 2015-09-22 by ssokol

cyoung, 2015-09-22:
Might want to add "factory reset" to the NEO-M8N init

jpoirier, 2015-09-23:
Doing a reset without the users consent is a bit heavy handed and certainly would not be appreciated by anyone that may have spent time experimenting with, and/or tweaking, the settings.

The documentation clearly states 10Hz is supported for GPS only and 5Hz for all other combinations.

cyoung, 2015-09-23:
Using UBX protocol to set up on init - 115200 baud, "Airborne with >2g Acceleration" dynamic model (so they call it), 3D fixes only, and 10Hz update. Other than 5Hz updates do you have some other settings you think might be ideal?

jpoirier, 2015-09-23:
Nothing I'd consider ideal at this point but there are some things/settings
I don"t really understand and haven't had a chance to do any
research/testing/confirmation on.

There may have been some small performance benefits tweaking the min and
max values under CFG/GNSS, but again nothing confirmed at this point.

Once I'm back home I'll connect via GPIO and try what I have.