cyoung/stratux #51: Feature Request - Client Mode Wifi

ssokol, 2015-09-21:
This would be useful both for debugging and for scenarios where there is already a wifi network in the aircraft. At first booth Stratux should start up in AP mode as it does now. The UI should allow the user to toggle between AP and client mode. In client mode, the admin would specify the SSID and password. If the Stratux did not establish a connection to the specified SSID it should fall back to AP mode.

bradanlane, 2015-09-21:
Do you have a test case for "client mode"? If Stratux is a client, would ForeFlight (or other) still know how to connect?

I do see a use for "client mode" when back at home, FBO, etc. it could submit a log, bug report, or check for software updates.

ssokol, 2015-09-21:
My personal current use case is development / debugging. Being connected to
my local network makes it much more convenient to SSH in, pull logs, change
settings, etc. while still connected to the Internet.

I don't know if ForeFlight would work under every scenario. If it's
hard-coded to look for a device on 192.168.10.1 then you would have to set
up your home / lab network to allow the Stratux to take that IP - not a big
deal. If it uses something like Avahi / Bonjour to find data sources we
might be able to add support. Other EFBs and experimental EFIS are probably
more flexible given that they're designed to interface with multiple data
sources.

In a "production" scenario it might also be a good way to update the
Stratux. Appareo uses ForeFlight as the means of installing an update -
something we can't do. The easiest solution to that may be to take it home
and either plug it into an Ethernet port (if you have one available, and if
your Stratux case exposes the Pi's port) or to kick it into client mode and
let it pull down updates.

bradanlane, 2015-09-21:
Thanks. It looks like we are on the same page for te use cases of "client mode".

cyoung, 2015-09-21:
gen_gdl90 finds client using the DHCP leases, so that would need to be changed also. To put it into client mode update-rc.d isc-dhcp-server disable, same for hostapd. Add wifi info in /etc/wpa_supplicant/wpa_supplicant.conf, and un-comment out the dhcp settings /etc/network/interfaces and comment the static IP stuff out.

ssokol, 2015-09-21:
Ok. That all sounds do-able. A couple of thoughts:

It appears that gen_gld90 currently starts sending UDP streams to any
device that receives an IP address from the DHCP server.  Could we instead
send a single stream to the broadcast address? The level of traffic will
actually be lower than with independent copies of the same stream being
sent out. (Not sure if this is a sane proposal or not...)

Rather than editing the /etc/network/interfaces it might be easier to just
recreate it.

If running in client mode, there needs to be some kind of watchdog that
checks to make sure the Pi has a connection and an IP from the access
point. If it doesn't, the watchdog should toggle the system back into AP
mode. Perhaps give it 1 minute

Client mode will need to support both manually configured static IPs and
DHCP. Static is pretty easy. DHCP poses the challenge of letting the user
know what IP address the system received. If we're already pumping data to
the broadcast address we could simply beacon on a known port, providing the
IP. Otherwise we could add the avahi daemon and use bonjour (multicast DNS)
to find it. IIRC, Linux and Mac come with support for mDNS, but Windows
does not.

cyoung, 2015-09-21:
>Could we instead send a single stream to the broadcast address?

The older versions used this method and it was unreliable - especially for Android devices. There's a thing called "Multicast Lock" that must be acquired by the app - a lot of them don't - before it receives broadcast messages. Like this: https://github.com/apps4av/avarehelper/commit/18bded376a44df94f22a835e6f6fbc9023488fe6

Also, for one reason or another there was a lot of jitter in the multicast packets.

ssokol, 2015-09-21:
Gotcha. So in AP mode it would continue to do what it's doing now - find clients via the DHCP reservations list. In client mode it would require some additional configuration to set destination addresses for the data streams.

gms5002, 2015-09-30:
FWIW you can easily add another wifi dongle for 10 bucks (ie http://amzn.com/B003MTTJOY) which will allow you to run the AP and connect to your local wifi network at the same time.

bradanlane, 2015-10-20:
the current AP mode is taking 90 seconds from teh time I select "Stratux" to the time I get the check-mark. This is iOS9.0.2 on an iPhone 6.

What are others seeing?

AvSquirrel, 2015-10-20:
See https://github.com/cyoung/stratux/issues/76 .

It's known issue with the 0.0.0.0 default route on iOS 9 (iOS 8 had similar behavior, though it timed out after a shorter time). However, the network (SSH, web UI, ForeFlight connectivity) is usable nearly instantly upon selecting "stratux", even though the spinning icon is displayed.

bradanlane, 2015-10-20:
yes. I guess my point is 90 seconds is a long wait and yesterday I wasn't thinking it would ever happen so I set a static connection. Should I re-open that issue or a new one or jsut forget about it?

rgutlon, 2015-10-20:
I think the connection delay is a non-issue *as long as user expectations are set* in a either a read-me-first or release notes file. Otherwise, it has the potential to generate repetitive support postings as folks (incorrectly) presume the system is not working ... or waste time fiddling with stuff that is not necessary.

cyoung, 2015-10-20:
Well, as far as I know right now the decision is either 1) have fast connects for iOS (Android is no issue at all) or 2) have working cellular internet, and not both. #76 didn't produce any solutions which allowed us to have both. This just came along recently with newer versions of iOS. Chances are it's breaking something else.

bradanlane, 2015-10-20:
Since I'm the only one whining, I'd say we don't mess with it. 

At the same time, if you could tell me how to "make it fast but no cellular" I'd like to use that on my personal setup.

cyoung, 2015-10-20:
subnet 192.168.10.0 netmask 255.255.255.0 {
		range 192.168.10.10 192.168.10.50;
		option broadcast-address 192.168.10.255;
		option routers 192.168.10.1;
		default-lease-time 12000;
		max-lease-time 12000;
		option domain-name "stratux.local";
		option domain-name-servers 4.2.2.2;
	}

At the bottom of dhcpd.conf. The only difference is the "routers" line.

rgutlon, 2015-10-20:
edit: /etc/dhcp/dhcpd.conf
change: option routers 0.0.0.0;
to option routers 192.168.10.1;

JohnOCFII, 2015-10-20:
> yes. I guess my point is 90 seconds is a long wait

Just to confirm are you aware that although iOS UI doesn't show the checkbox for 90 seconds, you you can start to use the WiFi connection within the normal 5-10 seconds -- you just don't get the visual indication of a connection in iOS.

bradanlane, 2015-10-20:
My experience has been different. For the 90 seconds I have no IP address, route, or anything else ... and sometimes I have the dreaded 254... address.

JohnOCFII, 2015-10-20:
> My experience has been different. For the 90 seconds I have no IP address, route, or anything else ... and sometimes I have the dreaded 254... address.

Interesting.  I get no data in the Settings --> Wi-Fi area to show a connection, but once I see 'stratux' in the list, and select it, I can then switch over to ForeFlight or the WebUI and start to see data within seconds.  I agree that if I wait on the Settings --> Wi-Fi page, it can be 90 seconds to see anything.  I just stopped waiting :) and instead started to wait for valid connectivity within ForeFlight or the WebUI and was rewarded.

loather, 2015-10-24:
Why even have a routers option in the first place? If you remove the option completely it won't send a default route and therefore won't send a broken route either.

cyoung, 2015-10-24:
Give it a try - how's that work?

loather, 2015-10-24:
Checkmark shows up in a second or so and cellular still works (posting this from my iPad). Still don't get the wifi logo in the upper-left but I'm not worried about that too much.

JohnOCFII, 2015-10-24:
> Checkmark shows up in a second or so and cellular still works (posting this from my iPad). Still don't get the wifi logo in the upper-left but I'm not worried about that too much.

Nice find!

Checkbox is maybe 3-4 seconds for me, but way faster than before.  I can still get to the Internet via Cellular on my iPhone too.  While there is still no WiFi icon, at least there is a relatively quick visual indication that you are connected (via the checkmark on the Settings --> WiFi page.

I also used to get a message about not being connected to the Internet when launching apps.  I don't get that message any more.

cyoung, 2015-10-25:
Excellent! @kdknigga - are you able to try this on Android 6 (Marshmallow)?

kdknigga, 2015-10-25:
I'll give it a whirl Sunday evening and report back.

kdknigga, 2015-10-25:
Looks good!

cyoung, 2015-10-26:
Awesome, thanks for testing.

Gold star for @loather.

cyoung, 2016-03-08:
Duplicate #228.

closed 2016-03-08 by cyoung