cyoung/stratux #52: Feature Request - Development Stack On Release Images

ssokol, 2015-09-21:
It would be quite useful if the Golang development stack came preinstalled on releases - at least any alpha / beta releases. The software is already available in /root/stratux. If would be convenient if that could be edited, compiled and deployed without having to connect the device to the Internet and manually install the development tools.

Alternately, we could include a script to quickly install the necessary software.

cyoung, 2015-09-21:
Makes the image about 60MB bigger.

JohnOCFII, 2015-09-21:
> Alternately, we could include a script to quickly install the necessary software.

Maybe go with this option -- space only used for those who want/need it, but makes it easy to get installed.

ssokol, 2015-09-21:
I'm not sure anyone would notice the extra 60 MB - either in storage used
or in the additional download time - but I'm good either way.

haykinson, 2015-09-21:
+1 on the request to just add it to the image. This way the environment is ready for making changes.

bradanlane, 2015-09-21:
Just an FYI for those who forgot what "most of America" has for internet connectivity, that extra 60mb translated to another 3-6 minutes of download.

ssokol, 2015-09-21:
True, but the process of installing and configuring the dev tools is also a
lengthy process. I'm all for stripping it down to the minimal size for
stable / production releases, but it seems like a win to have the tools in
place on any pre-release / alpha / beta build.

cyoung, 2015-09-21:
Already included for debug images, 60MB comment for documentation for non-debug release images.

ssokol, 2015-09-21:
Hmmm... I'm running 0.3b3 and I don't find the go complier, mercurial, or
any of the other build tools on the image. There's a copy of the source in
/root/stratux, but I have to manually install go, etc. to build it.

cyoung, 2015-09-21:
Already included for future debug images, 60MB comment for documentation for non-debug release images.

ssokol, 2015-09-21:
Ack. Thanks!

commit e43c509a (2015-09-22): Fixes #52 (mostly).

go doesn’t run under qemu, so you’ll need to get the go imports on
there somehow.
closed 2015-09-22 by cyoung