cyoung/stratux #54: [0.3b2] Occasional 0 altitude issues, mostly just 10 hours of logs.

egid, 2015-09-22:
[stratux-uat.log - 107MB](https://drive.google.com/file/d/0B3DKMWgF5KKtdEpzbWRndDlxRzg/view?usp=sharing)

Took a trip from Seattle to Sacramento this weekend, about 5 hours each way. Had a couple instances of the 0 altitude issue; I also had an issue where ownship would get stuck at 4000-odd feet, which screws up nearby traffic altitudes (at 8500, aircraft on the ground at 500msl show as -35, ish).

cyoung, 2015-09-22:
I see 11 startups logged here - did you have to restart it for having locked up at all?

[image]
[image]

When you got the 0 relative altitude issue, did you do anything specific to resolve it? Force close, etc. Anything special seem to work?

Looks like there were only a handful (5-6) intervals of 5 minutes where you went without the latest METARs (QoS graph) within 100nm.

egid, 2015-09-22:
I didn't have any lockups. I did restart it once to clear the 0 alt issue, and there were a few tests from the hotel in Reno just to see if I could get a tower. 

0 alt was fixed a few ways: cycling wifi on and off (had to wait for ADS-B to drop off); cycling power on and off (both iPads had the issue and I was lazy); force-quitting ForeFlight and relaunching (least-good option, loses route progress). Best solution seems to be cycling iPad wifi, but I didn't try it more than maybe twice so I'm not sure if it's a magic bullet.

egid, 2015-09-22:
Actually, now that I think of it, most of the fixes I listed were not for 0 alt but the ownship relative alt. The only time I had 0 I just cycled wifi and it resolved itself.

cyoung, 2015-09-24:
Thanks for sending this!

closed 2015-09-24 by cyoung