cyoung/stratux #55: Request: Binding 978/1090 process to specific SDR?

dmurray14, 2015-09-22:
This came up in a Reddit thread but I'm not sure it was really fleshed out. I've hand some 978 and 1090 antennas made (hopefully will be here this week then I can get them out to everyone). Since they are tuned to each respective frequency, it would be good to bind the UAT and ES processes to the correct receivers. Has anyone determined if there is a way to detect which one is which, and bind the process to one or the other? Is one USB port always the lower numbered one?

cyoung, 2015-09-22:
Is the order that USB devices are "ordered" on bootup deterministic?

bradanlane, 2015-09-22:
I did some more searches and the following discussion thread suggests the device order will change when devices are added or removed but more important there is the change they change from boot sequence to boot sequence based on timing of processes and race conditions. 
http://raspberrypi.stackexchange.com/questions/12246/why-does-usb-port-enumeration-change

My experience *appears* that devices are stable from boot to boot and ports are sequential. **However** I may find its just current coincidence and it all changes a week or month from now.

lastdeadmouse, 2015-09-22:
It could be possible but only if the SDRs supply a unique identifier. In
the past, I've resolved certain issues by creating symlinked devices this
way for other stuff, like tuners in a htpc. Haven't looked into with the
SDRs though.

ssokol, 2015-09-23:
IIRC (and this is coming from memory, so please don't trust it as gospel truth) there is a way to map USB devices based on the port into which they are inserted. You can create udev rules that then enforce device identity based on that mapping - maybe? Here's a link that might help:

http://askubuntu.com/questions/49910/how-to-distinguish-between-identical-usb-to-serial-adapters

cyoung, 2015-10-19:
Duplicate, moving to #79.

closed 2015-10-19 by cyoung