cyoung/stratux #57: Stratux v0.3b3 hangs on soft reset (reboot)

ssokol, 2015-09-23:
When executing the "reboot" command as root (sudo su -), the system hangs a some point in the cycle. I can't tell if it's during the shutdown phase or the subsequent startup phase. During the shutdown process both the red power / user LED and the the green disk activity indicator LED come on for a sustained period. Eventually (10 seconds?) the green LED goes out leaving a solid red. Network and USB are both shut down (LEDs go off on the SDR and Wifi modules).

This has happened on two different Pi units, each running a different copy of the 3b3 image. If somebody else can confirm this, I'll try to hook the Pi up to an HDMI monitor and see what that shows during the reboot process.

fjwarren, 2015-09-23:
I have noticed that "sudo shutdown -h now" does not power down the pi (Stratux v0.3b3), as it seems to on other pi projects I have built.  Perhaps this is related; I have not noticed a -r failing, but can't say I have used it.  I'll set it up with a monitor in my lab after lunch and see what is displayed.

ssokol, 2015-09-23:
This might be a clue... I have two v0.3b3 Straux systems. One is a
single-band UAT - no 1090, no RY835AI. The other is currently a single band
with the RY835AI attached and both AHRS and GPS enabled. The unit without
the RY835AI reboots and powers down cleanly. The until with the RY835AI
does not. Identical build on both.

Makes me wonder if having the I2C bus and/or UART active is causing some
kind of issue. I'll get the unit with the RY835AI on a monitor this evening
and see what it shows.

trippingly, 2015-09-23:
I'm running v0.3b3 from source on a UAT-only setup, no RY835AI. 

Like your single-band, UAT-only setup, "sudo shutdown -r 0" performs a normal reboot for me, with no continuous green LED - just the normal flickering green LED.

fjwarren, 2015-09-23:
I have a RY835AI connected to I2C & Serial.  I just tried a "sudo shutdown -r now", and the reboot appears successful.  When I do a "sudo shutdown -h now", I see what appears to be a proper shutdown, ending with "[info] - will now halt." About 2 seconds later, a fresh boot up starts, as though the power were cycled. 

With the RY835AI disconnected, no change.  If I disconnect both RTL-SDR dongles, the -h works as expected.  With either or both  RTL-SDR dongles connected, the pi will start again approximately 2 seconds after the halt is complete.

So, I guess this is different than the "hang" the OP saw, but interesting.  Anyone have any suggestions?

cyoung, 2015-10-20:
Reproduced, defunct gen_gdl90 hangs the soft reboot. Fixed with #63.

closed 2015-10-20 by cyoung