cyoung/stratux #59: request: interesting data

bradanlane, 2015-09-24:
While it may not be used extensively once we get Stratux to a "release" level of capability, it would be useful *and get users to stay engaged with the project* if the WebUI had the option to show data such as:

* current tracking aircraft (aka, ones we have heard from in teh past 60 seconds) - the WebUI would display a list with tail number, lat, lon, last time received, etc.
* weather messages - display a list with type, date/time, location (likely ICAO or FAA identifier), and content
* other ?

I can easily integrate this into the new WebUI 'app' if someone can provide the data from teh backend via the socket layer.

ssokol, 2015-09-24:
I've been poking at the idea of a web-based PFD / EFIS for the past couple of years. I would love to see all of the ADS-B data made available over the websocket on a pub/sub basis. All the text products could be encoded in JSON. The binary products (NEXRAD) could either be sent as Base64, or better yet simply write out a file and pass the URL. This opens Stratux up as the basis for all kinds of cool web and mobile apps.

cyoung, 2015-09-24:
https://github.com/cyoung/stratux/commit/63173fc88d805acd6df3bfcd2552c15941b9e09a

closed 2015-10-08 by cyoung