cyoung/stratux #61: FlyQ 2.1 Incompatibility

rgutlon, 2015-09-25:
FlyQ 2.1 was released the day before yesterday and it's not working (again) with stratux.

Some may recall FlyQ doesn't allow users to select the ADS-B device ... they attempt to automatically connect based on protocol or some other variable.

Under FlyQ 2.0 CYoung determined changing the SSID to NavWorxWIFI was the solution ... and it worked! But reports are coming in this is not working with either stratux v0.3b3 or v0.2r1 and FlyQ 2.1. Same behavior as before ... immediate (and repeated) connect and disconnects from the ADS-B device. A bit disappointing as a number of us wrote Seattle Avionics and were under the impression this would no longer be an issue in this release.

Thought I would mention this here in case someone sees an easy workaround on the backend.

https://www.reddit.com/r/stratux/comments/3iv2zk/flyq_efb_stratux_not_working/cve3tsq

cyoung, 2015-09-26:
I just tested it out. The problem is it is now listening for everything (you don't need to change the WiFi SSID name). It gets confused because it is receiving ADS-B data on more than one port.

Write this to /etc/stratux.conf

     {"UAT_Enabled":true,"ES_Enabled":false,"GPS_Enabled":false,"NetworkOutputs":[{"Conn":null,"Ip":"","Port":43211,"Capability":5}],"AHRS_Enabled":false,"DEBUG":false,"ReplayLog":false,"PPM":0}

This will disable the other ports. Just tried it out and it works well.

closed 2015-09-26 by cyoung

rgutlon, 2015-09-27:
With your code in /etc/statux.conf, I'm able to connect to the stratux device with FlyQ 2.1

Can you think of any reason why the GPS indicator within the FlyQ app, which previously indicated green (accurate) when used with the NavWorxWIFI SSID, is now "blue?"

According to the FlyQ docs, blue means FlyQ is in simulator mode. However, it did (correctly) track my location. Is there a setting in the conf file that could trigger this?

It's kind of a shame this is the one app we all seem to be struggling with. FlyQ tries to automatically detect the +- 12 ADS-B devices they say they are compatible with. Have a feeling it would be a seamless integration if users could select which one they have manually.

cyoung, 2015-09-27:
Is your GNSS light blinking (red) and you see satellites on the status page http://192.168.10.1?

rgutlon, 2015-09-27:
Using 2r1 with iPad internal GPS. No satellites indicated on the status page. See attached image - FlyQ recognizes internal GPS and it works ... but something is throwing off the status indicator.
[image]

stevepodradchik, 2015-09-30:
The graphic shows GROUND STATIONS (ADS-B Towers) not satellites.  It's the ADS-B tab, not the GPS tab.

The indicator is not blue, it's purple.  That means the ADS-B device is not supplying the GPS feed and we're getting it somewhere else.  Virtually all commercial ADS-B receivers have a built-in GPS but the Stratux project does not.

- Steve
Seattle Avionics

cyoung, 2015-09-30:
@stevepodradchik - thanks for posting. So in this case, FlyQ for rgutlon is using the iPad's GPS?

@rgutlon - do you have the RY835AI hooked up? The stratux status page shows the number of GPS satellites that stratux is receiving and isn't related to what FlyQ is using (which appears to be in your case iPad internal GPS). If you don't have the RY835AI/NEO-M8N GPS hooked up, then that is expected.

stevepodradchik, 2015-09-30:
Hi Chris,

Yes, it's using either the iPad's internal GPS (if applicable) or, if a Bluetooth device is connected, would be using the BT GPS.  Note that the iPad sends the GPS location to us, we don't directly connect to the GPS and the iOS message does not indicate if the location is from the internal GPS or an external GPS so we just display "iPad" in the GPS source in the GPS portion of that popup.

rgutlon, 2015-09-30:
Steve - thanks for taking the time to add to this post.

Under FlyQ 2.0, the GPS status indicator was green when used with the stratux and the iPad's internal GPS.  When we all had trouble connecting again after the 2.1 update, and the GPS indicator went "purple," many us presumed the GPS function was not working properly as this color is not documented as a valid GPS indicator in the 2.x documentation or videos.

Chris - understand your comments - in this test scenario I was simply using the iPad's internal GPS.

stevepodradchik, 2015-09-30:
Chris, is there an ADS-B message that Stratux sends with the device name?  I'm not an iOS or ADS-B developer so I may not have the terminology right but if there are any non-stanard messages we could use, can you email them to me?  In particular, we handle ADS-B systems w/o a battery (I presume like the Stratux) a little differently so the ADS-B Battery indicator does not flash red.  If we could trap for a particular device ID, we can do that properly.

The Purple GPS indicator is clearly documented on page 12 of the Pilot's Guide.

"Purple (rarely seen) means you’re using an ADS-B receiver but the GPS feed is coming from a different source such as the iPad’s internal GPS or a separate GPS."

https://www.seattleavionics.com/documents/FlyQEFBGettingStarted.pdf

rgutlon, 2015-09-30:
> "Purple (rarely seen) means you’re using an ADS-B receiver but the GPS feed is coming from a different source such as the iPad’s internal GPS or a separate GPS."

Steve - thank you for pointing that out ... I along with others looked all over for an explanation.

cyoung, 2015-09-30:
@stevepodradchik - The GDL90 "Ownship" message is sent with the callsign 'Stratux' (capital S). That should uniquely identify the device as a stratux.

stevepodradchik, 2015-09-30:
Thanks Chris.  That should help. If you also have s/w version # or similar, we can process that, too.

BTW - typo on my part.  The GPS status colors are on page 15 not 12.

rgutlon, 2015-10-05:
Great news: Seattle Avionics mentioned the following in a recent mailing:

>**Flash!** New version 2.1.1 just given to Apple for their review. INcludes support for more ADS-B systems including **SkyGuard** and **Stratux** as well as a fix for a problem that some pilots have had with iOS 9 and the iPad's internal GPS.

bradanlane, 2015-10-05:
Can omeone can test when it's available to see if it works with the default stratux.conf network settings. If it still needs a special setting, I'd like to update the WebUI to make it easy for users to setup for FlyQ

rgutlon, 2015-10-05:
>Can omeone can test when it's available to see if it works with the default stratux.conf network settings. If it still needs a special setting, I'd like to update the WebUI to make it easy for users to setup for FlyQ.

Yes - happy to do that.

cyoung, 2015-10-05:
@bradanlane - I've got the pre-release that I tested with. It doesn't work with the default conf. Need to use the above one.

bradanlane, 2015-10-05:
Thanks. Unless there is objection, I'd like to add a mechanism via the WebUI for FlyQ users to update the network settings. Since it is a binary configuration, the WebUI would not expose the actual network settings. I'd only give a switch and help text then under the covered "do the the right thing".

stevepodradchik, 2015-10-06:
Just one quick thought … I’d change the default config.  Let me explain...

According to a 3rd party survey firm, Foreflight is the most popular app out there and FlyQ EFB is #2.  The only issue w/ the default config from FlyQ’s perspective is that the data was being sent on ports like the Clarity/FreeFlight system AND the one used for SkyRadar/iLevil/etc.  If only sent on one port, it works fine.

I doubt that any app that works w/ SkyRadar doesn’t also work w/ Clarity.  Therefore, I’d change the default config to simply output on the Clarity/FreeFlight port .. I doubt any app will fail to work w/ that config and, even if it does, you’ve optimized for the two most popular apps.

Thanks, Steve

cyoung, 2015-10-06:
Good point, we need to take a look at what works with port 43211 and not 4000. I think it was for WingX?

stevepodradchik, 2015-10-06:
WingX definitely supports the Clarity and I believe the FreeFlight RANGR.  It should work just fine on 4000.  No idea what major app would’t (other than Garmin which won’t work any any of these).

Thanks, Steve

rgutlon, 2015-10-06:
Like FF, WingX has worked out the door from the first release w/o any user modifications to the Stratux config. I believe a couple of the Android apps (including AvAre) required the port change with their helper app.

stevepodradchik, 2015-10-06:
WingX should work just fine w/ the change to sending only on the Clarity port.  So you’d get ForeFight, FlyQ, and WingX to work w/ the default config.  That covers all the major iPad apps that you can possibly talk to (again, no joy w/ Garmin).  There may be less popular apps that require the other port but my point is to optimize for the most users and allow manual (or WebUI) changes for the less popular ones.

Thanks, Steve

stevepodradchik, 2015-10-06:
Incidentally, we’re making changes to FlyQ EFB to support Stratux as a first-class device in the release we’ll make tomorrow to Apple.  That is, it’s in every way a peer to all the commercial systems (so the name ‘Stratux’ appears in the FlyQ EFB when it says which ADS-B is connected, etc.) so we’re doing some work specific to Stratux so it would be great to see it work perfectly out of the box w/ any user configuration.  I don’t know if any other app directly recognizes the Stratux.

Thanks Steve

On Oct 5, 2015, at 5:27 PM, Steve Podradchik <stevep@seattleavionics.com<mailto:stevep@seattleavionics.com>> wrote:

WingX should work just fine w/ the change to sending only on the Clarity port.  So you’d get ForeFight, FlyQ, and WingX to work w/ the default config.  That covers all the major iPad apps that you can possibly talk to (again, no joy w/ Garmin).  There may be less popular apps that require the other port but my point is to optimize for the most users and allow manual (or WebUI) changes for the less popular ones.

Thanks, Steve

cyoung, 2015-10-06:
Much appreciated, Steve. I found the app that it was added for (there was only one) and going to roll back port 43211. That should clean things up quite a bit. Great support from you guys - you really make things happen.

stevepodradchik, 2015-10-06:
We’re all pilots here, just like you.  We just want to make the best app and have everyone fly safely and economically.  Having a very inexpensive ADS-B is good for everyone.

You’ve done a remarkable job with this, Chris.  Do you have a day job in addition to this?

Thanks Steve

rgutlon, 2015-10-06:
v0.4r1 Compatibility Issues with FlyQ 2.1?

* Loaded v0.4r1
* Noticed /etc/stratux.conf was already in place given new UI
* FlyQ 2.1 has same behavior that prompted the need for the custom conf in v0.2r1
 * "ADS-B Connected" flashes on screen every few seconds
 * New behavior: FLYQ GPS indicator, which was reflecting "purple" (internal GPS) is now switching between purple and red each time the ADS-B Connected message loads.
* iPad iOS 9.0.1

Let me know what other information I can provide.

cyoung, 2015-10-06:
A permanent resolution to this is just a day or two out - use that custom config in the interim. Thanks!

Randman80, 2015-10-07:
OK, is it port 4000, or 43211 that Q is listening on for GPS data? Same port for the AHRS? Thanks for your patients....and all your hard work!

cyoung, 2015-10-07:
We're going to go with 4000 and eliminate 43211. The above config still works as a temporary fix for now, which is using 43211, but it doesn't really matter much for FlyQ - as long as it's just a single port.

stevepodradchik, 2015-10-07:
That sounds great.

Thanks for doing this!

You’ll see a final beta (actually release candidate) today for 2.1.1 that has cleaner display for Stratux.

Thanks, Steve

HiltonG, 2015-10-08:
Hi,

Any EFB manufacturer can make any claims about numbers, I'm not going to join in, suffice it to say that I just looked and WingX Pro7 is the #1 Top Grossing Navigation iPad app and the other EFB isn't in the top 10 places (this isn't unusual although it sometimes changes when new versions are released).

OK, ports...  WingX Pro7 was the first major EFB to add ADS-B weather and traffic - long before FF, FlyQ, AOPA's cancelled FlyQ, or Bendix King's cancelled myWingMan).  We used port 43211 because the one and only ADS-B receiver out then was SkyRadar and they used 43211.  Then because we were the only EFB app with ADS-B (and later the EFB app supporting the most ADS-B receivers) other manufacturers would ask us which port to use, we told them that just using 43211 would work fine so they did and they would 'just work'.  Sagetech was the 'outsider' and used port 4000 (so we added support for that).  By far, the most popular port for ADS-B receivers and EFB apps is port 43211.  In addition, WingX Pro7 for Android will be using port 43211.  We will be announcing a new product next week (watch our Facebook page), that too will use port 43211 for its ADS-B implementation.  Hilton Software LLC is the DoD contractor for mobile aviation on iOS, Android, and Windows (contrary to various claims by another company in aviation magazines).  Guess which port the DoD app will use for its ADS-B implementation...  Not sure why you would deviate from THE standard port based on one person's view.  I would recommend staying with your first choice, port 43211.

Moreover, we have (helped) define status messages for many ADS-B companies to ensure their proprietary GDL90 messages don't clash so if you ever want to throw in a status message, I'd be happy to help.  The status message can include useful info such as firmware and hardware versions, number of satellites, number of 978 and 1090 traffic targets, number of msgs received per second/minute, number of towers 'visible', and more.  Again, let me know.

BTW: I may keep up with this thread, but I probably won't post again here because I don't want this to turn into a back-n-forth about apps.  Bottom line: it's your software, your project, and your port choice.  Port 43211 is such the obvious choice (you got it right the first time); I'm curious why someone would want you to change that, hmmm...

Feel free to contact me directly if you'd like and congrats on your work - very impressive.

Hilton
hilton@hiltonsoftware.com

cyoung, 2015-10-08:
Hilton, thanks for the message. We were actually duplicating all messages on multiple ports. Ideally we could narrow it down to one port and "all" EFBs would be able to make use of that since it is more efficient overall. I'll send you an email.

stevepodradchik, 2015-10-08:
Folks,

The suggestion was simple math — FlyQ EFB, WingX, and ForeFlight all work fine off 4000.  WingX and FlyQ work fine on 43211 but ForeFlight does not.  Thus, using 4000 works for the majority of users who can possibly use Stratux as Garmin Pilot doesn’t use anything but their Bluetooth device.  Making the default setting use 4000 makes all the major apps happy. Using the SkyRadar port is only necessary for one app (I don’t know which it is but I get the feeling their market share is very low) and a Stratux user who needs that port can modify the config file to do it.  We’re just talking about out of the box experience as FlyQ EFB does currently have a problem if the Stratux transmits on both ports and correcting it is delicate work while changing a config file takes 1 minute.  Unless I’m missing something, WingX is not hurt in any way by simply using 4000.  More happy Stratux users with just using one port.  It’s just about optimizing the default config for the maximum # of users while retaining the flexibility for all users.

Hilton, completely unrelated to the ports was my comment that a neutral 3rd party survey group (not one we commissioned or even knew about until after the fact) recently did a survey of thousands of users and found ForeFlight to be the most commonly used app, followed (in order) by FlyQ, Garmin, and WingX (FlyQ was just ahead of Garmin, WingX had 1/3 the share of either Garmin or FlyQ).  I’d prefer if you didn’t imply we made “claims” (i.e. “lie”).  You do often have the highest grossing app in the Navigation category in the App Store (FlyQ EFB averages well within the top 10, btw).  Of course, that doesn’t take into account sales made outside of the app store and we both know that the daily difference between #1 and #20 grossing is usually a few hundred dollars per day and your higher price makes a difference there.  Moreover, since ForeFlight and Garmin are in the Weather category, not the Navigation category and therefore not part of the same list, you can’t really make any rational claim for being #1 about anything from that since ForeFlight and Garmin were not included in the comparison.  Your #1 claim is good marketing for you because most people don’t understand the rather critical distinction between different primary categories in the App Store.  In my opinion, it’s technically accurate but misleading.

Thanks, Steve

ssokol, 2015-10-08:
What if we default to the most common port and add an option to the UI that
allows the user to enable other ports by selecting their EFB(s) from a
menu? The "normal human" menu could allow you to pick the EFB apps to
support while the "advanced" menu could allow users to specify ports, or
event IP / port combinations (for statically configured devices).

stevepodradchik, 2015-10-08:
Obviously that would work but as just one EFB has a problem w/ 4000 and it’s (apparently) not a commonly-used one, seems you’re adding some additional setup complexity and confusion that 99.99% of the users don’t need.  I’m personally a bug fan of KISS (Keep It Simple Stupid) so I don’t like to ask users questions when they only apply to a very small # of people.  The list you’re talking about might also imply to users that EFBs not listed won’t work w/ Stratux — which may or may not be true but would cause additional user questions.

During Setup, I might suggest something like:

Stratux is pre-configured to work with virtually all the popular iPad and Android aviation apps (ForeFlight, FlyQ EFB, WingX, etc.).  If it doesn’t work with your app, click the Advanced button below to manually configure it.

<Advanced>

Thanks, Steve

jpoirier, 2015-10-08:
Just a friendly reminder to stay on topic.

rv8roester, 2015-10-11:
FWIW, I test flew my build (dual channel, no GPS) with the beta FlyQ 2.1.1 today for about an hour or so for the first time. It worked as advertised. I never saw any UAT traffic but 1090ES was depicted. Looks like it's working.

cyoung, 2015-10-11:
v0.4r2? Works out of the box? That's what we're looking for. Thanks for the report.

rv8roester, 2015-10-11:
Yep. I was a happy camper. I still need to verify the UAT 978 traffic
though. I wasn't seeing anything displayed, but the web UI was looking
good in real time during the flight and the ground station was showing the
appropriate updates.

Thanks for all of your work on this project! you're a prince.

David

Randman80, 2015-10-11:
Yep! I can  report that it is working , with no changes. I am running it with the gps  module, connected via the gpio interface. Traffic was being reported and presented, weather was updating. I disabled the gps in my iPad ( disabled location services for FlyQ ) and it switched to the adsb for the gps. So it works on iPads with or without built in gps. I can also report that it is working with iflight planner. Great job!

rgutlon, 2015-10-11:
>v0.4r2? Works out of the box? That's what we're looking for. Thanks for the report.

Yes - no need to touch the config file

navyav8er, 2021-12-22:
I know this thread is from 7 years ago, but it described an issue I was dealing with and allowed me to find a solution.  My issue was that I was losing GPS position and traffic every so often on my newer iPad, especially on longer flights (annoying but not critical).  My father's older iPad wouldn't even keep a ADSB supplied GPS position, it would switch to the internal iPAD GPS even though the ADSB showed a solid GPS position (the real reason I was trying to find a fix).  Here's what I have:
iPad running OS 12.5.5  FlyQ 6.0.2
iPad running OS 9.3.5  FlyQ 3.3.1
Merlin ADSB running Stratux v1.6r1-eu027-3d07b461-us

I'll admit I don't know how to write this code, so I was just blindly changing the code to see what worked.  Based on the thread above, I opened the "stratux.conf" file and edited it a text editor.  In that code it lists three different ports: 49002, 4000, 2000.  I deleted the two extra ports listed by removing everything between the {} and kept "Port":4000,"Capability":5

This seemed to fix the GPS stability issue with the older iPad and it seemed to keep the other ADSB data stuff too (Traffic, weather, etc).  I'm not sure if this will cause problems down the road, I'll test and report back if it doesn't work long term.  Feel free to respond if this is not a good idea.