cyoung/stratux #63: Stratux with GPS / AHRS fails after a while

ssokol, 2015-09-28:
Sorry for the utterly vague subject line on this. I've been trying to nail it down to a specific pattern but so far it seems to be just that  - vague. I'm running the image from v0.3b4 but with the latest code pulled from Github. System has either single or dual SDRs and an RY835AI. Tried on two different Pi units with a combination of three different SDR dongles. Only have one RY835AI, so that is common across all tests.

Here's the fact pattern as I've seen it: 

- Happens in as little as 5 minutes or as much as an hour
- The "FreeFlight" device disappears from ForeFlight
- The "Stratux" device is still showing as connected
- AHRS and baro altitude functions continue to work in FF*
- GPS and 1090 ES functions stop working ("No Fix" and all traffic disappears)
- The uptime counter stops in the admin UI
- I SSH in and check, and gen_gdl90 is still running
- free -mh shows no indication of a memory leak
- It continues to dump data to the stratus.log file including an uptime value
- Log memory values remain consistent at Alloc = 5 - 10mb, Sys = 17mb
- If I try to kill it either with Ctrl+C in the screen session or using kill -9, it becomes 'defunct'
- Unable to cleanly reboot - hangs and requires a hard reset
- If the thread (?) has not yet hung, I can cleanly kill the process and I can cleanly reboot

I've tried multiple power sources, all rated at 2.1 amps and above. I've changed power cables. The rig is not in a case, so it's not heat. Here's the kicker - when I unplug the RY835AI and turn off GPS / AHRS, it doesn't seem to happen. I'm running it overnight to see if that holds true for a really long run. If it fails, I'll update this issue with the log files.

Is anyone else seeing this on the current version of the software?

cyoung, 2015-09-28:
The defunct part is interesting.

Could you run

     strace -ff -p (proc number of gen_gdl90) 2>&1 >/tmp/somelog

Then try to kill it to reproduce the defunct state? First thought is issue with the i2c code.

ssokol, 2015-09-28:
Will do. FWIW - just checked and *without* the RY835AI the Stratux has been up and running for 9 hours.

ssokol, 2015-09-28:
Probably a side-note / separate issue but... when I re-connected the RY835AI and powered it back up, it had been off for 9+ hours and took perhaps a minute to acquire a fix. The Stratux never received GPS data. I tried toggling the GPS function off and back on - no joy. I tried a soft reboot - still showed 0 satellites. Finally did a hard reboot and after it showed 11 sats and a 3 meter accuracy in ForeFlight.

ssokol, 2015-09-28:
Ok, caught it. Here's the last (and hopefully most relevant) part of the strace log file.

````
[pid  2252] <... poll resumed> )        = 1 ([{fd=23, revents=POLLOUT}])
[pid  2258] clock_gettime(CLOCK_REALTIME,  <unfinished ...>
[pid  2252] ioctl(23, USBDEVFS_REAPURBNDELAY or USBDEVFS_REAPURBNDELAY32 <unfinished ...>
[pid  2258] <... clock_gettime resumed> {1443452697, 339047755}) = 0
[pid  2252] <... ioctl resumed> , 0x6b7fec64) = 0
[pid  2258] clock_gettime(CLOCK_REALTIME,  <unfinished ...>
[pid  2252] timerfd_settime(22, 0, {it_interval={0, 0}, it_value={0, 0}}, NULL <unfinished ...>
[pid  2258] <... clock_gettime resumed> {1443452697, 339954780}) = 0
[pid  2252] <... timerfd_settime resumed> ) = 0
[pid  2258] futex(0x10c3eba0, FUTEX_WAKE, 1 <unfinished ...>
[pid  2252] futex(0x10c3f960, FUTEX_WAIT, 0, NULL <unfinished ...>
[pid  2258] <... futex resumed> )       = 1
[pid  2164] <... futex resumed> )       = 0
[pid  2258] clock_gettime(CLOCK_MONOTONIC,  <unfinished ...>
[pid  2130] ioctl(26, 0x707 <unfinished ...>
[pid  2258] <... clock_gettime resumed> {13680, 847198988}) = 0
[pid  2164] futex(0x10c3f960, FUTEX_WAKE, 1 <unfinished ...>
[pid  2258] futex(0x504214, FUTEX_WAIT, 0, {0, 51296} <unfinished ...>
[pid  2252] <... futex resumed> )       = 0
[pid  2258] <... futex resumed> )       = -1 ETIMEDOUT (Connection timed out)
[pid  2164] <... futex resumed> )       = 1
[pid  2252] clock_gettime(CLOCK_MONOTONIC,  <unfinished ...>
[pid  2258] clock_gettime(CLOCK_MONOTONIC,  <unfinished ...>
[pid  2252] <... clock_gettime resumed> {13680, 848749394}) = 0
[pid  2258] <... clock_gettime resumed> {13680, 848840747}) = 0
[pid  2252] epoll_wait(0xa, 0x6b7fe564, 0x80, 0 <unfinished ...>
[pid  2258] futex(0x107e93e0, FUTEX_WAIT, 0, NULL <unfinished ...>
[pid  2252] <... epoll_wait resumed> )  = 0
[pid  2164] futex(0x10c3eba0, FUTEX_WAIT, 0, NULL <unfinished ...>
[pid  2252] clock_gettime(CLOCK_MONOTONIC,  <unfinished ...>
[pid  2130] <... ioctl resumed> , 0x1071fe80) = 2
[pid  2252] <... clock_gettime resumed> {13680, 850154593}) = 0
[pid  2130] sched_yield( <unfinished ...>
[pid  2252] futex(0x107e93e0, FUTEX_WAKE, 1 <unfinished ...>
[pid  2130] <... sched_yield resumed> ) = 0
[pid  2258] <... futex resumed> )       = 0
[pid  2256] <... read resumed> "80915,,,", 2022) = 8
[pid  2258] futex(0x10c3eba0, FUTEX_WAKE, 1 <unfinished ...>
[pid  2256] futex(0x107e8e60, FUTEX_WAIT, 0, NULL <unfinished ...>
[pid  2258] <... futex resumed> )       = 1
[pid  2252] <... futex resumed> )       = 1
[pid  2258] ioctl(23, USBDEVFS_SUBMITURB or USBDEVFS_SUBMITURB32 <unfinished ...>
[pid  2252] clock_gettime(CLOCK_MONOTONIC,  <unfinished ...>
[pid  2258] <... ioctl resumed> , 0x76a7d1e0) = 0
[pid  2252] <... clock_gettime resumed> {13680, 852610253}) = 0
[pid  2258] ioctl(23, USBDEVFS_SUBMITURB or USBDEVFS_SUBMITURB32 <unfinished ...>
[pid  2164] <... futex resumed> )       = 0
[pid  2130] futex(0x504570, FUTEX_WAKE, 1 <unfinished ...>
[pid  2164] clock_gettime(CLOCK_MONOTONIC,  <unfinished ...>
[pid  2129] <... mmap2 resumed> )       = 0x76f59000
[pid  2164] <... clock_gettime resumed> {13680, 854193316}) = 0
[pid  2129] read(33,  <unfinished ...>
[pid  2164] clock_gettime(CLOCK_REALTIME,  <unfinished ...>
[pid  2129] <... read resumed> "480\n", 4096) = 4
[pid  2252] clock_gettime(CLOCK_MONOTONIC,  <unfinished ...>
[pid  2258] <... ioctl resumed> , 0x76a7d20c) = 0
[pid  2252] <... clock_gettime resumed> {13680, 855954137}) = 0
[pid  2258] ioctl(23, USBDEVFS_SUBMITURB or USBDEVFS_SUBMITURB32 <unfinished ...>
[pid  2252] futex(0x107e8e60, FUTEX_WAKE, 1 <unfinished ...>
[pid  2258] <... ioctl resumed> , 0x76a7d238) = 0
[pid  2256] <... futex resumed> )       = 0
[pid  2258] ioctl(23, USBDEVFS_SUBMITURB or USBDEVFS_SUBMITURB32 <unfinished ...>
[pid  2256] read(27,  <unfinished ...>
[pid  2258] <... ioctl resumed> , 0x76a7d264) = 0
[pid  2252] <... futex resumed> )       = 1
[pid  2258] ioctl(23, USBDEVFS_SUBMITURB or USBDEVFS_SUBMITURB32 <unfinished ...>
[pid  2252] clock_gettime(CLOCK_MONOTONIC,  <unfinished ...>
[... 933 lines cut ...]
[pid  2039] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 2039 leader 1887
[pid  2081] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 2081 leader 1887
[pid  2045] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 2045 leader 1887
[pid  2017] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 2017 leader 1887
[pid  2015] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 2015 leader 1887
[pid  1928] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 1928 leader 1887
[pid  1924] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 1924 leader 1887
[pid  2256] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 2256 leader 1887
[pid  2164] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 2164 leader 1887
[pid  2294] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 2294 leader 1887
[pid  2291] +++ killed by SIGKILL +++
PANIC: handle_group_exit: 2291 leader 1887
````

Apologies for not simply attaching the full file, but somehow it did not write to /tmp correctly - not sure why. I copied this from the terminal buffer.

egid, 2015-09-30:
I just got the GPS wired in over GPIO, and I'm updating to .3b4. b3 was stable for very long periods (2.5-3h) in flight. Once I get it on b4, I'll leave it up and running in the house and see how long it goes.

egid, 2015-09-30:
Ok, I can repro. I've had it up for several hours now, running 0.3b4. WebUI shows 00h17m15s uptime, but correctly loads & connects. GPS satellite count doesn't change, although I know I have pretty poor coverage where the unit is. I seem to be able to set settings - I just toggled 978 off and force-quit the browser, and it stayed off.

Would any logs in particular be helpful debugging?

ssokol, 2015-09-30:
You might try grabbing a copy of the system call trace using strace command Chris included above:

    strace -ff -p (proc number of gen_gdl90) 2>&1 >/tmp/somelog

"Then try to kill it to reproduce the defunct state? First thought is issue with the i2c code."

Beware - it spews a lot at you, and for some reason I wound up with an empty file. I had to capture the dump you see above from the terminal buffer on my Mac. Probably easiest to do with two windows open. Capture the trace in one and run the kill command in the other.

ssokol, 2015-10-01:
Possible clue: the bug occurred for me with AHRS and GPS enabled, but with the RY835AI disconnected. I turned off the AHRS and GPS buttons, power cycled and it's been stable for several hours.

egid, 2015-10-01:
I'll try to ssh in once the battery has recharged. 

> the bug occurred for me with AHRS and GPS enabled, but with the RY835AI disconnected

Not for me - connected _and_ enabled. I got about 3 hours out of the second attempt, with 978 on and picking up a few random UAT flights overhead.

vontresc, 2015-10-06:
I am still seeing the same issue with the new 0.4 release. Are there any logs I can provide that would help out in troubleshooting this?

cyoung, 2015-10-06:
On v0.4r1 enable replay logs. Is everyone here that's having issues also have GPS AND AHRS settings enabled? If not, what are your settings?

ssokol, 2015-10-07:
I've tried it with only GPS enabled and get the same symptoms. I've also
tried it with GPS and AHRS enabled but without the RY835AI connected. Same.

vontresc, 2015-10-08:
Quick question. In checking the logs for this bug, I noticed that the system time was set to September 27/28. Does Stratux update the system clock based on the GPS date when connected via serial io?

bradanlane, 2015-10-09:
OK, it sounds like it would be helpful for the WebUI to add the first phase of GPS and AHRS support.

I have the RY835AI board (sitting in its little Mylar bag). Where can I find the wiring and setup?

vontresc, 2015-10-09:
Brandan, Check this reddit post. https://www.reddit.com/r/stratux/comments/3mf8xg/ry835ai_gps_connections_nonusb/

bradanlane, 2015-10-09:
_"bradan" is just my login :-)_

Is there any config for the needed for the RY835AI prior to powering up Stratux? (I recall something about 10hz ?)

vontresc, 2015-10-09:
nah Just hooked everything up, and it worked

cyoung, 2015-10-19:
Reproduced on:

     root@raspberrypi:~/stratux# vcgencmd version
     Apr 21 2015 14:42:19 
     Copyright (c) 2012 Broadcom
     version 2d5ad04b63af4233440c3f7c8587108223201102 (clean) (release)
     root@raspberrypi:~/stratux# uname -a
     Linux raspberrypi 3.18.11-v7+ #781 SMP PREEMPT Tue Apr 21 18:07:59 BST 2015 armv7l GNU/Linux
     root@raspberrypi:~/stratux#

cyoung, 2015-10-19:
@ssokol @egid - Think I was able to find the source of the problem (RPi temperature probe, seems to only have issues when other serial devices like RY835AI is hooked up and being used).

If you could try reproducing this with the latest code that would help. The problem will still occur, but the failure mode should be different. Everything should work normally, except for in the web interface you will see the "CPU Temp" hang at 0C. Also the process will go defunct if you try to kill it, but should function normally otherwise.

cyoung, 2015-10-19:
Stable on
     root@raspberrypi:~/stratux# vcgencmd version
     Sep 23 2015 12:12:01 
     Copyright (c) 2012 Broadcom
     version c156d00b148c30a3ba28ec376c9c01e95a77d6d5 (clean) (release)
     root@raspberrypi:~/stratux# uname -a
     Linux raspberrypi 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux
     root@raspberrypi:~/stratux# 

(do apt-get update and apt-get upgrade, then reboot).

Could you guys give it a try with the old kernel/RPi firmware then with the new and confirm it's fixed also?

ssokol, 2015-10-19:
Sure. Will ground test today / tonight and hopefully flight test tomorrow.

egid, 2015-10-19:
I'll have access in a few days and I'll give it a shot then.

JohnOCFII, 2015-10-19:
> (do apt-get update and apt-get upgrade, then reboot).
> 
> Could you guys give it a try with the old kernel/RPi firmware then with the new and confirm it's fixed also?

I can test (ground) tonight.  I'll run on old kernel until I go to sleep, then update.

JohnOCFII, 2015-10-20:
I tested for 2 hours on the old kernel (after a git pull and make install, plus manually copying over the new web files).  All seemed stable.

I just updated and upgraded to 4.1.7-v7+.  Running fine to the WebUI and to ForeFlight on my iPad for 10 minutes.  I'll let Stratux run over night.

John

JohnOCFII, 2015-10-20:
I ran the current Stratux code (as of 7 pm CDT last night) on the upgraded pi (4.1.7-v7+) for about 9 hours.  It all seemed to be running fine in the morning - I was still receiving valid traffic targets via 1090ES.  I can't receive 978 at my house, so I have nothing to report in that area.  The WebUI seemed to be working fine, and the clock/uptime display in the WebUI continued to increment properly.

cyoung, 2015-10-20:
Nice, going to call it fixed. Re-open if any issue.

closed 2015-10-20 by cyoung