cyoung/stratux #64: Implement GPS accuracy check

AvSquirrel, 2015-09-28:
**Problem statement:**

GPS position accuracy is hard coded as NACp = 11 (0xb) in gen_gdl90.makeOwnshipReport() . ForeFlight thererfore reports accuracy as 3m any time a valid ownship GPS position is received, regardless of actual accuracy.

**Solution**

- Update ry835ai.go to extract DGPS status from the GGA message, then calculate and output an NACp value based on HDOP, type of nav solution, and possibly number of satellites.
- Update ownship msg[13] to use variable NIC = 0, and the variable NACp value from updated ry835ai.go
- Correlation between NACp, HDOP, type of navigation solution, and number of satellites received is TBD

&nbsp;

**Additional comments**
ForeFlight appears to be using accuracy estimates based on the ownship NACp value. Any value less than 8 will show as "Poor". Values of 8 or greater will show a position estimate of (IIRC) 100m, 30m, 10m, or 3m per the table below.

[image]

Getting a decent Estimated Position Uncertainty (EPU) may be a challenge. EPU is ["defined as the radius of a circle centered at the true horizontal position within which the measured horizontal position lies with 95% probability"](https://www.mitre.org/sites/default/files/pdf/10_3026.pdf). In other words, the 2σ horizontal position variation. This can be calculated readily for a GPS receiver in a fixed location.

The NMEA messages include HDOP (horizontal dilution of position), which indicates the uncertainty of the measurement due to the geometry of the satellites used for the position solution. However, this does not account of other error sources (multipath, clock error, atmospheric effect, etc.)

I think that combining the presence / absence of DGPS (i.e. WAAS) along with HDOP may get us close. I'm running a series of long (i.e. 8+ hour) evaluations with the GPS module located in one position to calculate total position variation for multiple conditions (5, 8, 12 satellites with / without WAAS, and with / without GLONASS) to get a baseline. I'll update this issue with my test results as they complete.

I also think we should report NIC = 0. This is an intergrity / containment measurement of "trust" in the GPS system. It's relevant for traffic separation, but not for evaluating the performance of the raw position data. ForeFlight doesn't seem to use it, so it would make sense to drop it.

224XS, 2015-09-28:
I think I don't understand your concern with reported GPS accuracy in the context or Stratux, 
but I would like to.
The accuracy parameters NIC, NAC, and HDOP are really only relevant to the GPS data if it 
were to be fed back into the National Airspace System, i.e., ADS-B Out.  Are you planning
on building an open source extended squitter or UAT?

JohnOCFII, 2015-09-29:
@224XS One reason to try to improve reported accuracy to is to increase confidence in the user when seeing the position plotted on a map in ForeFlight while on an instrument approach, for example.  Another reason is feature parity with how the data is displayed in ForeFlight when using a Stratus 1 or Stratus 2 unit.  The Stratus units will regularly change the displayed accuracy varying between 1M and 30M or greater than 30M.  It helps me have a better understanding for the quality of the data.

bradanlane, 2015-09-29:
As an end user, I regularly look for the "GPS accuracy" as a means of knowing if my navigation device is receiving enough satellites. It's "front and center" in ForeFlight and is easy to reference.

224XS, 2015-09-29:
I guess I would just urge a wee bit of caution.  The GPS card we are talking about is clearly more than a toy, but it isn't exactly an approach certificated WAAS GPS.

AvSquirrel, 2015-09-29:
Prototype is here (https://github.com/AvSquirrel/stratux/commit/e0ba1a35a6da46d76cff2cb3d8a7379274589813) and on three prior commits to my fork. (Bear with me as I learn to use git.)

Changes include:
- Determine if WAAS is being used in nav solution
- Multiply HDOP by 4.0m (WAAS) or 8.0 m (non-WAAS) to estimate 95% confidence of position (likely to be refined after additional testing)
- Determine appropriate NACp category based on the position estimate
- Pass NIC = 8 (was NIC = 11 in 0.3b4) and calculated NACp in the Ownship message
- Add above data to stratux.log

@224XS - appreciate the concern for caution, but there's nothing here that isn't being done by any other internal or external GPS. As @JohnOCFII and @bradanlane note, an estimate of GPS accuracy is one of the default "gauges" shown on the bottom of the ForeFlight screen. Having an indication of what the external GPS is seeing a useful tool, and the current betas sort of lie to us by telling us that we always have 3m accuracy. This change would put Stratux in line with the expected behavior.

One thing worth noting is that dropping NIC to 0 will prevent ForeFlight from detecting Stratux data as ownship, but this will also disable the pressure altitude that is sent with the ownship message. Keeping a valid value for NIC will allow ForeFlight to intepret pressure altitude data, and give a better vertical reference point than internal GPS for detecting relative altitude of other aircraft.

cyoung, 2015-09-30:
@AvSquirrel - great idea tagging "Stratux" on the ownship in the case that it gets displayed by the app.

Your changes integrated into master: https://github.com/cyoung/stratux/commit/aafcd4892c3bf9fb303582369180e94392cc1531

cyoung, 2015-09-30:
Good enough for now, re-open if issues with the new code.

closed 2015-09-30 by cyoung