cyoung/stratux #66: no AHRS to wingxpro with RY835

peeve, 2015-09-29:
I get gps via pin or USB, I get AHRS in foreflight,
but in wingxpro I can see the embedded AHRS status is green but on the synthetic vision page I get NO AHRS and no input.
I have tried changing the port in the stratux.conf per the flyq directions and that made no difference

ScraboTower, 2015-10-18:
Has anyone reached out to Hilton about this issue on WingX ?  This is something I have been waiting for since i got the chip

peeve, 2015-10-19:
the only complaint I have about wingx is complete lack of tech support. Even if I COULD get someone to respond to a message I just don't know enough to even begin the discussion personally or I'd take it on.

cyoung, 2015-10-20:
I'm getting an "Embedded AHRS" indication on WingX. Tried X-Plane simulator protocol also. When showing the attitude indicator, neither option shows that AHRS data is being received. @HiltonG, any ideas?

[image]

ScraboTower, 2015-10-20:
I Have pinged Hilton over in the Aopa forum to see if that can get involved in resolving this

peeve, 2015-10-21:
thanks guys!
I get exactly the same indications.

HiltonG, 2015-10-21:
peeve: email us at support@hiltonsoftware.com - unfortunately we get a LOT of Junk and sometimes good emails go to Junk.  We do go through the Junk folder but granted we probably miss a few here and there.  We have a support phone number number (see top-right right on our home page, I checked with Support yesterday and all voicemails had been called back and everything was up-to-date.).

I try really hard to reply to the forums, I just cannot go to each and every one every day, so if an issue comes up, please email and I'll try respond on that forum.

I did look into the AHRS issue a week or two ago and IIRC and something in the datastream was wrong - I seem to remember it was a msg that should contain AHRS, but I didn't even have an AHRS connected, so that green indication is wrong because it appears that the datastream contains bad data.  I am happy to contribute to the discussion of fixing, improving, and correcting the datastream, ports, msg formats etc, but, and please correct me if I'm wrong here, all efforts seem to be to get this working with FF instead of working on the 'correct' msgs etc - so I decided to take a backseat and let is play out.  I'd love to help if the help is wanted.  I have offered to help with the msgs (I have helped/coordinated with the design of msgs (status, AHRS etc) with many companies in the industry) - I do it for free, never asked any companies once for recognition or pay of any kind.  Perhaps I misread the email replies, but with this big push at 'just get it working with FF' any other input just didn't seem that important.

BTW: I was at FLYING Expo a few days ago and was being interviewed by the Media.  We were talking about WingX Pro7 for iOS, Android, and (new) Fire and I gave Startux a good 'plug' - showed all the pieces, mentioned the cost etc and of course stressed that it wasn't my work that it was another pilot's efforts.  Now, I don't know how much of what I said will be kept and what will be thrown away, but watch for that.

I'm waiting for someone to put together a company and make these things with a $50? premium for their efforts; e.g. $199-249 (or higher) for a single/dual channel plus GPS/AHRS nicely put together in a box, soldered, with the short antennae.  I'd buy 5 of those right now.

Hilton

peeve, 2015-10-21:
a guy is already working on exactly what you said.
https://www.reddit.com/r/stratux/comments/3pbhdk/a_stratux_kickstarter_campaign/
a lot of people seem to think it's a little shady, I hope he does ok with it.

As a side note, I do wish there was (or if there is I could figure it out) a full screen chart mode for us ipad mini users, drop all the menus and headers and just display the chart. Maybe a double tap to exit kind of deal.

rwilmoth, 2015-10-21:
This is on my list, as well... I'm Sales/Marketing for a major 
technical/Open Source company. Hilton - I'll be in touch with you 
regarding some advice on the business side.

skypuppy, 2015-10-21:
> <mailto:support@hiltonsoftware.com> - unfortunately we get a LOT of

That is so cool that you would take the time to respond, Mr. (?) 
Hilton.  Thank you.

skypuppy, 2015-10-21:
I haven't looked up the original author's licensing/release details but 
some open source software is freely released to the world to do with as 
they please, even make a profit.  Personally, I would REALLY be 
disinclined to put any further effort into a project that others 
marketed as their own.  (This actually happened to me once and when I 
stopped supporting it, that group fell flat on their face.)
Other licenses allow the free software to be used anywhere for any 
purpose as long as it is not for profit.

In the kickstarter guys' defense, several people on reddit have 
practically begged for a turnkey solution in this price range and I 
imagine he is simply responding to that.  I considered it myself. :)  On 
the flip side of that, it seems unethical to market someone else's code 
AND design for profit.

There could be a lot of profit in it as 2020 approaches, just in the US 
alone, much less worldwide markets.  If I did it, I'd *have* to at least 
get the major software and design contributors on board and figure out 
something equitable with them.  But that's just me.  I would also modify 
the physical packaging to make it more practical for the non-tinkerer.

Skypuppy7

rwilmoth, 2015-10-21:
Here's the major difference - the company i work with pays the 
developers that develop the code. We encourage that it continues to be 
freely released. The only thing that we make money off of is putting it 
into a package and ensuring that there is support, incase people need it.

Also, there are several people around me that are asking for it, and if 
the guys from Hilton are asking for 5, then there is a market for it. I 
wouldn't be selling the code, as with most open source licenses, you 
can't exactly sell something that isn't yours. I would, however, sell 
the hardware platform as a pre-assembled *thing* for those who are 
interested. I think most of the money in this is more along the lines of 
the 3d-printed case, which isn't readily available to be purchased.

> skypuppy <mailto:notifications@github.com>
> October 21, 2015 at 12:15 PM
> I haven't looked up the original author's licensing/release details but
> some open source software is freely released to the world to do with as
> they please, even make a profit. Personally, I would REALLY be
> disinclined to put any further effort into a project that others
> marketed as their own. (This actually happened to me once and when I
> stopped supporting it, that group fell flat on their face.)
> Other licenses allow the free software to be used anywhere for any
> purpose as long as it is not for profit.
>
> In the kickstarter guys' defense, several people on reddit have
> practically begged for a turnkey solution in this price range and I
> imagine he is simply responding to that. I considered it myself. :) On
> the flip side of that, it seems unethical to market someone else's code
> AND design for profit.
>
> There could be a lot of profit in it as 2020 approaches, just in the US
> alone, much less worldwide markets. If I did it, I'd *have* to at least
> get the major software and design contributors on board and figure out
> something equitable with them. But that's just me. I would also modify
> the physical packaging to make it more practical for the non-tinkerer.
>
> Skypuppy7
>
>
>
> >
> > a guy is already working on exactly what you said.
> > 
> https://www.reddit.com/r/stratux/comments/3pbhdk/a_stratux_kickstarter_campaign/
> > a lot of people seem to think it's a little shady, I hope he does ok
> > with it.
> >
> > As a side note, I do wish there was (or if there is I could figure it
> > out) a full screen chart mode for us ipad mini users, drop all the
> > menus and headers and just display the chart. Maybe a double tap to
> > exit kind of deal.
> >
> >
>
>
> peeve <mailto:notifications@github.com>
> September 29, 2015 at 9:27 AM
>
> I get gps via pin or USB, I get AHRS in foreflight,
> but in wingxpro I can see the embedded AHRS status is green but on the 
> synthetic vision page I get NO AHRS and no input.
> I have tried changing the port in the stratux.conf per the flyq 
> directions and that made no difference

kdknigga, 2015-10-21:
rwilmoth,

Your employer really isn't a hardware company, but it would be really cool
to see the likes of them take an interest in this.

rwilmoth, 2015-10-21:
Correct, and the fact that this runs on small IOT style ARM makes them 
look a bit the other way.

I'm talking to some folks in the Fedora project to see if they have any 
interest, and I've done my best to evangelize this project wherever I 
go. I'm about 90% travel for a living and carry my device around with me 
showing whatever the closest flight school/FBO to my customer location 
what it is, what it does, and helping garner additional interest if 
possible.

> Kris Knigga <mailto:notifications@github.com>
> October 21, 2015 at 12:46 PM
> rwilmoth,
>
> Your employer really isn't a hardware company, but it would be really cool
> to see the likes of them take an interest in this.
>
> > Here's the major difference - the company i work with pays the
> > developers that develop the code. We encourage that it continues to be
> > freely released. The only thing that we make money off of is putting it
> > into a package and ensuring that there is support, incase people 
> need it.
> >
> > Also, there are several people around me that are asking for it, and if
> > the guys from Hilton are asking for 5, then there is a market for it. I
> > wouldn't be selling the code, as with most open source licenses, you
> > can't exactly sell something that isn't yours. I would, however, sell
> > the hardware platform as a pre-assembled *thing* for those who are
> > interested. I think most of the money in this is more along the lines of
> > the 3d-printed case, which isn't readily available to be purchased.
> >
> >
> > > skypuppy <mailto:notifications@github.com>
> > > October 21, 2015 at 12:15 PM
> > > I haven't looked up the original author's licensing/release 
> details but
> > > some open source software is freely released to the world to do 
> with as
> > > they please, even make a profit. Personally, I would REALLY be
> > > disinclined to put any further effort into a project that others
> > > marketed as their own. (This actually happened to me once and when I
> > > stopped supporting it, that group fell flat on their face.)
> > > Other licenses allow the free software to be used anywhere for any
> > > purpose as long as it is not for profit.
> > >
> > > In the kickstarter guys' defense, several people on reddit have
> > > practically begged for a turnkey solution in this price range and I
> > > imagine he is simply responding to that. I considered it myself. :) On
> > > the flip side of that, it seems unethical to market someone else's 
> code
> > > AND design for profit.
> > >
> > > There could be a lot of profit in it as 2020 approaches, just in 
> the US
> > > alone, much less worldwide markets. If I did it, I'd *have* to at 
> least
> > > get the major software and design contributors on board and figure out
> > > something equitable with them. But that's just me. I would also modify
> > > the physical packaging to make it more practical for the non-tinkerer.
> > >
> > > Skypuppy7
> > >
> > >
> > >
> > > >
> > > > a guy is already working on exactly what you said.
> > > >
> > >
> > 
> https://www.reddit.com/r/stratux/comments/3pbhdk/a_stratux_kickstarter_campaign/
> > > > a lot of people seem to think it's a little shady, I hope he does ok
> > > > with it.
> > > >
> > > > As a side note, I do wish there was (or if there is I could 
> figure it
> > > > out) a full screen chart mode for us ipad mini users, drop all the
> > > > menus and headers and just display the chart. Maybe a double tap to
> > > > exit kind of deal.
> > > >
> > > > 
> > > >
> > >
> > >
> > > peeve <mailto:notifications@github.com>
> > > September 29, 2015 at 9:27 AM
> > >
> > > I get gps via pin or USB, I get AHRS in foreflight,
> > > but in wingxpro I can see the embedded AHRS status is green but on the
> > > synthetic vision page I get NO AHRS and no input.
> > > I have tried changing the port in the stratux.conf per the flyq
> > > directions and that made no difference
> > >
> > >
> >
> >
> >
> > Rob Wilmoth, RHCDS
> > Solutions Architect
> > Red Hat, Inc
> > 919-805-1393
> >
> >
>
>
> peeve <mailto:notifications@github.com>
> September 29, 2015 at 9:27 AM
>
> I get gps via pin or USB, I get AHRS in foreflight,
> but in wingxpro I can see the embedded AHRS status is green but on the 
> synthetic vision page I get NO AHRS and no input.
> I have tried changing the port in the stratux.conf per the flyq 
> directions and that made no difference

kdknigga, 2015-10-21:
Pidora doesn't seem to get much love, which is sad.

rwilmoth, 2015-10-21:
Agreed....

> Kris Knigga <mailto:notifications@github.com>
> October 21, 2015 at 12:54 PM
> Pidora doesn't seem to get much love, which is sad.
>
> > Correct, and the fact that this runs on small IOT style ARM makes them
> > look a bit the other way.
> >
> > I'm talking to some folks in the Fedora project to see if they have any
> > interest, and I've done my best to evangelize this project wherever I
> > go. I'm about 90% travel for a living and carry my device around with me
> > showing whatever the closest flight school/FBO to my customer location
> > what it is, what it does, and helping garner additional interest if
> > possible.
> >
> >
> >
> > > Kris Knigga <mailto:notifications@github.com>
> > > October 21, 2015 at 12:46 PM
> > > rwilmoth,
> > >
> > > Your employer really isn't a hardware company, but it would be really
> > cool
> > > to see the likes of them take an interest in this.
> > >
> > > > Here's the major difference - the company i work with pays the
> > > > developers that develop the code. We encourage that it continues 
> to be
> > > > freely released. The only thing that we make money off of is 
> putting it
> > > > into a package and ensuring that there is support, incase people
> > > need it.
> > > >
> > > > Also, there are several people around me that are asking for it, 
> and if
> > > > the guys from Hilton are asking for 5, then there is a market 
> for it. I
> > > > wouldn't be selling the code, as with most open source licenses, you
> > > > can't exactly sell something that isn't yours. I would, however, 
> sell
> > > > the hardware platform as a pre-assembled *thing* for those who are
> > > > interested. I think most of the money in this is more along the 
> lines
> > of
> > > > the 3d-printed case, which isn't readily available to be purchased.
> > > >
> > > >
> > > > > skypuppy <mailto:notifications@github.com>
> > > > > October 21, 2015 at 12:15 PM
> > > > > I haven't looked up the original author's licensing/release
> > > details but
> > > > > some open source software is freely released to the world to do
> > > with as
> > > > > they please, even make a profit. Personally, I would REALLY be
> > > > > disinclined to put any further effort into a project that others
> > > > > marketed as their own. (This actually happened to me once and 
> when I
> > > > > stopped supporting it, that group fell flat on their face.)
> > > > > Other licenses allow the free software to be used anywhere for any
> > > > > purpose as long as it is not for profit.
> > > > >
> > > > > In the kickstarter guys' defense, several people on reddit have
> > > > > practically begged for a turnkey solution in this price range 
> and I
> > > > > imagine he is simply responding to that. I considered it 
> myself. :)
> > On
> > > > > the flip side of that, it seems unethical to market someone else's
> > > code
> > > > > AND design for profit.
> > > > >
> > > > > There could be a lot of profit in it as 2020 approaches, just in
> > > the US
> > > > > alone, much less worldwide markets. If I did it, I'd *have* to at
> > > least
> > > > > get the major software and design contributors on board and figure
> > out
> > > > > something equitable with them. But that's just me. I would also
> > modify
> > > > > the physical packaging to make it more practical for the
> > non-tinkerer.
> > > > >
> > > > > Skypuppy7
> > > > >
> > > > >
> > > > >
> > > > > >
> > > > > > a guy is already working on exactly what you said.
> > > > > >
> > > > >
> > > >
> > >
> > 
> https://www.reddit.com/r/stratux/comments/3pbhdk/a_stratux_kickstarter_campaign/
> > > > > > a lot of people seem to think it's a little shady, I hope he 
> does
> > ok
> > > > > > with it.
> > > > > >
> > > > > > As a side note, I do wish there was (or if there is I could
> > > figure it
> > > > > > out) a full screen chart mode for us ipad mini users, drop 
> all the
> > > > > > menus and headers and just display the chart. Maybe a double 
> tap to
> > > > > > exit kind of deal.
> > > > > >
> > > > > >
> > > > > >
> > > > >
> > > > > 
> <https://github.com/cyoung/stratux/issues/66#issuecomment-149949079
> > > > >
> > > > > peeve <mailto:notifications@github.com>
> > > > > September 29, 2015 at 9:27 AM
> > > > >
> > > > > I get gps via pin or USB, I get AHRS in foreflight,
> > > > > but in wingxpro I can see the embedded AHRS status is green but on
> > the
> > > > > synthetic vision page I get NO AHRS and no input.
> > > > > I have tried changing the port in the stratux.conf per the flyq
> > > > > directions and that made no difference
> > > > >
> > > > >
> > > >
> > > >
> > > >
> > > > Rob Wilmoth, RHCDS
> > > > Solutions Architect
> > > > Red Hat, Inc
> > > > 919-805-1393
> > > >
> > > > 
> > > >
> > >
> > >
> > > peeve <mailto:notifications@github.com>
> > > September 29, 2015 at 9:27 AM
> > >
> > > I get gps via pin or USB, I get AHRS in foreflight,
> > > but in wingxpro I can see the embedded AHRS status is green but on the
> > > synthetic vision page I get NO AHRS and no input.
> > > I have tried changing the port in the stratux.conf per the flyq
> > > directions and that made no difference
> > >
> > >
> >
> >
> >
> > Rob Wilmoth, RHCDS
> > Solutions Architect
> > Red Hat, Inc
> > 919-805-1393
> >
> >
>
>
> peeve <mailto:notifications@github.com>
> September 29, 2015 at 9:27 AM
>
> I get gps via pin or USB, I get AHRS in foreflight,
> but in wingxpro I can see the embedded AHRS status is green but on the 
> synthetic vision page I get NO AHRS and no input.
> I have tried changing the port in the stratux.conf per the flyq 
> directions and that made no difference

dmurray14, 2015-10-21:
I kind of went off on that guy a bit in the kickstarter thread, but I stand
firm that it's not right for someone else to make a buck on this before the
authors do. A lot of you guys have put tons of hard work into this and even
if you are too gracious to admit it, you are the ones that should see the
dollar signs far before anyone else. I don't have anywhere near the skill
needed to put together what you guys have done, and frankly none of us
deserve to get this for free, despite you guys all giving it to us. There
is tremendous value in this, and YOU are the ones that have created value,
therefore YOU are the ones that deserve to determine how and when a
commercial offering is presented.

The beauty of open source projects is how quickly they snowball into mature
projects - and I've seen that with this project. You all have way more
skill and selflessness than you do a desire to earn money, but soon enough
(perhaps already) someone will come by with the opposite ingredients and
look to make a quick buck off your hard work, and that's not fair.

I wholly support you guys putting this under a license that keeps with the
spirit of open source development, but restricts the ability for
non-contributors to come along and sell your hard work. This doesn't mean
you can't/shouldn't commercialize it - but it should be done under your
terms, with due, equitable return to the original creators.

I recognize the effort required to develop custom hardware to go along with
this, but it's important to remember that the need for this started with
(and relies on) the software, and you guys are the ones with the original
initiative to create this. To paraphrase one of my favorite movies, "if
they were capable of inventing the stratux, they would have invented the
stratux."

Thank you guys for what you're doing - if it hasn't already, one day your
work will save a life - whether by traffic avoidance, weather awareness, or
otherwise. It's a pretty cool thing you've come up with.

cyoung, 2015-10-21:
@HiltonG - AHRS has been working with FF since the start, then tested with Naviator and came up with this GDL90 message: https://github.com/cyoung/stratux/blob/master/main/ry835ai.go#L496-L535. On my testing platform it's sending messages that have attitude information in that format. It might have something to do with a lack of battery level and other identifying messages since this format mimics that of another receiver, but only sends the AHRS message. See the above link and if there's a format error we can pretty quickly remedy this.

Thanks for getting the word out at FLYING Expo.

HiltonG, 2015-10-22:
Chris, I wonder who 'designed' the first four bytes of that message and why those specific numbers? :)  BTW: There appears to be a problem with your logic "globalStatus.RY835AI_connected && globalSettings.AHRS_Enabled" - if I read that correctly this expression should only be true if I have an RY835AI attached, I don't and I'm still getting the GDL90 msgs sent by makeAHRSGDL90Report.

cyoung, 2015-10-22:
That's strange, maybe there is something connected that's being detected as the RY835AI. In either case, I do have an RY835AI connected and the AHRS data is being read from it and sent successfully over the XPlane protocol, and this 0x4C GDL90 message is working in Naviator.

ScraboTower, 2015-11-03:
Any update on this - its been quiet nearly 2 weeks :(

RobertGary1, 2015-11-21:
@cyoung, are you seeing this work in Naviator? I was not able to get it to work. I guess at this point we can only wait for @HiltonG to tell us what is wrong in the protocol stream?

cyoung, 2015-11-21:
Naviator works fine (upgrade to the newer version). WingX ... tried what I could, will have to wait for Hilton

ScraboTower, 2015-12-08:
Hilton published an update to his app today that includes "Enhanced Stratux Support"...  no clue as to what that is, or AHRS issues addressed

peeve, 2015-12-08:
no change as far as I can tell, still get a NO AHRS flag

HiltonG, 2015-12-08:
Can someone email a data stream to me please (with AHRS data)

Thanks,

Hilton

HiltonG, 2015-12-13:
Stratux AHRS support has been added to WingX Pro7 - this version will be released within days.

BTW: I have offered my help improve the Stratux status message etc to display a WAAS flag and more, but the offers were rejected.  So, you won't see a WAAS flag and other useful info bits in WingX Pro7 and other EFBs until they are added to the status message.

Hilton

skypuppy, 2015-12-13:
Then it might be time to add an LED or 5 coming off the GPIO pins, to
show various important states.

cyoung, 2015-12-13:
@HiltonG - thanks! Not rejected, just not sure what all we need to send over yet. I'll put WAAS on there to start sending in the stratux heartbeat. I see also GPS satellites was in there. Going to start filling out the blank fields with the heartbeat, then.

HiltonG, 2015-12-13:
Hi,

Clearly (at least) the initial focus of Stratux was on a cheaper ADS-B solution for FF than the AFS company's Stratus product; the name, some of the coding etc. I get that and I have no problem with it plus as I've always said, it's your project and a really cool one, amazing what you've done with this.  But let's be real here, FF magically no longer works with Stratux because they cleaned up the logic just a very short time after you got AHRS working in FF without Stratus?  Sure.   FF's explanation is that they don't want FF connected to FreeFlight and a sim at the same.  My question to them is exactly how would you do that when to connect to FreeFlight you need to connect to FreeFlight's local network?  No sim on there. Their intent is apparently obvious. The AFS company want to keep their Stratus, FF, Sportys cash cow and Stratux threatens their business.  They just did something similar to an EFB app (to be fair that EFB was apparently not playing nice and holding on to a communications port thereby preventing FF connecting to Stratus).

I have 'designed' status messages for many of the ADS-B receivers on the market. How about I design one for Stratux?  I will happily take input from you and anyone else willing to contribute, continually update it to a point where we're happy with the message format and its contents then we all implement it - this shouldn't take more than a few days to put together and implement. My guess is that unfortunately FF will never implement the message.

I will try post one tonight as a starting point and we'll go from there. Again, this is your project so the final choice is yours.

PS: If someone from FF would like correct anything I've said, please do.

Hilton

duecedriver, 2015-12-13:
Hilton... first let me say thanks for jumping into the pool!  I was a software engineer and U-2/KC135 pilot in the USAF.  I helped transition both those aircraft to glass cockpits and was an expert on human factors and glass integration as a graduate of the USAF School of Aerospace Medicine (Physiologist) .  I have done consulting on many of todays EFBs that go all the way back to the compaq ipaq.  I use your product because you are one of the few developers to not be playing the lock in game with hardware and your support of flight instructors.. I have actually walked away from paying contracts because developers wanted to hinder interoperability and am currently out of my exclusivity agreement with one of your competitors.. that said... 

for GPS... it would be nice to have  a numbers of sat lockup, a 3D solution warning, a comparator if multiple GPS units are in use.. i.e. a warning if the iPad GPS drifts X out of the attached external, predictive raim would be a bennie i.e. a lookahead in the flight plan or the ability to run raim at a destination.

here is a good open source lib of sources I think

http://www.gpstk.org/foswiki/pub/Documentation/GPSTkPublications/GaussiranBeaconGPSTK.pdf

i dont particularly like the way wingx displays any of the ADS-B message status... its just a indicator button.. other apps give breakouts
under GPS it would give lock and constellation status
under weather (FIS-B) it would give minutes from last update of each breakout item
etc
that would be far more useful

Also, in wing X I would prefer to see a green circle on traffic that is outside 5 miles and getting regular updates, a yellow ring for inside 5 and a red for inside 2... blinking red if its a target that has lost updates and its within 5. predictive flight path warnings too!

I have also posted sources to AHRS and IMU/INU that would be all the code needed for that to be world class in a different post

duecedriver, 2015-12-13:
Oh.. and if we can make the whole output stream compatible with AIRINC standards perhaps it could play well with a connected cockpit...

cyoung, 2015-12-13:
@HiltonG - another GDL90 custom message isn't a big deal really. Would it be out of the way to do something completely different? If the EFB could connect to Stratux, there's a ton of data already available in JSON format over a websocket. Custom GDL90 messages aren't really the most convenient solution nowadays, might have been in years past.

ScraboTower, 2015-12-14:
Separate issue perhaps but WingX does not always resume receiving the heartbeat if you have left the app and then resumed.  75% of the time its does not resume.   On today's flight it stopped after 25 mins and never resumed and the WingX was the only thing running on the ipad.  Reddit has other reports of similar issue reported by other WIngX users

HiltonG, 2015-12-15:
I have finished my initial draft of the Stratux Status Message.  Going to read through it again and post it here for comment hopefully later today.

Hilton

duecedriver, 2015-12-17:
whats the verdict....

HiltonG, 2015-12-17:
Proposed Stratux Status Message published here: https://github.com/cyoung/stratux/issues/150

peeve, 2015-12-19:
working beautifully with the new wingx release. Thank you!

cyoung, 2015-12-22:
Tested also - working great.

closed 2015-12-22 by cyoung