cyoung/stratux #68: Request: project status / milestones in Readme

egid, 2015-09-30:
I (and others, mostly on Reddit) have occasionally gotten a bit confused about the state of various features, especially AHRS.

Might help avoid confusion if there was a list of things that work, mostly work, are in progress, on the roadmap but unstarted, that sort of thing.

bradanlane, 2015-11-21:
@all - do we want to plan a release with what has been accomplished since 4r4?

Are there fixes or features we should get wrapped up?

_(I tend to keep tweaking stuff until there is a deadline)_

egid, 2015-11-21:
I can't speak to that, but this is addressed by #123!

@bradanlane, any interest in another js dev working on the web ui? I'd like to help out, and that's my area of expertise.

bradanlane, 2015-11-21:
@egid - stratux is "open source" so anyone is welcome to contribute. My suggestion is to do you dev in a branch or your own fork; then make a pull request.

I was about to work on a little refactoring of the code to bubble up the common settings so the various pages can visually "do the right thing". 
* weather page should display informational message when stratux configured for 1090 only
* traffic page should display informational message when stratux configured for 1090 only
* gps/ahrs page should display just gps, or just ahrs, or just informational message with stratux is configured for just gps, just ahrs, or neither respectively

I have not dont this yet but I did create bug #120 to track it.

If you are familiar with AngularJS, I would appreciate a code review to best practices since I was "learning on the fly" for this project.

egid, 2015-11-21:
I'm only passingly familiar with angular - the majority of my webapp work was Dojo, and now React - but I'll take a look.

I mostly want to make sure that if I'm going to mess around with stuff, I'm not stepping on work in progress. Once I get oriented in the code I can grab an issue and see where it goes.

bradanlane, 2015-11-21:
I railed against Dojo after far too much of it for a company I use to work for :-)

As for React, it was a tossup between React and Angular and I tossed a coin :-)

egid, 2015-11-21:
Oh yeah, Dojo is awful. React + Flux is quite nice! We needed the ability to quickly render predominately-js pages server-side, so React it was.

cyoung, 2015-11-21:
>@all - do we want to plan a release with what has been accomplished since 4r4?
>
>Are there fixes or features we should get wrapped up?
>
>(I tend to keep tweaking stuff until there is a deadline)

@bradanlane - biggest thing holding it back now is the new SDR code needs a bit of testing. I've tried it a bit, maybe we could get a count on how many people are using the latest? If more than a few we can roll it into a beta release and get more testing.

bradanlane, 2015-11-21:
Yeah, formal (or even deliberate) testing on the code would be ideal. I'll admit I have not flown the latest code. I can schedule a local hop but it would only be a "static test" - eg I will startup Stratux with replay on and fly but I can't monitor it in flight.

Also, if we determine the REST APIs appropriate for the SDR "tagging", then that should get added along with the UI for it.

cyoung, 2015-12-22:
@egid - thanks for resolving this, any further enhancements or additions are welcome.

closed 2015-12-22 by cyoung