cyoung/stratux #69: UPTIME different if unplugged vs shutdown

bradanlane, 2015-10-02:
Assume I power up Stratux at 09:00, then turn it off at 12:00 (noon), and finally power it up again the next day at 09:00. I get idfferent results based on how I turned Stratus off.

If at 12:00 I pull the power from Stratux, then on the "next day" the UPTIME starts at 00h00m00s.
If at 12:00 I SSH to the device and execute "shutdown now" then on the "next day" the UPTIME starts at 24h00m00s.

It is as if a controlled shutdown somehow preserves the original start time.

cyoung, 2015-10-03:
Isn't "shutdown now" just switching to singleuser mode? "shutdown -h now" probably is what you want to use

bradanlane, 2015-10-03:
While logged in a root, issuing "showdown now" yields a different result than simply unplugging the power. I'm just confused as to why.

kdknigga, 2015-10-03:
I've seen similar behavior with 'shutdown -h now'.

I thought it was odd, but I haven't looked into it.

cyoung, 2015-10-03:
Just tried 'shutdown now' after about 5 minutes uptime, left it off for a minute, pulled the plug and plugged it back in (isn't the only way to bring it back up after a 'shutdown' is to unplug it?), and the uptime started back up at 0. Any other way to reproduce it?

JohnOCFII, 2015-10-03:
> Assume I power up Stratux at 09:00, then turn it off at 12:00 (noon), and finally power it up again the next day at 09:00. I get different results based on how I turned Stratus off.

Are you referring to Uptime as displayed in the WebUI, or when issuing the -uptime- command at the Linux prompt?

bradanlane, 2015-10-03:
The WebUI "uptime" is what ideas looking at. The value comes from the /status websocket messages.

JohnOCFII, 2015-10-11:
Just adding an additional datapoint.  I just re-imaged my SD card from v0.4r1 to v0.4r2.  Even through the Pi was power-cycled and has a new image, the WebUI is showing an uptime of 19h39m42s. (and counting...).  I confirmed that the uptime command on the Pi itself is showing a more reasonable 7 minutes.  I force-quit mobile safari on my iPad, but no change in behavior.  It seems the field is not being initialized properly, or something similar.

bkwny, 2015-12-30:
I saw this too.  POR and then immediately via the web interface uptime showed about 10 hours.

JohnOCFII, 2016-01-05:
I just upgraded my Stratus to the latest Beta (v0.5-b4).  I powered up the stratux, which has been powered off for about a month.  Looking at the Web UI, it reports Update of 23h17m55s.

skypuppy, 2016-01-05:
Since RPi's don't have a battery backed-up clock, they're not real good
at time functions until they have finished booting up and getting time
set (somehow.)

commit 6303b0c5 (2016-01-07): Use monotonic timer for uptime.

Fixes #69.
closed 2016-01-07 by cyoung