cyoung/stratux #70: WebUI Weather experience

bradanlane, 2015-10-02:
The **Weather** page in the new WebUI needs more user control than other areas of the web experience.

While the **Traffic** page will typically see a handful of active targets at any one time and targets come and go within a few minutes, **Weather** is a different story.

I did a small sampling of pilots and came up with a couple suggestions. 

* separate METAR+TAF, PIREP, WINDS, ???? into sections so the user is looking at one section at a time
* limit METAR+TAF to a user configurable watch list

I'm interesting in more feedback.

JohnOCFII, 2015-10-02:
For weather, a quick, nicely formatted list of weather along my route (with some color for IFR/MVFR/VFR) would be nice.  I'm a ForeFlight user, so would turn to that App for my primary information.  I think of using the web-based Stratux app primarily as an administration/debugging tool for the Stratux device, in the same way that I consider my Stratus 2 a "turn it on and forget it" device.

bradanlane, 2015-10-02:
The WebUI is definitely growing far beyond "setup and diagnostics". That may be a poor decision. No real data to point one way or another.

I'd love to color-code weather but the algorithm is not simple. The best example I've seen is WeatherMeister which is a paid service. No one else has even come close. In an interview with the developer, he said it was a huge undertaking even to maintain the code base.

Similarly, finding reporting stations along a route requires a database of all possible reporting stations and a geolocation API to determine proximity.

A baby step in that direction is letting the user enter a series of stations.

JohnOCFII, 2015-10-02:
> I'd love to color-code weather but the algorithm is not simple. The best example I've seen is WeatherMeister which is a paid service.

I agree WeatherMeister looks great.  There is a free app for iOS that does a pretty good job.  I think there are similar apps for Android as well:  https://itunes.apple.com/us/app/aeroweather-lite/id288286079?mt=8
Based on the FAA or ICAO definitions for LIFR, IFR, MVFR, and VFR, the author displays a colored icon.

bradanlane, 2015-10-02:
@JohnOCFII - would you happen to know where I could find those FAA or ICAO definitions ?

mikey1299, 2015-10-02:
They are available here: https://www.aviationweather.gov/adds/metars/description/page_no/4

JohnOCFII, 2015-10-02:
> @JohnOCFII - would you happen to know where I could find those FAA or ICAO definitions ?

http://www.aviationweather.gov/adds/metars/description/page_no/4

I believe the ICAO definitions are the same, but expressed in meters.

bradanlane, 2015-10-02:
And in typical fashion, that doesn't define "ceiling". For that we go to part 14,of the FARs:

**Ceiling** - lowest layer of clouds or obscuring phenomena that is reported as "broken", "overcast", or "obscuration", and not classified as "thin" or "partial".

224XS, 2015-10-03:
@bradanlane:  while you are adding to the UI, remember to add a menu item and plate for Airspace uplink messages like SUA and TFRs.

For colors, I vote for the NOAA palette

For display of wx, I'd like to see some kind of notification or flagging of METARs as they change content during a flight.

bradanlane, 2015-10-03:
I'm not sure how useful SUA messages will be. here is an example:

SUA 211925 2233071|5550|W|M|ANNE LOW MOA|1012211925|1012212125|001|069|A|Y|53|ANNE LOW|MOA US 01022|ANNE LOW MOA, AR
I doubt anyone will find that information useful in-flight.

If Stratux receives **all** vs **local** SUA and TFR, the user will have to read and find the applicable reports. It's risky for the WebUI to presume to filter these.

The WebUI can include these "for completeness" but not "for use".

224XS, 2015-10-03:
True, but the SUA usually come with graphic overlays. It should be possible to distance sort/filter from the data in the graphic portion

bradanlane, 2015-10-03:
The WebUI will only know where "it is" if there is a GPS available. Even then, it may be best if this level of filtering is done somewhere in the backend where the code is more efficient. 

For the near term, I suggest we add this to a "futures" stack. We can get into the details and decision tree at some point.

224XS, 2015-10-03:
Please forgive me if I am being obtuse, but let me suggest that geolocation is a feature of nearly all
display devices running an app like FF or WingX, etc., thus a target device for the Stratux UI. It seems
all current HTML5 browsers support the geolocation API from what I see on W3C and other places.  Sometimes, the geolocation is not very accurate, if its just cell-tower based, but I think that market share is shrinking.
It just seems very valuable to have GPS for the planned UI including Traffic, Wx, self ship and self status, etc., at least as an option.

Easy enough to test for:   http://www.w3schools.com/html/tryit.asp?filename=tryhtml5_geolocation

bradanlane, 2015-10-03:
The issue is back to what JohnOCFII was eluding to. The WebUI was intended as an administrative interface to the Stratux device. If a user if fling with Stratux is is because they are using on of the apps you describe. Those app can and will always do a much better job at providing an interface to weather, traffic, etc.

The intent of displaying some amount of traffic and weather information was to confirm the device is functioning and to add with any issues a user may have in setup and configuration of the device.

The reason for this discussion was that it became immediately obvious that displaying all UAT uplink data would swamp the WebUI. Going back to the initial goal of "administration and diagnostics" displaying *any* of the UAT data is sufficient. I was looking for something that might also be interesting.

jpoirier, 2015-10-04:
What problem is the WebUI trying to solve? The WebUI is suppose to be an
"administration and diagnostics" page correct? If so, then that's what it
should be.

Personally, I don't want to have to bring up a page/s that has/have major
functionality overlap with my primary app just to get to the admin and
diagnostics parts.

bradanlane, 2015-10-04:
Do people agree with the following descending order (and if not, please provide your own defending list):

* first purpose of the WebUI is administration
* second purpose of the WebUI is diagnostics
* third purpose of the WebUI is debugging
* forth purpose of the WebUI is to provide interesting and useful ADSB data

224XS, 2015-10-04:
For my part, I agree that Settings Control, testing and debugging are primary.

I do think however, that there are some relevant flight data that can be be better communicated in a fresh interface than
any of the Pilot Apps currently in use. I will stipulate that my primary experience is with FF, but I have looked at the others
at SnF and KOSH.

The main issue with the pilot apps is that they are very cluttered and do not easily declutter.  If you are specifically looking for
traffic in a busy area, I think you can see targets with the best S/N against a blank radar-type screen or maybe the cockpit
view paradigm I put an example up of previously. If you don’t believe that app displays are cluttered, look at my sample of the
airspace around KATL(look up Stearman in reddit/stratux).  If Im buzzing VFR around the Bravo, and Approach is controlling 
my nav, I don’t really need the sectional and a clean screen with just traffic and maybe airspace borders would be nice.  Just
my opinion.

Here’s another suggestion for simplified QA of signals.  Display some very simple heartbeat indicators on the primary web UI
page. What I am thinking is separate animated iconic graphics for UAT download messages, Upload Traffic, Upload Wx messages,
and 1090 messages.  Cycle the animations (like a beating heart, e.g.) as properly formatted messages in each category are received.
Other pages might then allow a peek at the scrolling messages for wx, traffic targets etc.  That kind of display should inform the
user that the messages are being successfully relayed into the display device and then reserve deeper drill downs for the primary app.

If a GPS is installed, the usual full display of GNSS status should be displayed on a web UI page since it might not be available in the primary
app, and the GPS module and its comm link to the Rasp Pi is the subject of a lot of message traffic on reddit.

Cheers

> On Oct 4, 2015, at 5:48 AM, bradanlane <notifications@github.com> wrote:
> 
> Do people agree with the following descending order (and if not, please provide your own defending list): 
> 
> * first purpose of the WebUI is administration 
> * second purpose of the WebUI is diagnostics 
> * third purpose of the WebUI is debugging 
> * forth purpose of the WebUI is to provide interesting and useful ADSB data 
>

bradanlane, 2015-10-04:
I like the suggestion for a cleaner, more graphical status page. I'll investigate. 

I have my Garmin EFIS setup with a small "radar scope" styled window lower right as my primary scan for ADSB traffic. It's exactly what you describe. It's very easy to interpret and works well in a busy cockpit scan.

bradanlane, 2015-10-04:
I'm going to finish the very basic implementation of the **Weather** page using a watchlist. After I commit that code, I'll go back to the basic prioritized principles we've been discussing here and make improvements in descending order - starting with the status page. Once the GPS and AHRS code is stable, I'll build my Stratux test device and focus on that work.

JohnOCFII, 2015-10-04:
> Do people agree with the following descending order (and if not, please provide your own defending list):

I agree with your order and your reasoning.

bradanlane, 2015-10-04:
here is a quick video of the WebUI for Traffic and Weather ...
http://player.vimeo.com/video/141345333

cyoung, 2015-10-04:
Looks great!

JohnOCFII, 2015-10-05:
Fantastic!

caravitor, 2015-10-05:
This looks really great!

jpoirier, 2015-10-05:
Very nice!

224XS, 2015-10-05:
Very nice!  Useful too!!

rgutlon, 2015-10-05:
Bradan - nicely done! One thought on the weather ... an app I use shows "minutes old" which is really useful when quickly glancing at defined metars. Would you consider this for the "time" column?

bradanlane, 2015-10-05:
I agree with "minutes" old and will make that change. @cyoung wants to deliver a new "release" so I put a stake in the ground to get the basics done. 

In addition to "minutes old" I also want to color code METARs with the NOAA coloring for "flight conditions".

ChevyFlyer, 2015-10-09:
Put a off switch on the web ui to shut down the pi.

bradanlane, 2015-10-09:
Hi @chevyflyer - the "shutdown option in the WebUI has been discussed (in this issue topic or another).  

The best solution is to make the SD card read-only for normal operations. We don't want to require the typical user to launch the WebUI at the end of each flight in order to shutdown their Stratux.

bradanlane, 2015-10-09:
With the exception of METAR coloring (which I am working on) I think this issue has been addressed. I'm going to close it. We can open a new issue if there is a bug or for other enhancement requests.

closed 2015-10-09 by bradanlane

jpoirier, 2015-10-09:
Yes, the off switch was discussed but if I recall it wasn't taken off the table, and where was it stated that the user would be required to use it?

bradanlane, 2015-10-09:
I could provide the supporting research but it would be a lesson in cognitive psychology and learning curves. Suffice it to say, "adding a feature 'because we can' in not good UX design". Software should add what is needed and do it in the most intuitive way possible.

jpoirier, 2015-10-09:
lmao

kdknigga, 2015-10-09:
> suffice it to say, "adding a feature 'because we can' in not good UX
design".

*Zawinski's** Law*:

“Every program attempts to expand until it can read mail. Those programs
which cannot so expand are replaced by ones which can.”

jpoirier, 2015-10-09:
“Every program attempts to expand until it can read mail. Those programs
which cannot so expand are replaced by ones which can.”

Most focus on the first part of the statement but it's the second part that's important. The reality is that usage is the main driver and things that allow users to do more end up replacing those that don't.