cyoung/stratux #72: problem editing weather watchlist

bradanlane, 2015-10-06:
(reported by @rgutlon)

**v0.4r1 Weather Watch List Bug**
> Settings -> Configuration -> Watch List
> Once you add entries into this field they cannot be removed. You can append entries to the list, but backspacing over existing entries has no effect (ie the change is not being written to /etc/stratux.conf).

bradanlane, 2015-10-06:
I tried this and it worked. Can you give me the exact steps (right down to keystrokes) you tried. I think the issue may be that the coee doesn't know when the user is "done" so it waits for them to hit ENTER. I can look into trapping an event if hte user touches outside the field.

At the very least, I will add help text.

rgutlon, 2015-10-06:
> I think the issue may be that the coee doesn't know when the user is "done" so it waits for them to hit ENTER.

Just went through the process again paying attention to "returns" and you are correct.

One needs to press Enter (or GO on my iPad keyboard) in order for the change to be recorded. As the cursor neither moves position or changes visually, I may have thought GO was unnecessary and tapping outside of the field would record the input .

Thanks for double checking.

bradanlane, 2015-10-06:
I'll see if I can detect the "out of field" tap event. 

Interesting your iPad keyboard says "GO" and my iPhone (iOS 9.0.2) shows "Return" in the keyboard.

rgutlon, 2015-10-06:
Bradan - just went through the exercise again with different results.

* If you backspace and leave some value in the field (character, digit) it is recorded upon pressing GO (or Enter).
* If you backspace and remove everything so the field is empty (null), this is where the change is not saved. Are you able to replicate that behavior?

bradanlane, 2015-10-06:
Yes, making it blank doesn't work. I can fix that. Understand that a blank watchlist means "watch none".

bradanlane, 2015-10-06:
I've submitted new code that allows the user to submit an empty watchlist. It also tracks fiedl focus changes. IT's not perfect but it's better.

closed 2015-10-06 by bradanlane