cyoung/stratux #74: v0.4r1 - client loses connection (WebUI and ForeFlight)

JohnOCFII, 2015-10-07:
I downloaded v0.4r1 last night.  ForeFlight and the WebUI both lose connection after 2:18.  I can still connect from the same device using SSH.

I've attached logs, but on quick look, nothing stands out.

I can restart with sudo shutdown -r now, but same thing happens.

[Well, I tried to attach logs - even renaming them .txt didn't help, so I zipped them up and they are on DropBox.

https://www.dropbox.com/s/sknrkyjrc5g1e21/Stratux-logs20151006.zip?dl=0

bradanlane, 2015-10-07:
I've seen something similar but not 100% reproduce able. The WebUI will show "connected" but the uptime will stop incrementing. The stratux.log continues to get updates.

ScraboTower, 2015-10-07:
Seen the same with WingX , 2 to 5 mins then nothing, reboot and all is fine for a while, just have GPS and two receivers connected

bradanlane, 2015-10-07:
I have 2 SDR and no GPS/AHRS connected nor enabled.

JohnOCFII, 2015-10-07:
I removed power (as opposed to doing a sudo shutdown -r now) and now I've been up and connected with ForeFlight and the WebUI for 22 minutes and counting...

AvSquirrel, 2015-10-07:
Reproduceable (two radios + GPS connected via USB).

Symptoms are similar to the loss of WiFi connectivity seen in high UAT traffic areas on the 0.1 and 0.2 releases. ForeFlight and WebUI (via WiFi) lost connection after anywhere from 3-25 minutes. As in earlier builds, WebUI and SSH remained responsive via Ethernet connection.

JohnOCFII, 2015-10-07:
Whatever it is, it sure is random. After the power-cycle mentioned above, all ran well for 90 minutes before I shut it down to go to sleep.

When I saw problems earlier tonight, it was only in fresh data reaching ForeFlight or the WebUI. I never lost my WiFi / SSH connection to the Raspberry Pi. 

John

jbast, 2015-10-07:
Had the same issue. WebUI froze just over 40 seconds on several occasions. Tried cycling Wifi on my iPad, and power-cycling the Pi/SDRs as well. Reproduced a few times. Tried again the next day and it worked for several minutes without issue. I hate inconsistent bugs. Two radios, no GPS/AHRS, Foreflight. Thanks.

AvSquirrel, 2015-10-07:
Same issue was observed when using an alternate version of hostapd with an nl80211 compatible (TP-Link TL-WN722N) wifi adapter. That configuration died after 2:08 of uptime.

ssokol, 2015-10-07:
This all sounds very similar to #63. These both feel like deadlocks to me, since some functions running on other threads (web server, AHRS) continue to work after the lockup.

bradanlane, 2015-10-07:
I suspect we can compose a reproducible dataset which will work with "replay". We need to capture Log sets.

bradanlane, 2015-10-07:
Would it make it easier if the logs were rotated each time Stratux started?

ssokol, 2015-10-07:
Maybe, but the devilish thing about debugging multi-threaded code is that the exact right set of conditions need to happen on all of the involved threads at the same time. That *may* be reproducible from the logs, but it may not depending on what triggers the lockup.

bradanlane, 2015-10-07:
I know I have had the issue while using @cyoung replay logs. It has not been consistent but it has occurred several times. I am hoping we can get other Log sets which also reproduce the issue.

Alternately, we just start going back through the commits to find a commit point that is stable.

rgutlon, 2015-10-07:
Does it make sense to change the *Current version (stable)*  on reddit back to v0.2r1 until this is sorted out? Appears to be getting a lot of donwloads and I don't think v0.4r1 is stable enough to take flying ... unless one is collecting log information to help troubleshoot.

bradanlane, 2015-10-07:
Just ran a test using replay. It got "stuck" after 5m59s. 

* The WebUI would still refresh (F5 in the browser) which means the http serving the files from managementinterface.go was still working. 
* The "connected" status was still correct which means websocket was still alive handleStatusWS() was still functioning. 
* The WebUI "uptime" no longer changed (stuck at 5m59s) which likely means updateStatus() was no longer being called.
* The logs were still reporting new data for totalNetworkMessagesSent which suggests one of the 'sender' functions was still alive

Perhaps we are looking at something with the heartBeatSender() ?

JohnOCFII, 2015-10-07:
> The WebUI would still refresh (F5 in the browser) which means the http serving the files from managementinterface.go was still working.
> The "connected" status was still correct which means websocket [...]

Those symptoms match mine as well.

bradanlane, 2015-10-07:
I'm done to the problem being in sendTrafficUpdates(). It's either the body of the function or  cleanupOldEntries() or makeTrafficReport(). Perhaps the Mutex ?

JohnOCFII, 2015-10-07:
> One thing I had not noticed before is that my iPhone (sitting on the desk) had connected to "stratux". I don't normally have it doing that.

In most of my test cases, I did have more than one device connected to the Pi via WiFi (either iPad and iPhone, or iPad and MacBook Pro).

bradanlane, 2015-10-07:
I placed a debug message at the begin and end of traffic.go:cleanupOldEntries(). Both of these are displayed when the problem occurs. This suggests it is either the mutex, the loop code, or the call to makeTrafficReport().

JohnOCFII, 2015-10-07:
> @cyoung - didn't you tell me at some point to not connect multiple clients?

Hopefully that is not a long term concern.  As a flight instructor, I usually have at least two devices connected to my Appareo Stratus 2 unit, and would expect to do the same with the Stratux.

bradanlane, 2015-10-07:
it appears the problem is with the mutex.

I have four debug statements: 
* first line of sendTrafficUpdates()
* line before calling cleanupOldEntries()
* line after calling cleanupOldEntries()
* last line of sendTrafficUpdates()

At the point of failure, i see only the first debug statement. The only then between the first two debug statements is the mutex lock (and defer unlock).

ssokol, 2015-10-07:
I'm a total Go noob, so forgive me if I'm way off base here. I thought one of the real advantages of Go over C and C++ was integrated message passing with channels, rather than using a locking scheme to directly access memory. Could we revise the system to use channels?

bradanlane, 2015-10-07:
I've coded traffic.go to use a channel as a mutex with timeout capability. I'm testing it and will submit a branch for review.

ChevyFlyer, 2015-10-07:
Stratux test in N601LT
IPad Air FE392LL/A (wifi only)
iOS 9.0.2
Foreflight v 7.3.2

Stratux Ver 0.4r1

Kmashi battery running,  57D to Y47
This version doesn't behave well in FF with the bad elf 1000 and both radios (uat and 1090),  lose FreeFlight from devices during flight to Y47.

Changed to the Anker A1208 battery pack connected to stratux, both radios up
Flight back from Y47 to 57D

Removed bad elf and tried both radios, not good, uat only works.

Stuck bad elf back in during flight back to 57D and had them both running for a while.  Lost the towers then at 2500' but both devices show live in FF.

Log files available not sure which ones to include, just getting my Linux hat back on.

bradanlane, 2015-10-07:
Found it! (I reverted all of my changes to be sure.)

parseDownlinkReport() locks the traffic mutex when processing each string. The next to last step before finishing is to call registerTrafficUpdate() which will stick the new tracking on the trafficUpdate channel. The channel size is 1024. When nothing is taking traffic off the channel, it fills up and waits. Then sendTrafficUpdates() wakes up from its timer and is blocked by the traffic mutex lock.

cyoung, 2015-10-07:
@JohnOCFII - multiple clients should be no problem.

@bradanlane - excellent find. Thanks! Working on this now.

bradanlane, 2015-10-07:
@JohnOCFII - I had from 1 to 4 clients connected. The bug was producible independent of client count.

cyoung, 2015-10-07:
Oh, @ChevyFlyer - I was out testing this too at the same time as you, heard you on CTAF :)

ChevyFlyer, 2015-10-07:
Thought I heard FreeFlight @cyoung or the 122.7.

cyoung, 2015-10-07:
Guys, I just committed some code that you can reproduce this with.

No dongles plugged in, no web interface connections, if you do:

     ./replay ../test-data/cyoung-09062015-noproblem-stratux-uat.log ../test-data/cyoung-09062015-noproblem-stratux-es.log 10 | ../gen_gdl90

Watch it in ForeFlight. You'll see towers build up, message rate in the logs builds up, and then drops. Takes about 2-3 minutes. It does the exact same thing every time (except for a delta of 1-10 messages received before lockup).

You see in the log status updates like:

     2015/10/07 23:32:42  - UAT/min 2,005/2,005 [maxSS=0.00%], ES/min 0/0

And it will repeat that.

bradanlane, 2015-10-07:
Yup. I've been using replay all day. For me, it ran until 5m59s.

cyoung, 2015-10-08:
Give that a try

ssokol, 2015-10-08:
Is this connected to the GPS / AHRS bug, or is that a separate deadlock
issue?

cyoung, 2015-10-08:
No, it was a bug introduced when I added backend code for some of the new UI features.

JohnOCFII, 2015-10-08:
> @JohnOCFII - multiple clients should be no problem.

Excellent. Thanks!

cyoung, 2015-10-08:
v0.4r2 - feedback requested

AvSquirrel, 2015-10-08:
v0.4r2 disk image seems bad. Image file size is about 1 GB smaller than other releases and seeing multiple services failing at boot on local terminal -- no WiFi connectivity.

Reverted to the 0.4v1 image, and am now up and running on 0.4r2 compiled from the master branch.

AvSquirrel, 2015-10-08:
So far, so good. Uptime of 21 minutes so far, receiving light amounts of 1090ES traffic.

The new default ICAO address of 0x000000 seems to break ForeFlight's ownship detection -- according to [this page](http://www.kloth.net/radio/icao24alloc.php) it's an invalid address that "shall not be used for any application".

Instead, I updated stratux.conf to use 0xF00000, which would be in a block of ICAO-administered temporary addresses. An actual ownship address would be better, but this seems to be a safe option, especially for a non-transmitting system.

cyoung, 2015-10-08:
Good point, https://github.com/cyoung/stratux/commit/fa4718c2128af4c89602ae4d43625b47a83dd084.

AvSquirrel, 2015-10-08:
Still good. I let it run overnight, and have more than ten hours of uptime on  https://github.com/cyoung/stratux/commit/a2a7b114949d400c7d642cfe233c6dcdbb5c6c9d  with both receivers and GPS (USB) enabled. Peak message rates were 418 (UAT) and 1920 (ES).

cyoung, 2015-10-08:
Fairly certain it was just that, really that was the biggest change since v0.3b4 - which most people reported stable (excluding AHRS/GPS)

closed 2015-10-08 by cyoung

cyoung, 2015-10-10:
Have put 5 hours flight time on v0.3b4 - has been solid for me.

CL604pilot, 2015-10-11:
I'm going flying next week all over the US in a corporate jet. Want to try out the dual system weather and traffic (dual antennas). 

I downloaded V0.4r2. What other program do I need to download that is current?

I will report back when we're done and send files if required.  

Thanks guys,

Owen

ChevyFlyer, 2015-10-11:
Put about an hour on the 4r2 image today, no crashes or lock up. Had
FreeFlight running before hooking up the Bad Elf, they seemed to behave.
No real weather in the Thumb of Michigan today but I did see the traffic
around DTW.