cyoung/stratux #75: GDL39

cyoung, 2015-10-07:
If anyone is interested in implementing GDL39 - 

http://www.chartbundle.com/tech/gdl39/

Interesting paper: http://cseweb.ucsd.edu/~savage/papers/CCS14MobileCockpit.pdf

bradanlane, 2015-10-20:
I assume, in addition to the coding for encryption and handshake, this would also require Bluetooth support- yes?

cyoung, 2015-10-20:
I think so - there are some avionics that take only GDL39 over serial also. In terms of using it for Garmin Pilot it's probably bluetooth only.

skypuppy, 2015-10-20:
I'm told that Garmin encrypts their transmissions, at least over the 
air.  Haven't heard about direct wired.

Skypuppy7

cyoung, 2015-10-20:
It uses blowfish encryption with a key that is negotiated in the handshake, the above link describes it.

skypuppy, 2015-10-20:
If they wanted to compress the data interchanges, that's one thing.  But *encrypting* it is downright <censored>!!!  Another good reason to never buy Garmin gear.  Bah.  /rant off

monadical, 2015-12-09:
The GLD39 looks pretty straightforward and interesting. The paper is very interesting, and I agree that security between devices would be a great idea. Of course it requires agreement on the approach between the app and the device makers. BTW: the Garmin approach to security from looking at the GDL39 code is embarrassing. I'm starting to get the impression Garmin doesn't care. :)

n980km, 2016-02-16:
What I'd love to see would be GDL-39 serial output so I can use Stratux to display traffic and weather on my Garmiin 696, instead of having to buy a GDL-39 to do it.  Anyone with a Garmin x96 GPS -- and there are a lot of them out there -- could benefit from it.

Snowflake6, 2016-12-11:
As with n980km, i'd like to see GDL-39 serial output so I could feed a Garmin handheld.  In my case an Aera 660.  I know many pilots with Aeras (500, 510, 550, 660, 760) and they have all been following the Stratux development and would all like to see this capability.

jamez70, 2016-12-20:
I'm going to see if I can get it to work

braice, 2017-05-06:
Hello,
Did you succeed to write something ?
It can be a great feature and I will be happy to provide help testing/debugging.
Thank you

lukepalmer, 2017-05-16:
Does anyone own a GDL 39 (preferably 3D) who would be willing to provide a few minutes of a bluetooth packet capture?

Here's a reference on how to collect the traffic:
https://www.nowsecure.com/blog/2014/02/07/bluetooth-packet-capture-on-android-4-4/

FlyingSikh, 2017-06-13:
I don't own a GDL39 but have an Area 796 with old Zaon connected to it via serial connection. Let me know if I can help in any way.

monadical, 2017-06-13:
The flight schools planes I rent now have GTX 345's which can all talk to ForeFlight (and Pilot) via the built in Connext link. I assume that's the same protocol data as the GDL 393D. I can work on capturing steams from that, probably just via a laptop, because laptops.

Snowflake6, 2017-06-13:
@lukepalmer, the bluetooth protocol has been documented previously... See the link in the FP.  Implementing Connext would only help people who are buying new Garmin hardware.  The goal with supporting GDL39 would be to support all of the existing Garmins in the field that can receive GDL39 data.  If you can afford new Garmin hardware that talks Connext, you probably don't need a Stratux.

monadical, 2017-06-13:
If all of the GDL39 (or Connext otherwise) is fully documented, then our job is done. It's just implementation work. If not, let me know, happy to capture streams to aid in full understanding the protocols. Of course we're borrowing GDL39s (or Connext setups) to get this data so the Stratux can act as a GDL39 (just like emulating Stratus). Obviously if we have either of those devices we don't need Stratux. :)

lukepalmer, 2017-06-13:
@Snowflake6 I'm not aware of any work on the 3D variant (with AHRS). Would be great if you know something I don't though! The thing that makes this interesting would be using Stratux with Garmin Pilot.

However: I think I've managed to get my hands on one. I'll work on getting a dump to post here.

Snowflake6, 2017-06-13:
@lukepalmer Personally I would consider the AHRS to be a bonus... I'm more interested in getting traffic and weather onto my Aera 660 without buying a GDL39 or an XM subscription.

jamez70, 2017-06-13:
We really need to get a hold of a GDL39. I was working on this months ago
but I wasn't able to make any headway due to not having captures from a
GDL39.

kdknigga, 2017-06-14:
I just ordered a GDL39.  I'll try getting a capture.

0x74-0x62, 2017-06-16:
I am also very interested in getting the GDL39 protocol to work (to feed Garmin Pilot). Unfortunately, I do not have access to a GDL39 source device. Please let me know if there is any way to support the development activities either by coding or testing.

jamez70, 2017-06-16:
Maybe I should just buy one with a good return policy :)

kdknigga, 2017-06-21:
Got my GDL39 today.  Here's a bluetooth capture from my living room.  As such, it's just GPS and air-to-air traffic (mostly 1090 traffic out of Midway and Ohare).

https://drive.google.com/file/d/0B5KKdvTfqQ4vN2VmbHhmOHk0SXc/view?usp=sharing

kdknigga, 2017-06-25:
Here's a trip I took over the weekend.  It should have everything: all forms of traffic, weather, GPS, and a couple of Garmin Pilot/GDL39 handshakes.

https://drive.google.com/file/d/0B5KKdvTfqQ4vWENjSmNMZGJIZ1U/view?usp=sharing

jamez70, 2017-06-30:
What did you use to capture?

kdknigga, 2017-06-30:
Android has a built-in Bluetooth capture facility in the developer menu.
Those logs will open in Wireshark.

jamez70, 2017-06-30:
@kdknigga From your coordinates, it appears you live about 5 miles from me. Maybe we should just get together sometime?

kdknigga, 2017-06-30:
@jamez70  Sounds good to me.

cptfbi, 2017-09-11:
any update on this?  Loving my Stratux but I like using garmin pilot as well as foreflight.  Wish I knew more to help this along.

Snowflake6, 2018-02-04:
Has anyone made any progress on this?  @jamez70 @kdknigga ?

kdknigga, 2018-02-04:
No, sorry.

ninelima, 2018-08-21:
Apps4Av has some GDL39 code. See: 
https://github.com/apps4av/avarehelper/tree/master/app/src/main/java/com/apps4av/avarehelper/gdl90

kdknigga, 2018-08-21:
While the GDL 39 does use the GDL 90 protocol, it requires a cryptographic
handshake before sending the data (and might even encrypt the data in
transit, I'm not sure).  Unfortunately, to my knowledge, no one has figured
out the handshake yet that gets the data flowing.

armeniki, 2018-09-02:
@kdknigga Just wondering ... isn't the encryption method described in the link above?

http://www.chartbundle.com/tech/gdl39/

kdknigga, 2018-09-02:
Perhaps.  It's over my head, though.

armeniki, 2018-09-02:
No worries!  Well, I'm definitely up for helping out with this.  I've done some developing (mostly in C and Visual Basic) for an access card system based on the AES2 and MiFare so the encryption stuff shouldn't be a problem.  I am, however, unfamiliar with the RPi hardware, especially Bluetooth.  Most of my experience has been on Atmel and PIC micro controllers.  In any case, I'll start looking into what's needed to get a test environment together.

upsetm16bandit, 2019-08-02:
How old is the information on the gdl39 scheme at http://www.chartbundle.com/tech/gdl39/  ?   Does anyone know if the handshake information is still applicable, or has garmin changed it via firmware update?

ecadenas, 2019-08-16:
2012-09-02 23:06

jamez70, 2020-06-26:
I started hacking this with a GPSMAP 696 and a GDL39. That is serial only, but I have learned quite a bit. I have a good idea how the weather and traffic come across, but then writing code to emulate a GDL39 is another task altogether.

It starts at 9600 baud then switches to 57600 baud serial. The serial isn't encrypted at all either. Since I live near a ground station I am able to receive basically all the different messages exchanged.

saabnut, 2020-06-26:
That's useful info.

> I started hacking this with a GPSMAP 696 and a GDL39. That is serial
only, but I have learned quite a bit. I have a good idea how the weather
and traffic come across, but then writing code to emulate a GDL39 is
another task altogether.
>
> It starts at 9600 baud then switches to 57600 baud serial. The serial
isn't encrypted at all either. Since I live near a ground station I am able
to receive basically all the different messages exchanged.
https://ci3.googleusercontent.com/proxy/g2Rj0BVEjY89JsMw3pCnRewaBKDWpBLsr6DUHDZe9oCX0LYjo5FzvjY1tKkHKCySiaKLwlsFFkROCM3IQTcZGGgqDL-URNKviud_InQqBnur_fxzx2vgYANG55a5V8nIrbiNlbw5L1RvGTlqQw2QySBBtmVwOBrD-DkGyMUVnbbsx0X85LLI0CdJEPfjGZqyGLoT29H14KZq-yWOkXBZ6R8no2YmornGhOk28eQC1Q=s0-d-e1-ft#https://github.com/notifications/beacon/AHAT66AXS7BKCUDKBVCB6RDRYTSODA5CNFSM4BRLLT42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOE3BUEVY.gif>

jamez70, 2020-06-26:
I could use some help if anyone wants to decipher some of these files...

I make a 2 port serial sniffer from an Atmel ATSAMC21 eval board, which sends the serial captured data to a windows application. In the windows application, I reconstruct the frames. Any frames that are files I wrote code to extract the files from them, and it saves them as files.

I'll show you a sample..

[image]

I basically have to figure out what the leading numbers before the '78 5e' mean. I think its loosely described in the document but its not exactly correct.

Jim

jamez70, 2020-06-27:
Here is the initial exchange at 9600:

`(10047) GDL: 10 0A 02 B2 00 42 10 03  '.....B..'
(11031) GDL: 10 0A 02 B2 00 42 10 03  '.....B..'
(12047) GDL: 10 0A 02 B2 00 42 10 03  '.....B..'
(13047) GDL: 10 0A 02 B2 00 42 10 03  '.....B..'
(13047) GPS: 10 06 02 0A 00 EE 10 03  '........'
(13719) GDL: 10 F9 00 07 10 03  '......'
(13719) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(13719) GPS: 10 06 02 F9 00 FF 10 03  '........'
(13719) GPS: 10 FF 89 E9 02 0C 03 47 50 53 6D 61 70 20 36 39 36 20 53 59 53 20 53 6F 66 74 77 61 72 65 20 56 65 72 73 69 6F 6E 20 37 2E 38 30 00 56 45 52 42 4D 41 50 20 4E 6F 6E 65 00 56 45 52 53 4D 41 50 20 4E 6F 6E 65 00 56 45 52 41 56 44 42 00 56 45 52 41 55 44 20 41 76 69 61 74 69 6F 6E 20 41 75 64 69 6F 20 50 6B 67 20 20 32 2E 30 30 00 56 45 52 53 50 4C 53 43 52 4E 20 53 70 6C 61 73 68 20 53 63 72 65 65 6E 20 32 2E 30 30 00 0C 10 03  '.......GPSmap 696 SYS Software Version 7.80.VERBMAP None.VERSMAP None.VERAVDB.VERAUD Aviation Audio Pkg  2.00.VERSPLSCRN Splash Screen 2.00....'
(13750) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(13750) GDL: 10 06 02 FF 00 F9 10 03  '........'
(13750) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(13750) GPS: 10 F8 7A 56 45 52 47 50 52 4F 4D 20 4E 6F 6E 65 00 56 45 52 53 54 44 42 20 41 6D 65 72 69 63 61 73 2D 39 20 54 65 72 72 61 69 6E 20 31 36 54 31 00 56 45 52 53 4F 44 42 20 55 53 2F 43 61 6E 61 64 61 20 4F 62 73 74 61 63 6C 65 73 20 31 38 42 31 00 47 50 53 20 4E 2F 41 00 56 45 52 47 50 49 20 41 4F 50 41 20 55 53 20 44 69 72 65 63 74 6F 72 79 20 44 61 74 61 20 31 38 44 31 00 9B 10 03  '..zVERGPROM None.VERSTDB Americas-9 Terrain 16T1.VERSODB US/Canada Obstacles 18B1.GPS N/A.VERGPI AOPA US Directory Data 18D1....'
(13765) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(13765) GDL: 10 06 02 F8 00 00 10 03  '........'
(13765) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(13781) GPS: 10 FD 96 50 00 00 4C 01 00 41 0A 00 54 01 00 41 64 00 44 6E 00 41 C9 00 44 CA 00 44 6E 00 44 D2 00 41 2D 01 44 38 01 44 2E 01 41 90 01 44 6E 00 41 F4 01 44 F5 01 41 58 02 44 58 02 41 59 02 44 59 02 41 8A 02 44 8A 02 41 BC 02 44 BC 02 41 20 03 44 20 03 41 22 03 44 22 03 41 84 03 41 85 03 44 86 03 41 86 03 41 87 03 41 89 03 44 84 03 41 91 03 44 91 03 41 8B 03 44 8B 03 44 8C 03 44 8D 03 44 8E 03 41 94 03 41 95 03 44 95 03 41 96 03 44 96 03 41 8C 03 44 8F 03 BB 10 03  '...P..L..A..T..Ad.Dn.A..D..Dn.D..A-.D8.D..A..Dn.A..D..AX.DX.AY.DY.A..D..A..D..A .D .A".D".A..A..D..A..A..A..D..A..D..A..D..D..D..D..A..A..D..A..D..A..D.....'
(13797) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(13875) GDL: 10 06 02 FD 00 FB 10 03  '........'
(13875) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(13875) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(14031) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(14031) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(14234) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(14234) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(14422) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(14422) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(14625) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(14625) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(14828) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(14828) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(15015) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(15015) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(15219) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(15219) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(15422) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(15422) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(15625) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(15640) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(15828) GDL: 10 00 02 20 00 DE 10 03  '... ....'
(15828) GPS: 10 06 04 00 00 20 00 D6 10 03  '..... ....'
(15844) GPS: 10 00 06 21 00 65 53 85 D3 C9 10 03  '...!.eS.....'
(16281) GDL: 10 06 04 00 00 21 00 D5 10 03  '.....!....'
(16281) GPS: 10 F9 00 07 10 03  '......'
(16281) GDL: 10 06 02 F9 00 FF 10 03  '........'
(16281) GDL: 10 FF 4E 3F 05 EB 01 47 44 4C 33 39 20 53 79 73 74 65 6D 20 53 6F 66 74 77 61 72 65 20 34 2E 39 31 00 47 44 4C 33 39 20 41 54 6D 65 67 61 20 53 6F 66 74 77 61 72 65 20 32 33 32 00 4D 54 33 33 32 39 20 53 6F 66 74 77 61 72 65 20 32 2E 37 30 00 CC 10 03  '..N?...GDL39 System Software 4.91.GDL39 ATmega Software 232.MT3329 Software 2.70....'
(16312) GPS: 10 06 02 FF 00 F9 10 03  '........'
(16312) GPS: 10 30 04 00 E1 00 00 EB 10 03  '.0........'
(16312) GDL: 10 FD 0F 50 00 00 4C 01 00 41 0A 00 41 96 03 44 96 03 55 10 03  '...P..L..A..A..D..U..'
(16328) GPS: 10 06 02 FD 00 FB 10 03  '........'
(16328) GDL: 10 06 02 30 00 C8 10 03  '...0....'
(16328) GDL: 10 31 04 00 E1 00 00 EA 10 03  '.1........'
`

jamez70, 2020-07-04:
so I have successfully paired a simulator with the 696 using the protocol. Its talking back and forth, but its not displaying traffic or weather yet. Once I figure out how to package the traffic and weather data I will add that to the simulator.

Once I can simulate it, I could potentially port it to the stratux to at least connect to a GPS over a serial port and display ADS-B data.

Snowflake6, 2020-07-06:
@jamez70 this is excellent news, looking forward to hearing more as work progresses!  My Aera 660 is standing by... :)

-Rob

jamez70, 2020-07-11:
Don't get too excited - I can't figure out how the GDL39 tells the GPS to request a file other than traffic_state. There is a whole slough of them that contain METARS, TAFS, Ground station logs, NOTAMS, PIREPS, Winds Aloft, etc. Its like looking in a pile of needles for a needle too

jamez70, 2021-04-28:
I have successfully cracked their blowfish key exchange and hopefully my simulator will be fully up and running

kdknigga, 2021-04-28:
Hot damn, man!

jwbruce, 2021-04-29:
I have a 696.  I sold my GDL39 years ago.  I would be happy to help out with the Stratux-GDL39 emulation in any way you need.  I would ultimately run my 696 with the Stratux over a hardwire serial connection.

ecadenas, 2021-04-29:
Aera500 with stratux would make great adsb traffic display...

Already have the aera500 tx line going to av30.

Could be a really nice redundancy to have the tailbeaconX waas output going to stratux aswell.

2cents...

Thanks for your support.

Regards,

E. Cadenas De Frutos
T: 1•514•906•7839
F: 1•866•208•3426

Snowflake6, 2021-04-29:
There are a lot of devices that would benefit from being able to make the Stratux emulate a GDL39.  AFAIK, everything from a 296 to the modern Aera devices would work, my Aera 660 included.  :)

jamez70, 2021-04-29:
@Snowflake6 you are right. Many to list and the benefits are overwhelming - if Garmin doesn't shut it all down.

I own a 496, 696 and an Aera 660 with a GDL39. I have connected the GDL39 to the 696 and wrote this embedded application to capture messages with timing references at both ends. I have about 100 different captures total. My capture app is pretty sophisticated in that it even saves the transferred files to disk for later viewing.

Anyway, about a year ago (6/2020) I got stuck and gave up. Recently, I set up the lab again and looked into the '22 00' packet. The reference about GDL39 out there at http://www.chartbundle.com/tech/gdl39/ is about half right, but they cover emulating a GPS, not a GDL39. After over 10 hours of trying I was finally able to generate a key successfully.

I added that to the simulator but it didn't get me anywhere else as of yet. There are about 15 different types of messages exchanged and its hard to recreate the sequence - meaning what messages drive what messages in order to emulate the communications properly.

I am still missing something but I believe that will be overcome pretty soon.

Jim

jamez70, 2021-04-29:
Here is the 57600 baud portion of a capture

`(8563) GPS: 0A 02 3A 00  '..:.'
(8969) GDL: 00 08 31 01 03 00 3A 09 00 00 
(8969) GPS:  06 04 00 00 31 01  '....1.'
(9266) GPS: 0A 02 3A 00  '..:.'
(9282) GDL:  06 02 0A 00 
(9282) GPS: 0A 02 3A 00  '..:.'
(9282) GDL:  06 02 0A 00 
(9282) GPS: 00 04 10 00 FF 03  '......'
(9282) GDL:  06 04 00 00 10 00 
(9282) GDL: 00 04 11 00 FF 03 
(9282) GPS:  06 04 00 00 11 00  '......'
(9282) GPS: 0A 02 00 3A 00  '...:.'
(9282) GDL:  06 02 00 0A 00 
(9282) GPS: 0A 02 00 3A 00  '...:.'
(9282) GDL:  06 02 00 0A 00 
(9282) GPS: 00 02 00 20 00  '... .'
(9282) GPS: 00 02 00 2E 01  '.....'
(9282) GPS: 00 07 00 2C 01 01 60 00 01 01  '...,..`...'
(9282) GPS: 00 02 00 20 00  '... .'
(9282) GPS: 00 07 00 2C 01 01 60 00 01 01  '...,..`...'
(9282) FILE REQUEST id (1): /RAM/TRAFFIC_STATE
(9282) GDL:  06 04 00 00 00 20 00 
(9282) GDL: 00 06 00 21 00 AF B6 7F 
(9282) GDL:  06 04 00 00 00 2E 01 
(9282) GPS:  06 04 00 00 00 21 00  '.....!.'
(9282) GDL: 00 04 00 2F 01 00 00 
(9282) GDL:  06 04 00 00 00 2C 01 
(9282) GPS:  06 04 00 00 00 2F 01  '...../.'
(9282) GDL: 00 07 00 34 01 01 60 00 01 01 
(9282) GDL:  06 04 00 00 00 20 00 
(9297) GPS:  06 04 00 00 00 34 01  '.....4.'
(9297) GDL: 00 06 00 21 00 AF B6 7F 
(9297) GDL:  06 04 00 00 00 2C 01 
(9297) GPS:  06 04 00 00 00 21 00  '.....!.'
(9297) GDL: 00 07 00 34 01 01 60 00 01 01 
(9297) GPS:  06 04 00 00 00 34 01  '.....4.'
(9297) FILE ACK id (1) handle (0) size 0
(9375) GPS REPORT
(9375) GDL GPS Report ACK
(9563) GPS REPORT
(9579) GDL GPS Report ACK
(9766) GPS REPORT
(9766) GDL GPS Report ACK

----------------- MARK 1 ----------------

(9969) GPS REPORT
(9969) GDL GPS Report ACK
(9969) GDL: 00 08 00 31 01 04 00 3A 09 00 00 
(9969) GPS:  06 04 00 00 00 31 01  '.....1.'
(10172) GPS REPORT
(10172) GDL GPS Report ACK
(10360) GPS REPORT
(10360) GDL GPS Report ACK
(10579) FILE REQUEST id (1): /RAM/TRAFFIC_STATE
(10579) FILE ACK id (1) handle (1) size 86
(10579) GPS REPORT
(10579) GPS: 00 0A 00 08 00 01 00 01 00 00 00 00 00  '.............'
(10579) GDL GPS Report ACK
(10579) FILE DATA handle (1) handle index 1 - 86 bytes: (10579) File /RAM/TRAFFIC_STATE received with 86 bytes 
(10579) GPS: 00 0A 00 08 00 02 00 01 00 00 00 00 00  '.............'

----------------- MARK 2 ----------------

(10766) GPS: 00 0A 00 22 00 9A 0B 10 3D ED 3D 39 26  '..."....=.=9&' <- blowfish seed?
(10766) GDL:  06 04 00 00 00 22 00 
(10766) GDL: 00 0A 00 22 00 38 FA D3 8C ED 7B 48 42  <- blowfish response?
(10766) GDL GPS Report ACK
(10766) GPS:  06 04 00 00 00 22 00  '.....".'
(11032) GPS REPORT
(11032) GPS: 00 03 00 65 00 1E  '...e..'
(11032) GPS: 00 03 00 35 01 01  '...5..'
(11032) GPS: 00 07 00 66 00 0A 00 00 00 02  '...f......'
(11032) GPS: 00 07 00 38 01 01 00 00 00 00  '...8......'
(11032) GPS: 00 0B 00 CB 00 05 0B 01 1D 1E 20 46 2A 0D  '.......... F*.'
(11032) GDL GPS Report ACK
(11032) GDL:  06 04 00 00 00 65 00 
(11032) GDL: 00 03 00 65 00 1E 
(11032) FILE REQUEST id (0): /RAM/GROUND_STATION_LOG
(11032) GDL:  06 04 00 00 00 35 01 
(11032) GPS: 00 02 00 C9 00  '.....'
(11032) GDL:  06 04 00 00 00 66 00 
(11032) GPS:  06 04 00 00 00 65 00  '.....e.'
(11032) GDL:  06 04 00 00 00 38 01 
(11032) GDL:  06 04 00 00 00 CB 00 
(11032) GDL: 00 08 00 31 01 04 00 3A 09 00 00 
(11047) GPS:  06 04 00 00 00 31 01  '.....1.'
(11047) FILE ACK id (0) handle (0) size 0
(11047) GDL:  06 04 00 00 00 C9 00 
(11047) GDL: 00 02 00 CA 00 
(11047) GPS:  06 04 00 00 00 CA 00  '.......'
(11047) FILE REQUEST id (14): /RAM/ASSERT_LOG
(11141) GPS REPORT
(11329) GPS REPORT
(11500) GDL GPS Report ACK
(11500) GDL GPS Report ACK
(11500) FILE ACK id (14) handle (2) size 56270
(11500) GPS: 00 0A 00 08 00 01 00 02 00 00 00 00 00  '.............'
(11500) FILE REQUEST id (1): /RAM/TRAFFIC_STATE
(11500) GPS REPORT

----------------- MARK 3 ----------------

(11719) FILE DATA handle (2) handle index 14 - 1015 bytes: 
(11719) FILE ACK id (1) handle (3) size 1181
(11719) GDL GPS Report ACK
(11719) GPS REPORT
(11719) GPS: 00 0A 00 08 00 01 00 03 00 00 00 00 00  '.............'
(11719) GDL GPS Report ACK
(11875) GPS REPORT
(11875) FILE DATA handle (3) handle index 1 - 1015 bytes: 
(12016) GPS: 00 0A 00 08 00 02 00 03 00 00 00 00 00  '.............'
(12032) GPS: 00 02 00 2E 01  '.....'
(12032) GDL GPS Report ACK
(12032) GDL:  06 04 00 00 00 2E 01 
(12032) GDL: 00 04 00 2F 01 00 00 
(12032) GDL: 00 03 00 35 01 01 
(12032) GPS:  06 04 00 00 00 2F 01  '...../.'
(12032) GPS:  06 04 00 00 00 35 01  '.....5.'
(12032) FILE DATA handle (3) handle index 1 - 166 bytes: (12032) File /RAM/TRAFFIC_STATE received with 1181 bytes 
(12047) GPS: 00 0A 00 08 00 02 00 03 00 F7 03 00 00  '.............'
(12157) GPS REPORT
(12157) GDL GPS Report ACK
(12360) GPS REPORT
(12360) GDL GPS Report ACK
(12547) FILE REQUEST id (1): /RAM/TRAFFIC_STATE
(12547) FILE ACK id (1) handle (4) size 1284
(12547) GPS REPORT
(12547) GPS: 00 0A 00 08 00 01 00 04 00 00 00 00 00  '.............'
(12547) GDL GPS Report ACK

----------------- MARK 4 ----------------

`

jamez70, 2021-04-30:
[image]
**I GOT IT!!!!!!!!!!!!!!!!!!**

saabnut, 2021-04-30:
KUDOS!
Be careful with updates to garmin devices in the future, so they don't
break it.

jamez70, 2021-04-30:
Well, not really feasible for them, because in order to do that, you'd have to update every device they make. The key exchange process would have to be changed on every device at the same time

saabnut, 2021-04-30:
good to know, great work. Is this some code we would add to stratux in
order to get emulation done?

jamez70, 2021-04-30:
We are quite a ways from that yet. I still have to figure out how to command the GPS to accept traffic and weather data

borgspace, 2021-05-08:
Hi Jamez
I stumbled across your work as I'm doing something similar to provide connectivity between an Aera660 and a SkyEcho 2 in the UK
I've had much the same experience as you with issues on the handshake - but I have some packet captures from a working Aero and GDL39 that show the interaction after the handshake and the traffic messages look to be sent in a clear (unencrypted) format (starting and ending with 7E)
The sky echo natively outputs GDL90, so I was hoping to get the handshake made and then pass the messages from sky echo to Aera unmodified.
I don't suppose you have any code you'd be willing to share on the handshake do you? Happy to share what I have in return - although I suspect you have most of it already from the same sources :)

jamez70, 2021-05-08:
I have everything completely figured out and I have connected to a Garmin 696 and an Aera 660.
Lots of work to convert stratux formatted things to Garmin formatted things, but this is quite possible,

jamez70, 2021-05-08:
> Hi Jamez
> I stumbled across your work as I'm doing something similar to provide connectivity between an Aera660 and a SkyEcho 2 in the UK
> I've had much the same experience as you with issues on the [...]

I can explain how its done, or share some code. Everything I needed was in the chartbundle but the key is incorrect (its commented out in the code though). Maybe we can chat directly?

borgspace, 2021-05-08:
> I can explain how its done, or share some code. Everything I needed was in the chartbundle but the key is incorrect (its commented out in the code though). Maybe we can chat directly?

@jamez70 that would be fantastic, thank you. 
steve@borgspace.co.uk

[49 more replies]