cyoung/stratux #76: iOS9 testing

cyoung, 2015-10-08:
1. What determines when the Wi-Fi icon is displayed (top left)? My home network shows it when connected, but stratux doesn't.
2. Any difference with newer hostapd versions?
3. Any difference with another adapter?
4. Any difference with Wi-Fi Assist OFF (Settings -> Cellular Data -> Wi-Fi Assist)?

AvSquirrel, 2015-10-08:
It seems to be related to the default route changes you made at v0.2b1.

With 0.0.0.0 as the default route, the Wi-Fi icon never appears. Reverting to 192.168.10.1, the Wi-Fi icon appear shortly after negotiating a lease. This behavior was consistent on both iOS 8.4 and 9.0.2, although the newer OS seems to take longer to "settle in" after connecting.

JohnOCFII, 2015-10-08:
I, too noticed that the WiFi icon is not showing when I connect to Stratux, but it does show when I connect to my home router (an Asus router running the Asus-Merlin firmware).

I know that if the iOS device is connected to a peer-to-peer network as opposed to a base-station network, you will get a different symbol.  This makes me think it is related to hostapd.   

Also - the connection to the stratux SSID takes quite a bit longer than connecting to my home router.

John

cyoung, 2015-10-08:
I don't think it's related to the DHCP default route. Unless iOS is caching that in a strange way, changing it back had has no effect.

Just got stratux to show the Wi-Fi symbol now on a fresh install of v0.4r2. I *think* turning off Wi-Fi Assist did the trick.

trippingly, 2015-10-08:
I'm going off of a 0.4r2 build rather than the image... I can confirm the connection to stratux has a longer "pending" circling on the wi-fi settings page than does a connection to a home router, though stratux is transmitting freeflight to FF successfully even while the pending icon is circling.

For me (again, not from image), turning off the wi-fi assist still does not result in a wi-fi symbol.

AvSquirrel, 2015-10-08:
With a default route of 0.0.0.0, I never get the Wi-Fi icon on 9.0.2, regardless of what I do to the cellular settings (Wi-Fi assist on / off; cellular data on / off; airplane mode on / off).

The symbol is displayed when I change to a valid default route and reconnect to the stratux network.  Simply renewing the lease after restarting isc-dhcp-server with the changed to the route isn't enough; I had to connect to another network first to make the icon appear / disappear.

I can't get it to toggle states if I just "renew lease" from Setttings > Wi-Fi > stratux ( i )  , but it does toggle if I connect

JohnOCFII, 2015-10-08:
I just changed /etc/network's default route to 192.168.10.1, restarted the pi,  and did not get the WiFi symbol.

AvSquirrel, 2015-10-08:
@JohnOCFII  - by /etc/network, did you mean /etc/dhcp/dhcpd.conf ?

Try disconnecting from the stratux network (i.e. connect to your home wifi), then connect back to stratux.

JohnOCFII, 2015-10-08:
> @JohnOCFII - by /etc/network, did you mean /etc/dhcp/dhcpd.conf ?

Nope, I changed /etc/network, changing the default route for the pi itself - which (really, as expected when I think about it) had no impact.  I'll try changing the DHCP configuration tonight.

dmurray14, 2015-10-08:
In my testing on the previous version, the wifi icon is present when the
device can access the internet through wifi. I'm not quite sure how it
figures that out (maybe using the captive portal detection). That said,
it's important to leave the DHCP-issued gateway as 0.0.0.0, as this
instructs the iOS device not to use the wifi as a default route, which
allows cellular data to continue working for devices which have it. If you
issue a valid gateway (like 192.168.10.1), the device will lose internet
access for the duration it is connected to the wifi network.

cyoung, 2015-10-08:
Yes that won't change in the releases, what I'm getting at is "how to get iOS to believe that it is connected without waiting the long (~2 minute) timeout period?"

Even though it connects in the same amount of time as before, a lot of users are reporting that it takes a long time to connect. It's mostly because of the false signals given by the Wi-Fi icon now in iOS9 and the differences in previous iOS versions.

AvSquirrel, 2015-10-08:
@dmurray14 , @cyoung -- no disagreement from me on keeping the 0.0.0.0 gateway. I like being able to use cellular data while connected to Stratux.

RE: captive portal detection, [this page](http://serverfault.com/questions/679393/captive-portal-popups-the-definitive-guide) suggests that iOS 8/9 attempts to connect to  http://www.apple.com/library/test/success.html or http://captive.apple.com/hotspot-detect.html with the user agent 'wispr' . If that request returns the string "Success" then iOS treats the connection as a good, open internet connection. No response triggers "no network" Other responses would trigger the captive portal popup.

So... how to fake a connection to the Apple network detection pages?

rgutlon, 2015-10-08:
>... a lot of users are reporting that it takes a long time to connect

That's the perception leading some folks to think the system is not working ... when its simply the connection delay.

I just timed it iPad iOS 9.0.1: 69 seconds to connect

bradanlane, 2015-10-13:
Since I don't need cellular service in flight, I'll set my gateway to 192.168.10.1 and see how it performs.

rgutlon, 2015-10-13:
@bradanlane curious if you noticed similar behavior. After changing the gateway to 192.168.10.1, rebooting / reconnecting to stratux, I noticed the WiFi activity symbol regularly "spin" within the application (FLYQ), where this did not occur when the gateway was 0.0.0.0. A wild guess, but could the 192.168.10.1 route *confuse* an app into thinking a data connection is available?

ChevyFlyer, 2015-10-14:
Running iOS 9.0.2 and FF 7.3.2, both dongle antennas and the USB RY835AI, stratux v04r2

Today 1.7 hours flight time, had to reboot both the iPad and the Pi.  Flew to JXN and back to 57D with known weather from ground stations before takeoff.  No radar services in the air till after the reboots.  No local echoes appear on FF and I can see the rain coming down out the window. I was able to see echoes over towards Pennsylvania but that was 500 miles away. The echoes were very rough also, large pixels.

Flew around the rain on the way back and saw some traffic near me in FF, no visual by eye, they were above the clouds and I was below at 2000'.

The GPS works on the usb but I will be wiring it later this week.

JohnOCFII, 2015-10-14:
Noticed today while in a Starbucks that iOS 9 generates a user-friendly message if it sees you on a network that can't connect to the Internet.  This happened (in that case) because of their portal, which I had not yet authenticated to.  Not sure if this is related to the WiFi icon showing or not -- or if there is more knowledge in this area to assist.
[image]

ChevyFlyer, 2015-10-19:
Running iOS 9.0.2 and FF 7.3.2, both dongle antennas and the USB RY835AI, stratux v04r2

2.7 hours of flight today and stratux was connected to both FF and Avare on my MotoX phone. One of the airplanes has ADS-B out and I was able to follow him back to 57D from KGDW. I was trailing him by 10 miles.

I did a fresh load of v04r2 on the sd card, no issues today.  Saw traffic and weather and the GPS led me directly to the destinations. using "sudo shutdown -h now" before pulling the power plug now.

cyoung, 2015-10-19:
@ChevyFlyer - Ron, I went flying for a couple hours today and reproduced what I think was your original issue. It has to do with the RY835AI and related to #63. Being worked on. If it comes up again, reboot THEN turn off AHRS/GPS via web interface - AFAIK v0.4r2 is solid with AHRS/GPS disabled.

bradanlane, 2015-10-19:
Thanks for the instruction @cyoung. I observed #63 the other day when running some ground tests. I'll be flying tomorrow and will keep an eye on the WebUI. If I see the updates stop, I'll do the reboot and disable GPS / AHRS.

ChevyFlyer, 2015-10-19:
Just posted on iOS9 testing #76

Flew today with GPS usb, no AHRS, no issues, two connections to stratux,
iPad and android.

cyoung, 2015-10-19:
http://www.apple.com/library/test/success.html
http://captive.apple.com/hotspot-detect.html

Pushing DNS to the iOS client to emulate a successful connection to show the "connected" icon could be done here. Then there's an issue when the user wants to use cell data. There is a solution but it's not worth pursuing now just to get the WiFi icon on the top left. Users will just have to be aware that iOS9 will not show the icon even when connected.

closed 2015-10-19 by cyoung