cyoung/stratux #77: v0.4r2 missing a GB?

Ergonomicmike, 2015-10-08:
First, thanks for this fantastic project. Now have wx in the Glasair.

I just downloaded v0.4r2. I notice its .img is 1GB smaller than previous releases, even tho the zip file is same as others. (I dnld'd & decompressed it twice.) It boots quickly, but it never makes the wireless WiFi blink. No stratux SSID either. Is it just me or is there something missing from the image?

cyoung, 2015-10-08:
There's something wrong with it, there was a problem with the upload it appears. Github afaik only has the web interface which takes some time to upload with. I'm re-uploading the zip now, looks like 10-15 mins more.

Ergonomicmike, 2015-10-08:
Thx. Whew - glad to know it wasn't me. Surprising that 7zip unzipped it as an image. (Serves me right for not checking hash.)

cyoung, 2015-10-08:
Alright, try again.

Ergonomicmike, 2015-10-08:
Yep - boots and connects. I can't get any towers from the couch, so I can't report whether it stays connected now, but I did have the uptime disparity in flight a few days ago.

I probably posted in the wrong forum for this kind of report. (Should have been in reddit instead.) My first time on github. I didn't mean to make an Issue (# 75) about it. So you should probably kill this thread.

Thanks again for bringing wx to our cockpit for less than the cost to fill the tanks.

Ergonomicmike, 2015-10-08:
Correction: Issue #77.

closed 2015-10-08 by Ergonomicmike