cyoung/stratux #78: Request: Read-Only Filesystem

ssokol, 2015-10-09:
I know this has been discussed ad nauseum on previous issue, but I've seen enough requests that it's probably worth collecting thoughts in one place.

## Background

The Raspberry Pi is not a traditional embedded system. It's a full Linux computer system on a very small board. Like all Linux systems, the Pi normally requires an orderly shutdown to prevent data loss and filesystem corruption. If the power is pulled in the middle of a write operation, bad things (tm) can happen, including a complete corruption of the system. This is always painful, but it is particularly painful in an avionics system.

## Stratux Read Only Requirements

1. Store DHCP lease information. This data *does not* need to be persistent between reboots.
1. Store the stratux.conf configuration data. This data *does* need to be persistent between reboots.
1. Store log files when logging is enabled. This data *may* need to be persistent between reboots.

[Please add other requirements here.]

## Strategies For Read Only Operation

1. Remove unnecessary stuff that tries to write to the filesystem:
  1. dphys-swapfile
  1. other packages?
1. Replace the default syslog system with the BusyBox syslogd which writes to memory
1. Use tmpfs (ramdisk) for:
  1. /tmp
  1. /var/tmp
  1. /var/log
  1. /var/lib/dhcp
1. Disable fsck by appending "fastboot" to the /boot/cmdline.txt file
1. Disable swap by appending "noswap" to the /boot/cmdline.txt file
1. Force read-only file access by appending "ro" to the /boot/cmdline.txt file
1. Add an "ro" and an "rw" script to "/usr/bin" to allow advanced users to easily switch between read-only and read/write modes.

That handles the basics. I've done some experiments with this and it seems to work with a small tweak to the isc-dhcp-server init file (have to add "touch /var/lib/dhcp/dhcp.leases" before you start up the server).

What this doesn't handle are:

1. Creation of and edits to /etc/stratus.conf made by the UI
1. Log file persistence

My personal thought on that is to create an additional small partition as part of the image. Something like 500 mb to 1 gb. Format it with something that can be read by any computer (i.e. FAT32). When Stratux boots up it checks to see if it can mount this partition. If so, great - it mounts it and uses it for the configuration file and for logging. If it can't mount / read / use the filesystem it uses the format command to recreate it. It then writes a default stratux.conf and alerts the user through the GUI that the configuration was recreated and needs some attention. (The stratux.conf config data can easily be cached on the browser in localStorage, making replacement trivial.)

A variation on this theme:

Mount the config / log partition read only at boot. Remount it read/write when necessary to update the configuration or to copy log data from memory to the filesystem. Have a log rotation feature built into Stratux that handles this so that every minute or so the log data gets saved. This minimizes the chance of data loss.

## Experience

I've tested the first part of this scenario and it seems to work. I have yet to try the additional partition. I'm not sure how much complexity that adds to the spindle configuration that is currently being used to build images.

I will attach the scripts that I've been using to this issue in case anyone wants to try it out.

cyoung, 2015-10-09:
Yes, please attach your scripts

ssokol, 2015-10-09:
Hmmm... Github does not seem to like tar files or shell scripts. Here's a link to the archive on my server:

https://www.stevensokol.com/stratux/readonlypi.tar.gz

bkwny, 2015-12-29:
Thank you for everything you folks have done so far.  I think your project holds a lot of promise, have built one, and will begin trials shortly.  It's my opinion this read-only-root requirement is the single most important requirement out there right now.  As much as we like to think of this device as a computing system operating as an ADS-B receiver, it's my opinion the pilot community will think of it as merely an appliance.  Turn it on and it works, and turn it off whenever you like.  As long as there is not a hardware failure of some kind, when you turn it on, it works.  I can't imagine being in the middle of the trip and having to reimage my SD card because I happened to turn off the appliance at an inopportune moment.  Thanks again for all you have done.  Brian.

Ergonomicmike, 2015-12-30:
Apparently implementing a RO file system is not as easy as it sounds? Else it would have been done back in October? (The new Shutdown feature mitigates most of this issue.) How about this for an option (a variation of @ssokol's model): For those who are geeky enough to know how to use gparted, say, on a live Linux CD, to create a Linux swap partition on their imaged SD card, the Stratux sees the swap, goes RO on the first, and uses the swap for logs, swap etc. If swap sets a dirty flag, Linux fixes it on boot. For those who aren't geeky enough, they take their chances with uncommanded shutdown, and are no worse off than now. (Brainstorming by a noob.)

skypuppy, 2015-12-30:
Not sure but I think that the swap partition is not used like a normal
partition.  One cannot write files to it or read files from it; it
belongs only to the kernel.
However, a separate partition just for the log files could be done and
we'd have to do a flush after every single write but that's not a very
big performance issue.
The largest problem with non-graceful shutdowns is that the OS (Linux)
kernel writes stuff all over the place as a normal part of it's
"operating" and *these* files are what can cripple a system.  It
crippled a desktop of mine, wiping out an X11 file due to a power outage
and I *still* can't rebuild X properly.  It is my main computer so there
is tons of stuff on it I cannot lose.  Due to mismanagement (and
cheapness!) on my part, it is the only computer in the house that did
not have a backup power supply (UPS.)  That is fixed now, but the horse
has already left the barn.
I have no idea what the complications are if the entire OS was in a
read-only partition.  Unfortunately, the Pi2 does not have enough RAM to
run as a diskless station and have enough room for stratux or anything
else (assuming it would run at all.)
The only solution I can see is it's own little (maybe a coin?) battery
that handles graceful shutdowns when external power is removed.  And
that ain't easy.  Cheap, but not easy.  A huge plus is that coin battery
could also maintain the system time and date clock, making other duties
less onerous.

Ergonomicmike, 2015-12-30:
I knew that I would probably embarrass myself by saying something stupid (re: swap). But since @skypuppy didn't shoot the whole idea down, how about if geeky users created an ext4 partition with a specific label that Stratux looked for? So if it saw "for_ro," Stratux would appropriate that for RO operation?

I did a cusory search for "keep alive" on the Pi. Someone had a hack-a-day, where two 9v batteries and a few components did a graceful shutdown on power loss. That wasn't a plug n play, off the shelf solution. So I doubt most users would do it. It's easier to carry a spare SD card with an image on it in case of catastophic corruption.

ssokol, 2015-12-30:
I've played with several shutdown circuits and found them all to be a PITA
to work with. They also add a good bit to the cost of a Stratux system. I
still think the best option is to make the root partition read-only.

I'm pretty sure that building a read-only system with a separate read/write
partition for updates, logs, and configuration is reasonably trivial. Given
that we're distributing a multi-partition image, it wouldn't take much to
add another FAT32 partition that, if corrupted, could be re-created by
Stratux. Under most circumstances all it would contain would be the
stratux.conf file. If you enable logging, logs would be stored there. If
you want to update the firmware, the upload would land there.

I linked a script that converted a running Stratux to RO. We just need to
take those same mods and build them into the build process.

@cyoung - any idea how hard it would be to either have the spindle process
create the new partition, or to move away from spindle and use some other
means of re-enabling the first-boot stuff (keygen, MAC generation, etc.)?

cyoung, 2015-12-30:
Probably best to do some post-processing on the image, mount it and modify the startup scripts.

skypuppy, 2015-12-30:
@Ergonomicmike:
1. There is nothing embarrassing about ideas.  This is a place for
discussions and ideas.
2. It is not up to me to yay or nay anyone's ideas.  I just spit out the
facts (and sometimes my conjectures) as I know them.
3. I'm not always correct.

Skypuppy

jamesshannon, 2016-01-05:
It seems to me that there are two types of files here with very different requirements:

Logs -- Not a big deal if you lose them (ie, the partition is corrupt and you reformat it).
Config -- Bigger deal. I predict that there'll be config scope creep over time, and people will dial in the config just the way they want it. Losing this in a corruption is more painful. I like the localStorage hack, but that'll require you to realize that something is amiss, reconnect from the iPad (hoping that it's the same iPad, etc).

As somebody said, in the longer term of stratux, a lot of pilots will see this as an appliance, not a computer. That means the logs loss is even less painful, and the config loss is even more confusing / painful (assuming they're not using the standard config).

So, I think the best answer is a combination of what's been proposed. The config file goes on a RO partition which is remounted just on updates (which will be infrequent, and unlikely to happen at the same time you're pulling the plug) while the logs can go on the more expendable partition.

1-prop-head, 2016-01-05:
Implementation of a method to protect highly customized configuration informaton rimes with config file backup-check-restore.  Perhaps the boot partition is a place to put a backup copy of sensitive config information to compare and restore is necessary.  I also thinik the web UI could provide a way to move configurtion information in the form of a text file (ini or xml) and keep externally not unlike a router backup.
As I recall Unix has substantial file security built in and moving some of the application outside of the root will add some complexity but may limit some potential fle system corruption. 
Usage of ipcs for interprocess communications is certainly a way to bridge communication between the less secure UI and the underlying priveledged processes.
Then the tuning using nice() to put the hardware where the need is can be benificial as well

1-prop-head, 2016-01-05:
As a user if you find that your setup is golden and you want to always get back you can copy the micro SD card image to an identical SD card as your backup (perrhaps 2 backup SD cards). Then you carry this backup in the plane and should trouble occure you simply swap the SD card and you are back as before.
Check out dd() on your RPi, Linux or Apple MAC to make the copy.

bkwny, 2016-01-05:
Instead of ext4, does one of the journaled file systems hold promise?  Is there a Linux file system somewhere that has the property that no matter when the disk writing stops, the file system is left in such a state that the committed files have integrity?  Losing the last update is not nearly as big a problem as losing the integrity of the file system.

kdknigga, 2016-01-05:
Ext4 is journaled.  As is ext3.

The purpose of all journaling file systems is to be able to make the file
system consistent after an interruption.

skypuppy, 2016-01-05:
iIRC, ext4 IS a journaling file system.  None of them are perfect. 
That's what makes graceful shutdowns so important.

yihchun, 2016-02-20:
Two suggestions on this approach:

1. Combine everything into a single file. One script that runs on the pi and sets everything up seems straightforward. Many of the accessory files could be integrated into a "echo ... > ..." or "patch << ..." This single script could possibly even be loaded through the new system update functionality in 0.7b2. Furthermore, if the script is python or something else that github tolerates, it could be checked in.

2. It'd be great to have an rw partition set as noauto mount, maybe mounted on /rw with soft-links from /tmp, /var/tmp, /var/log, and /var/lib/dhcp. Early during startup, a script can run fsck, and in case of error codes 8 and higher, the script can overwrite the old filesystem with mkfs.

cyoung, 2016-02-20:
Eventually there won't be any configuration options at all. Why would
anyone want to plug in a GPS unit and not use it? Or an extra SDR and not
use it? These are appropriate for a developer but not needed to the "user".

bkwny, 2016-02-20:
If nothing else, people need to be able to configure the wireless access point.

skypuppy, 2016-02-20:
I imagine most pilots want to just plug and go, not fiddle with things. 
Unlike geeks like us!  :)  :)
But sometimes you just can't do that with new technology.

Skypuppy

skypuppy, 2016-02-20:
> Are you referring to as an upgrade mechanism? Otherwise I'm not sure
> why you'd care about an access point. Im still of the opinion that the
> traditional appliance method works best (remove SD card and reimage
> than install). Works well for hand held GPS, Garmin panels etc.

It is a more convenient when YOU don't have to burn the SD card
yourself, resize it, verify operation, etc.  :)

For developers AND users, it is much more convenient to simply upload
the changed program (probably faster overall, too,) rather than the
entire 3+ Gb file transfer.  (Cheaper for the maintainers as well, those
who have caps on their monthly outgo or who have to pay by the megabyte.)

Skypuppy

bkwny, 2016-02-21:
Regarding the wireless access point or WAP, I'm referring to the idea that how Stratux operates the WAP has an impact on the security of Stratux, and on the performance of Stratux, and on the ability of the Stratux to coexist peacefully with its neighbors.

For the first, running the WAP in encrypted mode (e.g., WPA-2) helps the owner to secure the Stratux against unauthorized use or tampering.  For the second, because every user who binds to the WAP immediately creates a UDP broadcast burden for Stratux, being able to manage the WAP for security improves Stratux performance.  To these ends the owner needs a configuration tool that lets him set the WAP encryption mode and encryption key.  For the third, the SSID and wireless channel have a big impact on how well the Stratux coexists with whatever other technology might be deployed nearby.  To that end the owner needs a configuration tool that lets him set the SSID and the channel number.

We cannot assume the Stratux will always be deployed in a friendly environment where only one or two nonhostile devices will bind to the WAP.  We also cannot assume the environment will be free of other WiFi devices, so it's perfectly fine to pick SSID "Stratux" and channel 1.  For all of these reasons a successful Stratux will never be absolutely free of a configurator.

Like many other computing devices, the success of Stratux will probably depend more upon its owner's ability to manage, secure, and maintain it than upon the whiz-bang technology at its core.

RobertGary1, 2016-02-21:
That kind of argues for two sd cards. One for the harddrive and one for updates. I don't see pulling my Stratux out of the plane very often nor needing to update it often. 
But I agree that there really shouldn't be any user config data worth saving on such an appliance. 

-Robert

cyoung, 2016-02-21:
@bkwny - have you tried a WPA-2 config on Stratux? AFAIK none of this has been tested yet.

jzeevi, 2016-02-21:
Why would anyone need to secure a Stratux? In the air, there’s hardly any chance of another plane “camping on”. On the ground? If in a plane, again, hardly any chance. And if someone does, what can they “steal”? Even if they change the settings, that can readily be reset – and, as the devs move to a static config, there’s even less chance that something of merit can be changed. 

 

Can an injection attack be implemented? Sure – opens source pretty much allows anyone the ability to figure out the upload mechanism and use that as an attack vector. But, more to the point, so what? The absolute worst case is that one has to replace or re-image the microSD card. And, if the Stratux isn’t working, so what? It’s only a (fabulous) tool for assisting flight. Every pilot that uses one had darned well better be able to fly without it since it’s not on the practical test standards.

 

I  don’t see the use case that demands WiFi security.

 

There are many little bitty things that are still TDB to make this the great product it has the potential to be (Go Chris!). This is not one of them.

yihchun, 2016-02-21:
@cyoung I'm not using a standard setup, but I have WPA-2 PSK working on a RT5370 by updating hostapd, changing the driver to nl80211, and adding the following to /etc/hostapd/hostapd.conf:
`auth_algs=1`
`wpa=2`
`wpa_passphrase=(something clever)`
`wpa_key_mgmt=WPA-PSK`
`wpa_pairwise=TKIP`
`rsn_pairwise=CCMP`

bkwny, 2016-02-21:
@cyoung  Yes, I have tried WPA-2.  My experience has been that about half the time my iPad won't bind to the WAP.  When the bind fails, I get a message along the lines of "Failure to join" and then a place called "Dismiss" I can press.  If I reboot the Stratux enough times I can get it to work.  Here is what I add to /etc/hostapd/hostapd.conf.  Ref:  https://wireless.wiki.kernel.org/en/users/documentation/hostapd.

macaddr_acl=0
auth_algs=1
wpa=2
wpa_passphrase=abc4xyz
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

skypuppy, 2016-02-21:
I somewhat agree, Josef, but it still bears consideration. 
Like most "features," it isn't necessary for basic stratux operation
*but* it should be considered long run if it doesn't take a lot of
developer time.  Just my opinion (and I have a LOT of those!)  :) :)

Skypuppy

bradanlane, 2016-02-24:
OK, so I have a short script that makes Stratux read-only. It does not install any new packages. It consists of a small number of required changes and a set of optional changes.

Required Changes:
* update **/boot/cmdline.txt** with `fastboot noswap ro`
* update **/etc/fstab** to create a few `tmpfs` locations
* update /etc/init.d/isc-dhcp-server to generate a starter `dhcpd.leases`
* add `rpi-ro` and `rpi-rw` helper scripts so a developer can toggle the system to make changes, code, etc.

Optional Changes:
* add **/etc/environment* file to address block id table
* update **/etc/cron.hourly/fake-hwclock** to temporarily switch to RW to save the time and then switch back to RO
* update **/etc/cron.daily/logrotate** to save state info in the log directory
* update **/etc/init.d/hwclock.sh** so its test of `/etc/adjtime` will work

I have a script that runs all of the above changes.

Since I am not clear on the ssh keys that are generated during the spindle process, my script assume it is run after Stratux is booted the first time. This may not be an issue and the script changes may bve easily incorporated into spindle (_I'm just not clear on that issue_).

lastdeadmouse, 2016-04-04:
Sorry I've been on a hiatus for a while. Been running RO since the beginning. Finally did a new build based on the .8r2 image using unionfs-fuse for temp writing, and have it working great. Why go through the trouble (and risk) of writing logs and settings to a separate partition?  We already have the ability to download logs through the web interface which, though limiting in certain situations, is good enough for most issues. As for setting changes, I'd say it's preferable to add a web interface "save" button that calls a script to remount rw, write changes, and remount ro.

As it stands, I'm happy with my setup, and I don't mind remounting and making my changes manually, as I rarely need to make changes after my initial setup. I just think making the default image RO is often being made more complicated than need be.

bradanlane, 2016-04-04:
I agree with most of your comments regarding RO operations. I set it up without unionfs-fuse. I do toggle to RW to save settings changes and then immediately toggle back to RO.

peepsnet, 2016-04-04:
@lastdeadmouse How do you do this?? What is this process??

lastdeadmouse, 2016-04-05:
Install unionfs-fuse, replace syslog with busybox, create symlinks for things directories that may change on the fly, and unionfs entries to fstab/make the sd devs read only, and add kernel options. I didn't use a tutorial, and I left my stratux in the airplane so this is kind of off the top of my head, lol.  The closest tutorial I can find is in Italian, but can be found at:
http://guglio.net/raspberrypi-in-readonly/

Someone also had a read only fork with pretty good documentation, but finding it quickly is proving difficult.

Ergonomicmike, 2016-04-18:
I think this is an obvious comment, but I'll make it anyway. Seems to this dummy that jpoirier's stratux-setup script would be an easy way to implement ssokol's RO script.

bradanlane, 2016-04-18:
I seem to have missed it. Could someone Point me to @jpoirier 's script. I have one of my own I use to make Stratux RO. I'd be curious to compare.

jpoirier, 2016-04-18:
@bradanlane   https://github.com/jpoirier/stratux-setup

bradanlane, 2016-04-18:
I'm missing something. I can't find where the setup script makes the raspberry Pi "read only".

jpoirier, 2016-04-18:
It doesn't. The comment only stated that it'd be an easy way to implement the RO functionality to a setup.

cyoung, 2016-12-01:
10k+ or so devices so far and issues relating to microSD cards are mostly people bending them and breaking the card internally. Anything else has not been above noise level, so this is definitely not a priority even now. As far as I've seen, the read only implementations out there have had a higher failure rate than mainline RW stratux.

closed 2016-12-01 by cyoung

bradanlane, 2017-08-08:
Just an FYI / BTW ...

It's been more than a year since I was working on RO implementations and recently was working on a different project. It needed RO since it was always battery powered and had the probability of the batteries running down.

Ultimately I used an implementation based on [this blog post](http://wiki.psuter.ch/doku.php?id=solve_raspbian_sd_card_corruption_issues_with_read-only_mounted_root_partition). 

The "good" is that it was very easy to use and everything in the project "just worked" - eg, hotspot AP and home network toggling, etc. The "bad" was it doesn't provide a hot-toggle capability so switching ot RW required a reboot. I suspect there is a way to address this pain-point but it was not an issue since the project config was written to a different device than the OS and project program.