cyoung/stratux #79: Support two PPM if there are two SDRs

bradanlane, 2015-10-10:
I've run the Kalibrate utility against the two SDRs in my setup. I know have measurably different results for the two SDRs.

If others are seeing a similar result then should we allow users to set PPM for each SRR.

I suspect we only set it in the first since that is the one used for UAT and thus more critical - correct?

If we feel PPM will not provide an improvement on the 1090ES receive, then the resolution is to improve the help text for PPM to explain this to the user.

224XS, 2015-10-10:
I have two different model r820T sdrs, one older, the other the new version. Both seem to produce the same identifying data tags.  Is there a way to bind the respective frequency offsets to their proper SDR?

bradanlane, 2015-10-10:
It's actually worse that that. I've tested older R802T units and newer R802T2 units and they all report the exact same signatures.

ssokol, 2015-10-10:
The best option for this is probably to use the physical port, mapped
through udev. Here's a post on it:

http://askubuntu.com/questions/49910/how-to-distinguish-between-identical-usb-to-serial-adapters

tbirdrv, 2015-10-12:
Has anyone looked at [rtl-eeprom](http://manpages.ubuntu.com/manpages/trusty/man1/rtl_eeprom.1.html)? 

Would it be possible to encode a unique name or ID to each SDR? I'm not even sure this program works with our SDRs.

jpoirier, 2015-10-12:
Not sure if it's the right approach but rtl-eeprom is in librtlsdr
therefore it's available in gortlsdr via WriteEeprom.

cyoung, 2015-10-19:
https://github.com/osmocom/rtl-sdr/blob/master/src/rtl_eeprom.c

@jpoirier - do you think a "set/get serial string" function belongs in gortlsdr?

jpoirier, 2015-10-19:
@cyoung I hadn't looked at the rtl_eeprom.c file until just now. Having
similar functionality in gortlsdr would be useful. I can update the package
later this evening unless you or someone else wants to start on it now.

cyoung, 2015-10-19:
Working on a few other things now, if you could do that it would be really great - will implement the sdr.go end in stratux such that it makes use of it.

jpoirier, 2015-10-19:
Okay, will do.

loather, 2015-10-20:
fantastic. I suggested something like this on reddit a couple days ago; glad to see it's being implemented. I manually set the serial numbers of my units to 978 and 1090 and investigated the possibility of modifying the code myself.

jpoirier, 2015-10-20:
Added GetHwInfo() and SetHwInfo() methods, see the example file for usage. It's simpler and safer to just read all the info at once (there isn't that much) and modify a specific item/s then write the entire info struct.

E.g. to change the serial info:
info, err := dev.GetHwInfo()
info.Serial = "978"
err = dev.SetHwInfo(info)

jpoirier, 2015-10-20:
@cyoung if you think you'll need/want finer grained control just let me know.

bradanlane, 2015-10-20:
I'm getting a compile error for missing GetHwInfo / SetHwInfo. Do I need to manually 'make' something?

jpoirier, 2015-10-20:
What are you compiling? Did you update the package via go get -u
github.com/jpoirier/gortlsdr

Once the package is updated you can cd $GOPATH/src/
github.com/jpoirier/gortlsdr and go build rtlsdr_example.go

I just double checked go get on the package and built the example without
any problems.

cyoung, 2015-10-20:
@jpoirier - looks good. The current release from earlier today is using this now. It's not automatically setting the serial number and marking any SDR, right now the user has to use rtl_eeprom. But it's reading and looking in the serial strings now for people that are able to use rtl_eeprom and want to fix an SDR to a certain use.

yihchun, 2016-02-13:
I've checked in a possible solution to this problem. See #248.

ayarbro, 2016-03-12:
Hey all, first I appreciate this feature a ton!  So thank you.  I was wondering if it'd be possible to add the sdr serial next to the uat/1090 messages display on the status page, so that we know which sdr is receiving.  I'm trying to figure out why my sdrs are hooking up to the wrong dump, and this would make things go faster for troubleshooting.

peepsnet, 2016-04-23:
Could this issue be closed or are we looking for a way to add this to the webUI??

skypuppy, 2016-04-26:
I finally broke down and did this:
rtl_eeprom -d 0 -s stx:978:34
because of the tuned dMurray antennas.  It was not required with the stock antennas.  So, the inevitable questions:

1. is there an equivalent suggested string for 1090?  And,

2. If the "34" ppm value is used by stratux, it seems logical that the ppm value will differ for each and every radio.  Is there some method (kalibrate?) to obtain the optimum value after warmup?  Btw, the top of the 978 nano2 read a max of 109F (I'm sure that varies over time) with no bits being read (too far from tower.)

JohnOCFII, 2016-04-26:
Here's a description of using the kalibrate process that gives some better background info: http://314256.blogspot.com/2015/03/how-to-use-kal-software-to-workout-ppm.html

skypuppy, 2016-04-26:
Now this is odd.  The Nano2, using kal, with dMurray 978 antenna finds two GSM850 channels, one with power reading of 185242 and the other 313107, while the DVB-T (big blue case) with 1090 antenna sees none at all.  The DVB-T radio sees lots of 1090 traffic in stratux.

Axtel4, 2016-04-26:
> Now this is odd. 

It may not be. Try the 978 antenna on the Blue dongle and see if you get any GSM850 activity.

cyoung, 2018-01-17:
PPM setting now hidden except in developer mode.

Will likely remove the setting as it's rarely used and not necessary.

Helno, 2019-01-17:
The PPM values can be set using the SDR serial Script that @peepsnet wrote. It is included so it covers this issue for those that want to fine tune both SDR's.

https://github.com/cyoung/stratux/wiki/SDR-Serials

closed 2019-01-17 by Helno