cyoung/stratux #82: How are Logs going to be used?

bradanlane, 2015-10-16:
The last tab in the WebUI with nothing yet created is the **Logs** tab. 

How do we see logs being used and what would be most useful to enable for users to assist with that use?

cyoung, 2015-10-16:
Well, the only human readable data really is in stratux.log. And people will mostly be looking for errors there. So maybe some highlighting for error log entries. Then we can do the same (in the future) for the system logs based on some string matches or regex for common system errors.

eolinger, 2015-10-19:
Maybe this is obvious but a user friendly way to download the logs in order to submit them with bug reports (possibly even zip them for download).  

Size and last update time would be good, along with a way to truncate them for space.

bradanlane, 2015-10-20:
OK. I just ran a few tests. If we want to have some form of "submit logs", it is possible to collect the necessary log data; gzip it; base64 encode it; send it using a **mailto** link that uses the device native mail client. It get's queued and will send once the user gets network connectivity.

It does launch the mail client so the user sees it is going out as mail. They also have the option to add any text they want to the head of the email.

I generated the correct href for the link with a target email address, subject line, and the base64 text block as the body of the message.

If we want this ability, we would want to manage the size of the content. I did my test with 115K log file and ended up with a 15K base64 encoded block of text. We would also need an address to mail these to.

cyoung, 2015-10-20:
logs@stratux.me

It's just forwarding to me for now, will set up a mailbox and something to parse received logs when it gets to a certain level.

bradanlane, 2015-10-20:
OK, I'll code this up. Thanks.

ssokol, 2015-10-21:
Just a random thought: the primary log for realtime debugging is stratux.log. It would be great if the data being written to the log file could also be subscribed to over the websocket and displayed in the logging tab. It would avoid the need to SSH in and tail the actual file. Not sure how hard that is to do in go.

cyoung, 2015-12-28:
@bradanlane - how'd that log sender thing work out? Or did you decide against it?

cyoung, 2016-03-08:
Duplicate #228.

closed 2016-03-08 by cyoung