cyoung/stratux #84: Web interface: ModeS not being saved

cyoung, 2015-10-18:
As reported by a user:

https://www.reddit.com/r/stratux/comments/3p0js9/how_to_use_foreflight_ownship/cw38i5x

"I spent some time with this issue this morning. I'm using v0.4r2, with no GPS connected to the Stratux. The Ipad 2 is getting GPS data from an external GPS (Garmin GLO)
Preamble - I haven't used unix in 25+ years so I have no idea what I"m actually doing :)
I edited the stratux.conf file using the following commands (I put sudo in front of every command): kill -9 xxxx (gdl90 process) kill -9 xxxx (dump 1090 process) nano stratux.conf (put in the hex code, lowercase letters a4313a) poweroff, the poweron
When I restart the Stratux and launch the web interface (using the "Stratux" icon) - I see that for some reason the 1090 mode switch is off.
This is happening even throgh the ES_enabled is set to TRUE in the .CONF file
Sliding the switch to "ON" removes my Mode S code from the stratux.conf file.
It appears that the Web Interface doesn't like having a text value in the OwnshipModeS field. It further appears that the Web Interface rewrites the OwnshipModeS field to zero.
I re-editited the ModeS field and will avoide using the Web Interface to see if Foreflight will recognize it in the air.
"

bradanlane, 2015-10-18:
After re-reading this, the behavior does not surprise me. The code assumes the mode-S value is uppercase because it insures it is upper case before storing it. Since the user edited the config file directly and then also used the web settings, the mode-S didn't match up. 

I change the validation code handle this case.

gribbled, 2015-10-20:
I'm the original reporter for this issue.

I just checked and the behavior still exists and seems incorrect.   I edited the stratus.conf file to include "OwnshipModeS:A4313A"   After restarting I used the web interface.  The 1090 switch was set to off (incorrect, it was ON before and ES_Enabled:true is in the .conf file.)

Next, after setting the switch to on, the OwnshipModeS was reset to 0 in the .conf file.

Hope this helps

PS - I'd love to add a Feature Request to have the Mode S be editable in the Web Interface.  Is that allowed?

cyoung, 2015-10-20:
Confused - ModeS is editable from the web interface, no?

gribbled, 2015-10-20:
Not that I can find.   Not in Settings

bradanlane, 2015-10-20:
The "mode S" for own ship is on the settings page. However, it's relatively new code so I do not believe it is in 0.4r2 pre-built image.

gribbled, 2015-10-20:
Thanks - that explains it.  Can't wait to try it (wish I was smart enough to set up a build environment).  Thanks to both of you for the excellent work.

bradanlane, 2015-10-20:
I'm going to mark this as "no need to fix" since the webUI is now available with 0.4r3.

closed 2015-10-20 by bradanlane