cyoung/stratux #85: Weather "age" is incorrect

bradanlane, 2015-10-20:
I flight tested the code base from three days ago which includes color coded weather and the update to indicate how old a wether message is (rather than displaying a GMT timestamp).

Weather was displaying 2 days, 20 hours old in the WebUI.

rgutlon, 2015-10-21:
Is this due to the Pi not picking up the correct time? Or are you able to account for that in the raw data?  I recall there was a fair amount of discussion on being able to pass the "time" in Issue **#33**

bradanlane, 2015-10-21:
At this point I have no clue. I've not looked into it. Since I will have to debug this with playback data I'm going to have to first figure out what timestamps are available to me and what they originally meant when they were recorded.

rgutlon, 2015-10-26:
More than likely the Pi is starting up with the incorrect time (w/o access to the 'net) unless the user manually sets it ... or a script is run to pull the time from a GPS. Found a number of articles on this ... a couple of links below that might be of of help (programmatically):

>**Q**: I've got a USB GPS receiver permanently attached to my raspberry pi. At boot time, I would like raspbian to wait until the GPS has a signal, then set the OS's time to whatever the GPS says.

>**A**: There's two "official" ways to do it.
> gpsd can set the clock
> ntp has a special driver 127.127.28.0
>The benefit of gpsd is that it gives you a nice interface to the GPS data for other purposes. If you use the ntp driver you've tied up your GPS receiver:

Raspberry Pi: Set time from GPS dongle
http://blog.petrilopia.net/linux/raspberry-pi-set-time-gps-dongle/

>I would like to set the Raspberry PI's time clock on start up or once the GPS has a lock because the application this will be used on will not have access to the internet.

Script that sets system time through a GPS
https://code.google.com/p/gpstime/

bradanlane, 2015-10-29:
I'm not sure what the answer is to get the correct "age" of a weather item. I have checked both the gen_gdl90 code and the WebUI. The UI calculate the age of weather by taking the difference between the two date/time values from gen_gdl90: one is the timestamp for the item and the othe is when Stratux received the item.

For some reason, this delta is huge: aka more than a day old.

skypuppy, 2015-10-29:
arithmetic error somewhere?  Or data type conversion error maybe?

bradanlane, 2015-11-06:
since I can not debug and fly at the same time I'll make a temporary build that dumps the various calculated and received datetime data so I can determine where the delta is coming from.

rgutlon, 2015-11-06:
I'm scheduled to fly a couple of 100 mile legs on Sunday ... if I can be of any help let me know.

bradanlane, 2015-11-06:
I did a series of tests and the issue comes back to not having a reasonable system time from the raspberry pi. I powered up my system, which has been sitting for a few days, and it started with with the date from November 1st. All of the weather reports were then calculated based on this reference.

I've go ahead and changed the code back to just showing the short date/time of the message. Down the road, if we can always get some sort of useful date/time as a reference point, then the 'age' format for weather can be revisited. I will close this issue and we can either leave it or have an "enhancement request" for somewhere down the road.

rgutlon, 2015-11-07:
Probably not as elegant as you'd like, but in the short term, how 'bout giving the user the ability to enter in their local time via the UI?

bradanlane, 2015-11-07:
silly me. The problem is only with the WebUI so I can ignore the base time that is sent with the weather messages from the raspberry pi and just use the mobile device's time.

The fix is pretty small (weather.js) I'll check in the code.

**Can someone test it this weekend?**

If it doesn't work as expected I can back out the code.

rgutlon, 2015-11-08:
Just saw this ... leaving in about 45 minutes for the field. If you can get me the info I can test this morning.

bradanlane, 2015-11-08:
I committed the code change yesterday. If you're set up to build Stratux then you'll get the new code. You won't need to compile anything. Just use "make www" to load the new WebUI. Then open your mobile browser as usual. On the weather page, the age should be correct.

bradanlane, 2015-11-10:
Has anyone been able to confirm the "age" of weather reports in the WebUI is accurate (or at least reasonable close) and seems to make sense?

I'd like to go ahead and [re]closecthis issue.

commit 91f9242c (2015-11-10): Fixes #85.

Everything was showing a day older since “date” starts at 1.

Changed the display format of age, not the prettiest but should be
something that is obvious whose units are which especially since its
format will vary often.
closed 2015-11-10 by cyoung