cyoung/stratux #86: AHRS UI question

Bagarre, 2015-10-20:
Would you be interested in a Dynon D2'ish UI for your project?
When I saw the progress you're making on AHRS, I started playing with an SVG UI to load in a browser. My idea was to add a 4" touch screen to the Pi and make something similar in capability to the D2.
Should I just consider this a separate project for now?

bradanlane, 2015-10-20:
A touch screen interface would be very interesting. Personally, it may be best to start it as a separate projects. Two reasons:

1. The WebUI is deliberately staying focused on letting the user (a) configure theur Stratux; (b) verify it is working properly. Conversely, it is deliberately **not** attempting to implement features from the aviation App point of view.
2. we do not want even the hint that a Stratux is to be used for flight operations. That is dangerous place to be

It's totally possible to create an AI-like display with web code. I am using WebGL now for the AHRS graphical rendering. It would only be a matter of creating a different "shape". I deliberately chose a fun looking paper airplane to address #2 above.

Bagarre, 2015-10-20:
Good points, thanks. I'll work on it as a separate app that accesses stratux for it's data points.

Cheers

skypuppy, 2015-10-20:
Are you moving towards the tablet/ipad being able to fully control the 
stratux (outside of the data transmitted TO the tablet?)

Skypuppy7

bradanlane, 2015-10-20:
I'm not sure what you mean by "fully control the Stratux".

The design had always been that the WebUI is there to aid with setup and verification. Capabilities beyond those goals have generally been assumed to be the role of the aviation app or, as in this topic, a potentially new app that gets data from a Stratux.

skypuppy, 2015-10-20:
I refer to remotely control all the stratux settings via the tablet:
1. turn on/off each or both rtlsdr's
2. turn on/off ahrs
3. turn on/off gps
4. turn on/off gps WAAS (and equivalent) verification
5. future: set source of gps (a) existing unit, b) input via serial and 
not existing gps, or c) usb gps
6. future: set which ahrs unit to use, versus the default
7. actually, any setting that can be toggled, like logs, etc.

(I'm working on the code now for alternate gps and alternate ahrs 
units.  But it is kinda hard to figure out which parameters go where 
from the main/r8xx code.)

Thanks,
Skypuppy7

ps. In that thought train, coding just one or two choices for gps/ahrs 
in hard code is not much overhead.  However, if we want to support many 
more different ones, the smallest code would be loadable run-time 
modules for the particular hardware the builder chooses.  How do you 
guys prefer to do it?  I'm still in the mindset that embedded systems 
must have the smallest, tightest, least memory-hog code possible.  :)  
Yes, there is more wiggle-room, esp with a Pi 2 or BBB, but who knows 
just how much more?  Administering that could turn into a nightmare down 
the road as well. <sigh>  I *really* like the BNO055 as ahrs because it 
already has the sensor fusion cpu built into the chip, saving precious 
floating point cycles on the main system.  Plus, you can get it in 
Euler, cartesian, or Quaternions (which I'm still trying to figure out.)  :)

bradanlane, 2015-10-20:
Thanks for the background to your question. The WebUI supports enable/disable of all of the supported hardware within the Stratux. Since there are -at most - 2 SDRs, GPS, and AHRS, those are the hardware toggles. 

If there are software tuning requirements, those are in the WebUI as well. A good example is the ModeS setting.

Then there are software toggles to aid with debug and to verify operations. These include logs and tagging traffic with their source marking.  And to help with verifying some data, there are a few "helper settings" such as a watch list for weather to filter the potentially hundreds of messages.

If you were to add the ability to have multiple GPS or AHRS modules simultaneously, then you would need to support that in the UI. I'm not sure why you'd have multiples. If it's a matter of supporting different hardware, then the backend should detect. The existing WebUI would still just toggle whatever device were serving as GPS, AHRS, etc.

skypuppy, 2015-10-20:
I was thinking along the lines of someone (like me) saying, "well, I 
already have brand x gps and brand y ahrs but I don't know how to 
compile against all those different source languages, must less test 
them."  So a user has something besides the default hardware and doesn't 
know how or have the capability to modify the original source code, or 
doesn't know how to compile and field his own package.  Just thinking of 
opening up the hardware side.  Plus, getting away from RPi sensor fusion 
routines...  Load and go with different hardware if you guys want to 
even think of that for the future.

My webui does not support anything other than toggling gps on/off, ahrs 
on/off and a couple other simple (but hugely important) things, via the 
browser going to 192.168.10.10.  Just the basic important gear in 4r2.  
Or am I missing something?

bradanlane, 2015-10-20:
A lot of work has recently gone into the WebUI. The release images are just now starting to expose that work to end users.

If you are working with "code" then you have access to everything that has been developed whether or not it has yet been packaged into a release image.

As As you noted, hardware variability could quickly explode the branch complexity of the project. For that reason, and to simplify both support and users who do not have deep technical knowledge, the Stratux project deliberately limits the variants. As an example, the project supports a single WiFi adapter. 

The assumption is, any user who is technical enough and has the desire to support other hardware, is able to fork the code. For all of the user who only feel comfortable taking pre-built images from the release page, then the limited hardware support list simplifies any potential issues they could encounter.

cyoung, 2015-10-20:
@skypuppy - you're testing BNO055 and which GPS?

The AHRS implementation I'm working on now uses the MPU6050/9150 DMP "Digital Motion Processor". How does this differ from the BNO055's processor?

skypuppy, 2015-10-20:
GPS: tinkering with adafruit ultimate gps (mkt3339) via serial and just 
for fun, the microsloth usb gps (pharos 500).

ahrs: first I heard of anyone getting  the dmp portion of the mpu series 
to cooperate.  Who did you have to kill to get that code?  :)

The BNO055 allows raw sensor feeds should you wish it, and the sensor 
"fused" readings from the IMU portions in euler, cartesian, or 
quaternion readouts; your choice.  It is also an I2C chip.  I've 2 
versions, one from the Tindle store and one from Adafruit (recent 
release.)  Both vendors about $35 each.  Just received the adafruit ones 
yesterday and soldered on the pins but haven't actually hooked them up 
yet.  Still playing with the GPS.

So which coordinate version does the ads-b expect?  I assume cartesian, 
but please correct me.  Good thing about the quaternions is they don't 
have that flip problem at 180 degrees.  Not sure yet but if we have the 
Pi read the quaternion coords, maybe we can mathematically convert them 
to cartesian and avoid that 180 degree problem.

Aside: AIUI, the Bosch BNO055 algorithm is cleaner and more stable than 
the mpu (invensense?)

On the down side, it takes more tedious (not complex) code to get all 
the goodies out of the BNO055 but it gives you so much more.

I'm still playing around with the gps so haven't applied the ahrs yet.  
All the above statements are from the literature and posted user 
experience.  Not empirical data yet on my part.

Skypuppy7

cyoung, 2015-10-20:
>ahrs: first I heard of anyone getting the dmp portion of the mpu series to cooperate.

https://github.com/cyoung/linux-mpu9150

https://github.com/cyoung/stratux/blob/master/test-data/ahrs/ahrs.log

skypuppy, 2015-10-20:
well Hallelujah!  I didn't think they were ever going to release it to 
us!  Of course, Bosch has not either last I heard, but at least they 
gave us hooks to be able to use their sensor fusion machine, while 
Invensense did not do that for the first year, year and a half.  We knew 
it was there, just no docs to get to it.

Are you guys using the invensense anywhere?  Once I get the Bosch unit 
up, I'd like to compare accuracy and repeatability.  I've got an MPU 
6150 around here somewhere...

Skypuppy7

skypuppy, 2015-10-21:
Is there an example somewhere of the actual strings that get sent over 
the wifi: gps, ahrs, ad-b, weather, and traffic, please?

Skypuppy7

I think I'm putting too much on q 86...

cyoung, 2015-10-21:
https://github.com/cyoung/stratux/blob/master/test-data/ahrs/ahrs_replay.go

skypuppy, 2015-10-24:
I did find that one of my existing ahrs chips contains a 6050.  Woohoo.  Wired it to the rpi2 and on the second wiring setup :) it is seen by stratux.  Woohoo again.
However, even though the tablet Android version of iFlygps can see the ahrs via the stratux wifi, iFly can't use it in it's ahrs instrument.  Don't know why not.  
Now to get the serial gps data into the stream...

closed 2015-11-21 by cyoung