cyoung/stratux #87: 0.4r3 Compile error

Axtel4, 2015-10-21:
I pulled 0.4r3 including the latest commit (865ba68) and received the following errors when I tried to compile the code.
main/sdr.go:20: dev.GetHwInfo undefined (type *rtlsdr.Context has no field or method GetHwInfo)
main/sdr.go:28: undefined: rtlsdr.HwInfo
main/sdr.go:29: dev.SetHwInfo undefined (type *rtlsdr.Context has no field or method SetHwInfo)

Here is the full error:
pi@raspberrypi ~/stratux $ make
cd dump978 && make lib
make[1]: Entering directory '/home/pi/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o
make[1]: Leaving directory '/home/pi/stratux/dump978'
go get -t -d -v ./...
go build -ldflags " -X main.stratuxVersion=`git describe --abbrev=0 --tags` -X main.stratuxBuild=`git log -n 1 --pretty=%H`" main/gen_gdl90.go main/traffic.go main/ry835ai.go main/network.go main/managementinterface.go main/sdr.go main/uibroadcast.go
 # command-line-arguments
main/sdr.go:20: dev.GetHwInfo undefined (type *rtlsdr.Context has no field or method GetHwInfo)
main/sdr.go:28: undefined: rtlsdr.HwInfo
main/sdr.go:29: dev.SetHwInfo undefined (type *rtlsdr.Context has no field or method SetHwInfo)
Makefile:2: recipe for target 'all' failed
make: *** [all] Error 2

Note: Thanks for all your help. I was able to compile and run 0.4r2 after the Ownship Mode S was corrected (commit 0d0e3b8).

JohnOCFII, 2015-10-21:
@Axtel4 I ran into the same error.  Just for additional info, I'm not starting from the recent release version, but starting from v0.4r2 which I've been updating with 'git pull'  Not sure if that has caused something to be missed.

Axtel4, 2015-10-21:
Same here, I started with a compiled version of 0.4r2 where I pulled in 0.4r2 commit 0d0e3b8 and compiled from there. I found that 0.4r2 was very stable however, I wasn't running the GPS /AHRS option. I have one ordered and plan to start testing it with 0.4r3.

cyoung, 2015-10-21:
go get -u all

closed 2015-10-21 by cyoung

JohnOCFII, 2015-10-21:
That did the trick.  Thanks!

Axtel4, 2015-10-21:
I now have this error:
pi@raspberrypi ~ $ go get -u all
package cmd/internal/rsc.io/arm/armasm: directory "/home/pi/go/src/cmd/internal/rsc.io/arm/armasm" is not using a known version control system
package cmd/internal/rsc.io/x86/x86asm: directory "/home/pi/go/src/cmd/internal/rsc.io/x86/x86asm" is not using a known version control system
...Followed by a stream of warnings like below....
warning: code.google.com is shutting down; import path code.google.com/p/freetype-go/example/freetype will stop working

JohnOCFII, 2015-10-21:
@Axtel4 Those are just warnings.  They can be ignored.  It'll finish after a few minutes, and then you can re-run make, make install, etc.  Don't forget to also manually copy the /web/* files to /var/www

Axtel4, 2015-10-21:
Thank you JohnOCFII. It looks like I am back.

cyoung, 2015-10-21:
make www

For the web install

skypuppy, 2015-10-21:
Are you guys all using Eclipse or Qt or what?  And are you cross-compiling?

Thanks,
Skypuppy7   <sigh>

JohnOCFII, 2015-10-21:
@cyoung - Thanks

@skypuppy I'm just building right on the pi - no cross compiling.  For the little things I am doing, I'm not using an IDE, I'm just editing text files with vi.

skypuppy, 2015-10-21:
Man, that's hard core! :-)