cyoung/stratux #88: 0.4r3 console messages state 'can't start ES listening' but it's working

loather, 2015-10-21:
Here's the error from the 'gen_gdl90' process (also note CPU temp as '-99.00'):

```
usb_claim_interface error -6

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
2015/10/21 02:28:32 Can't start ES listening - no available RTL-SDR.
2015/10/21 02:28:32 stats [up since: 30 seconds ago]
2015/10/21 02:28:32  - CPUTemp=-99.00 deg C, MemStats.Alloc=3.2MB, MemStats.Sys=8.8MB, totalNetworkMessagesSent=15
2015/10/21 02:28:32  - UAT/min 54/54 [maxSS=5.40%], ES/min 61/61
, Total traffic targets tracked=2
```

I have my SDRs labeled as stratux:978 and stratux:1090.

cyoung, 2015-10-21:
Had you swapped them around and changed the settings without rebooting?

loather, 2015-10-21:
Nope. They were in their original positions and I had done a clean reboot to address that very situation.

cyoung, 2015-10-21:
After the reboot and no settings changes, did you get the crash as in #89 or was it stable?

loather, 2015-10-21:
Yes; the crash is persistent as are these messages.

cyoung, 2015-11-21:
Relevant code has changed since this issue.

closed 2015-11-21 by cyoung