cyoung/stratux #91: iOS 9.1

ChevyFlyer, 2015-10-23:
I bit the apple bullet and have upgraded my wifi only iPad Air.  Feel free to comment if this was a bad decision and I will keep you updated about my experience with it and FF 7.3.2.

ChevyFlyer, 2015-10-23:
iOS 9.1, FF 7.3.2, RY835AI is usb connect.
Loaded stratux v0.4r4 and went out to the driveway, turned on all hardware, saw 4 airplanes at 34000', GPS/AHRS shows my location on the map, AHRS is showing some information but is still in development.  Flying it tomorrow.

Ergonomicmike, 2015-10-23:
Since this isn't a code issue, I submit that it would be better if this comment were closed here and posted instead over in the general comment forum at https://www.reddit.com/r/stratux.

ChevyFlyer, 2015-10-23:
More will be revealed. Flight testing tomorrow. The web interface closed
unexpectedly using chrome. Logs weren't on.

cyoung, 2015-10-23:
I'm fine with it, there have been issues that popped up from new iOS versions concerning compatibility.

I've experienced the crashing browser in the interface before. It happened only once so far using Chrome and while looking at the GPS page. Is that what happened for you @ChevyFlyer ?

ChevyFlyer, 2015-10-23:
It happened after bringing the unit back under the roof, I was driveway
testing.  When it got back under the roof it crashed the browser, I was
working the interface and don't remember which screen may have caused it.

ChevyFlyer, 2015-10-25:
Flew it both yesterday and today and everything was turned on.  Yesterday I saw weather but no real traffic.

Today flew to JXN and back and everything seems to work.  Had to do a hard reset on stratux at JXN when leaving.  Un plugged stratux, went in to eat, plugged it back in and neither Avare or FF could see it.  Reset all and it all hooked up for the return flight.

Friend has adsb out on his RV-12 and I used stratux to find and trail him from ten miles away.  I sorta snuck up on him and could tell by his flying he could see me coming his way.

bradanlane, 2015-10-25:
I flew 0.4r4 on Friday. I had to power cycle it the first time because nothing would connect to the Stratux. iPhone would get a 254 address. After power reboot it was fine and I flew for a little over 3hrs with no problems.

CL604pilot, 2015-10-26:
Stratux connects on my Ipad and stratux has a check mark beside it.  Took approx 1 minute to have tick mark.  Radar connection is not showing.  Version 4r4

cyoung, 2015-10-26:
@CL604pilot - What do you mean by radar connection not showing? If you're using FF, look on the devices page for radar frames number and the time of the most recent radar update.

ChevyFlyer, 2015-10-27:
On the trip back from JXN yesterday my own plane was showing on screen just below my actual altitude.  The FAA hex code is in the web interface and it is holding it there, is that an issue?

cyoung, 2015-10-30:
Are you ADS-B Out @ChevyFlyer ? So you had your ModeS code in the web interface, ADS-B Out turned on (transmitting), GPS connected, and GPS enabled (stratux interface)?

ChevyFlyer, 2015-10-30:
I do not have ADS-B Out, only the Stratux with two SDR units and the
RY835AI.  Everything is on in the web interface for my last flight report.

cyoung, 2015-10-30:
Ah, you'll your shadow at times, then - whether or not you have your ModeS code in the interface or not.

CL604pilot, 2015-10-30:
I have the same setup. Challenger 604 with 2 SDR units. May have RY835AI hooked up. New enclosure being made so we will see. 

I'll be flying Sunday to Thursday all over western and central USA and will report.  (8 legs flying around FL380-FL400)

Can someone let me know how to download reports to web interface please. Im not near a computer until I depart Sunday. 

Regards,

Owen

cyoung, 2015-10-30:
Owen, go to settings in the interface http://192.168.10.1 (while connected to stratux).

Turn on "Record Logs".

On the ground, get
http://192.168.10.1/logs/stratux-uat.log
http://192.168.10.1/logs/stratux-es.log
http://192.168.10.1/logs/stratux.log

It will save across reboots (saves permanently on the memory card when "Record Logs" is enabled).

CL604pilot, 2015-10-30:
Thanks,

I'll be in touch next week.  

Have a great weekend.

CL604pilot, 2015-11-04:
Good Morning,

Ver 4.04

I have dual SDR setup and all appears to be running well. Have had to reboot unit 2 times in 8 hours run time due to no weather being picked up.  Showing traffic but on two occasions showing aircraft at incorrect altitude flying 1000 feet below on tcas it showed 2000 feet below on foreflight.   Flying between 36,000 and 41,000 feet.  Flying from west coast to New Orleans today and picking up everything.  Antenna location is key. Tried side windows with less interference from heated windows. Side did not do well.  Placed in the front windows and full coverage.  Pictures below.  GPS AHRS unit not connected this time. Using Dual xgps160. 

Weather in the Stratux 192.168.10.1 weather tab showing weather as old (see pic). I confirmed that the weather showing was up to date and correct.  

I will be flying back Thursday or Friday on a 5 hr flight and able to periscope live if you would like to watch for 10 minutes or so.  A perk with great WiFi.  If so send me a email so I can let whomever interested when.  

I will upload the log files when I get home.  

Cheers,

Owen

skypuppy, 2015-11-04:
I'd love to see your video!

David Merchant

cyoung, 2015-11-04:
Owen, great information and very valuable.

On the incorrect relative altitude on TCAS vs FF - was the GPS altitude on FF matching up with the altitude source of TCAS?

skypuppy, 2015-11-04:
For reboot, two things come to mind:
temperature drift of the rtl-sdr's, and
no high power tower near your route or no ads-b out to kick the ground 
transmitters into working.

bradanlane, 2015-11-04:
@CL604pilot - the excessively old indication for weather is being tracked under bug #85 . I'm struggling a bit since I need to do a small amount of debugging "real time" and I can do that while flying solo. I am aware of it and will figure out how to test it (and subsequently fix it).

rwilmoth, 2015-11-04:
Where are you located? Perhaps we can help you find somebody to pilot 
while you debug...

skypuppy, 2015-11-04:
My thoughts exactly.  I'm in north Alabama, with Moontown 3m5 as my base 
airport, if that helps anyone.  I do not have a bird that is currently 
flyable.

David M.

CL604pilot, 2015-11-04:
I just checked and the altitude was relative altitude not pressure altitude so yes that would be the difference.   On FF tag showed proper altitude on tag @38,000 but showed -20 by aircraft symbol.  

About to decend and confirmed above scenario with another aircraft.  

Will have AHRS hooked up next week so can report again with that then.

CL604pilot, 2015-11-04:
Download periscope and add @owenstacey 

I will send a email when I know departure date and time. 

Owen

dmurray14, 2015-11-04:
Added you! Can I send you a set of antennas to test? Would love to know how
they work up high.

skypuppy, 2015-11-04:
er, which periscope?

CL604pilot, 2015-11-04:
David,

Send me a email to owenstacey@gmail.com

Would gladly test some antenna's. 

I'll give you address when you send me a email. 

Thanks.

CL604pilot, 2015-11-04:
It is a Twitter thing. Sign up for Twitter then download periscope on android or Apple.  It's an App. 

Add me @owenstacey

CL604pilot, 2015-11-05:
I will be doing a live demo via Periscope today(Twitter app add on in App Store or Google Play).  Download onto phone or iPad/tablet. 

Wifi on the Challenger is very good most of the time so let's shoot for 1:45 pm PST  (2145 UTC).  Will be flying between 36,000 and 41,000 feet. 

User name @owenstacey on periscope.  Just add me and keep an eye out for me to start scoping. 
Dual SDR set up with stock antennas that came with SDR's.  No AHRS this time. Using Foreflight.  Ver 4.04

Regards,

Owen 

®

cyoung, 2015-11-05:
How long will the wifi be up on it? Have you used SSH before? Might be able to set up a reverse SSH tunnel and I can log into the stratux and try some changes while at altitude? Had another guy doing tests at ~FL360, tweaking to get better signal strength, but it was limited because he had to load a new image on the ground after flights.

CL604pilot, 2015-11-05:
Sure. Never used SSH on iPad. Tell me how and we will give it a shot.

CL604pilot, 2015-11-05:
15 mins let's say. Live.

cyoung, 2015-11-05:
On second thought, it won't work unless you have a laptop that you'd take along which also has an ethernet port and WiFi, and you have an ethernet cable handy.

Do you have two devices, one for Periscope and the other to access stratux / EFB on? You'll have to use them on different WiFi networks.

CL604pilot, 2015-11-05:
Don't have that.  

Will set up for next time.

cyoung, 2015-11-05:
Alright, no problem - curious to see how it works even so. If things don't work out with Periscope, try unplugging the tuner dongle and plugging back in instead of resetting the Pi if you have signal drop.

rwilmoth, 2015-11-05:
I have a laptop on me and will have in flight wifi tomorrow. I don't have my Stratux but we can try the reverse ssh connection and see if you can get to a VM that I can launch as a test.

bradanlane, 2015-11-05:
@CL604pilot - if are are able at some point during the flight, would you open:

http:192.168.10.1/logs/stratux.log

I'd be interested in the date/time that is being recorded in the log. 

The answer may help with diagnosing the erroneous "age" of weather reports within the WebUI.

Thanks.

CL604pilot, 2015-11-05:
Will do. 

®

CL604pilot, 2015-11-05:
Starting Broadcast a few minutes early.

CL604pilot, 2015-11-05:
Email me address you want me to look at. 

Re broadcasting in a few minutes. 

®

cyoung, 2015-11-05:
Got sound now, sent you that message

CL604pilot, 2015-11-05:
Howdy,

Hope you all got to watch that and it helped out.  Send me a email whenever you want something tested.   I'll be going to California next week and Hawaii and Mexico next 3-5 weeks.

Have a good one,

Owen

skypuppy, 2015-11-05:
Age of weather reports:
For the first time today, I drove up to an ADS-B tower.  Traffic info 
was good.  Weather info was, worst case, 29 minutes old and included 
some pireps (although I could not catch them.)  Weather reports only had 
a maximum reported number of 10, but at one point there were pages and 
pages of them.  I still couldn't catch a pirep though.  :)
Traffic said peak of *hundreds* of 1090 and thousands of 978 at one point!
Range of that tower on the ground while driving was roughly 5 miles in 
both directions.  Looked like an ordinary cell tower so I assume they 
rented "space" on it.
It was the first time I had positive connection and the iFly little 
ads-b report box did not say, "out of range."  Yay.
Sorry I couldn't manage a screen shot with the weather reports (which I 
saw on the 'web page.')

stratux with dual tuners.

David M.

skypuppy, 2015-11-05:
Is it saved anywhere so others that could not be there can still view it?

Thanks,
David M.

CL604pilot, 2015-11-06:
Yes.  Go to Periscope App. Find me and there should be video history. 

Let me know if it's not there.  

Owen

ChevyFlyer, 2015-11-09:
IOS 9.1 seems to be working for me on the wifi only ipad air and FF 7.3.2,  Just sent two zip files @cyoung  with the logs from two flights this week.

Should I close this issue?

cyoung, 2015-11-10:
Yep seems good, thanks.

closed 2015-11-10 by cyoung