cyoung/stratux #94: 1090-only 0.4r2 vs 0.4r4

nickulman, 2015-10-24:
1090-only mode works for me in 0.4r2 with /usr/sbin/stratux edited properly.  1090-only does not work for me in 0.4r4.  I did not edit /usr/sbin/stratux in that version.

cyoung, 2015-10-24:
The changes that made this possible were reverted from v0.4r3 to v0.4r4, it was causing some instability. Will be taking a look at this soon and the feature will be re-integrated. It seems to work fine for some users and not for others. Give v0.4r3 a try and if it has issues, will have to wait.

CL604pilot, 2015-10-26:
I have had unit running for 3 hours now.  Im watching planes in and out of Seattle and Vancouver, BC Canada and 978 Mhz not working.  1090ES is appearing to work great.  confirming with flight aware traffic I am tracking.  Not picking up weather due to my location.   

I am going to run V4.3 and will report if anything significant no one else has not already reported.

cyoung, 2015-10-26:
Do you have a reason to believe that you should be picking up weather on the ground?

ADS-B ground tower signals typically aren't easy to pick up on the ground. Have you done a flight test?

CL604pilot, 2015-10-26:
I know it's not available on the ground.  Flew the other day to PDX and was getting some weather then. 

I will be flying tomorrow to the East coast at FL390 and will let you know.  

Owen

rwilmoth, 2015-10-26:
He never stated he was on the ground.

What was your position/altitude with no 978? I'd simply like to see it on a map.

Also, I know others have been reporting that high altitudes don't seem to do well with 978 wx.  Can you confirm?

cyoung, 2015-10-26:
@rwilmoth - don't you say you fly commercial a lot anyways? Where are your test results?

rwilmoth, 2015-10-26:
I do, however my typical test results are the following:

1) Board plane with Stratux already running
2) 1090 traffic goes crazy
3) PAX next to me begins to worry
4) I hold 1090 antenna next to window during take-off roll
5) I have a conversation with a flight attendant and agree to turn it 
off and discontinue the practice of freaking out my fellow PAX.

This has happened about 6 times. I'm on 6 Delta flights this week (just 
hit my 120th leg last week for the year), however it's not always 
practical for me to take my Stratux with me, as I don't yet have a 3d 
printed case. I do have my new Antennas from DigiKey as of last week, so 
that should help with me keeping from looking too suspicious.

I'm still waiting for the glorious flight where the guy next to me 
either doesn't exist, is a pilot, or happens to be asleep.

I have also discussed this with a few pilots on a delayed flight, who 
thought it sounded cool, and may end up building them on their own. I am 
unwilling to ask a pilot to put something next to his window or on the 
glare shield as that would almost guarantee me a trip to a small room, 
likely with somebody that i'd rather not get intimate with.

I've been able to make an attempt to get signals after take-off, at 
cruise, however it really does not seem to work well. I just got my 
AHRS/GPS chip, but have not wired it up yet, so that'll be my next 
addition.

Due to a shipping error on Amazon's part, I actually have an extra RbPi 
that I plan to do a simple 978 only case with for testing just that 
functionality in a smaller package.

For reference, here's my test setup, and my normal flight schedule and 
abilities:

RbPi2
NooElec NESDR (per recommended hardware list) x2
Anker 25,600mah battery
Digikey 916mhz 1/2 wave antenna

I've tested this set up from my house (for 1090mhz) and it works well. 
I've also tested in-flight at altitudes ranging from 600 MSL (no signal) 
operating out of KOFP to the north of Richmond, VA, where 978 picks up 
around 500ft above the ground, with a tower at Richmond. 1090 traffic 
works well up to 8000ft (as high as i went).

I'll have a layover in Atlanta and post some additional results, as well.

closed 2015-10-26 by nickulman

cyoung, 2015-10-26:
This is somewhat getting beyond the scope of the original issue (this ticket is mostly open for posterity as the issue and cause are known and being worked on now) - but 1) Digikey antennas show poor performance in tests compared to the other ones (especially dmurray14's), 2) @PM-ME-YOUR-EMPENNAGE tested quite a bit from the cockpit of an ERJ-145 and we made a number of tweaks. We were able to get a QoS >1.0 for fairly long flights, meaning that the level of 978MHz at the altitudes he was flying at (FL300+) was "good", 3) ground tower signal strength is weaker than at small aircraft altitudes, but can be acceptable given a decent antenna and good placement.

skypuppy, 2015-10-26:
Maybe a belly-mounted, fixed antenna would be best in that environment 
(and maybe lot of others!)  :)

David M.

bradanlane, 2015-10-26:
I've flown the Stratux with an ADSB tuned belly mounted blade antenna. It is better but I wouldn't call it a deal maker.