cyoung/stratux #95: ahrs not seen by iFlyGPS

skypuppy, 2015-10-27:
Using the 6050, iFly is not utilizing the AHRS in it's displayed instrumentation.  I don't know if the disconnect is in iFly or in Stratux as the iFly people are VERY closed-mouth about their products engineering.

cyoung, 2015-10-27:
Probably stratux side issue. The GDL90 AHRS format was made to work with Naviator and have not added to it since. Going to work on WingX compatibility first and the rest should follow.

cyoung, 2015-10-30:
Tried contacting them and they're ignoring emails. See if you can get them interested in integrating.

skypuppy, 2015-10-30:
In one of their forums, I recall they said something about looking into 
adding more functionality via the stratux data but they were kinda 
vague.  They do seem as excited by stratux as most people are.  HOWEVER, 
they actively sell commercial ads-b's so read that as you wish.  
Hopefully, they will see that marketability of iFly improve as they get 
compatible with more external hardware, and if I don't misunderstand the 
signs, stratux is going to own the non-certificated ads-b industry in no 
time at all.

scubadogrick, 2015-12-16:
I got this from Ifly today:

Richard, 

Glad it is a benefit to you.  The Stratux open source project is definitely something we are keeping an eye on, and due to our open nature we want to support it and its users.  I have a couple of renditions here in the office and it's coming along way.  We are going to begin working on the AHRS support. Do you have the AHRS is your's, if you have one yet?  If you, you might be able to help use with some test log data.

There are still some bugs before it is fully ready for primetime, but it's a great alternative that offers an awesome value.  It is still needed to have the ADS-B out for Full Traffic as the Stratux is a In or receiver only.  Thanks again and we thank you for your business and support.

Fly Safe,
 
Shane Woodson | VP Business Development
Adventure Pilot, LLC.

cyoung, 2015-12-16:
Hey, someone was finally able to get through to them. I think the "Contact Us" on their website is broken.
The AHRS packet is as straightforward as possible, direct them to this:

https://github.com/cyoung/stratux/blob/master/main/ry835ai.go#L496-L535

duecedriver, 2015-12-16:
nope.. I just have the basic box and a single radio to play with at the moment.

I dont get to do much flying since leaving the service.. not much work here for a retired spy plane pilot but my wife wanted to start a teaching career so thats how we ended up here.

I was waiting to see what shook out of the tree during development as far as the ahrs hardware.  this by far will be the hardest to get right, but it looks like there are several good possibilities and with the drone market taking off .. more by the day.

scubadogrick, 2015-12-16:
It's been forwarded. Hopefully this will help get the AHRS going on Ifly.

duecedriver, 2015-12-16:
speaking of radios.. I was trying to tune my sdr.. it the newer version that you link but I think the crystal they use is very temperature unstable.. had a hard time pulling in noaa weather radio.. it would drift left and right of peak.. it needed +54 to get it close but it drifts..

I notice they make a + version with a more accurate crytal type.. its like 4 bucks more at $29 on amazon and even marketed as ADS-B..

I think this might be the goto radio.. since the nano get heat complaints and also has the cheaper oscillator \\

looks like enclosing in in a metal case also helps with signal to noise ratios…

thoughs

duecedriver, 2015-12-16:
Shane

Do you activate the pi’s internal pulldowns on the i2c circuit?  the GPS/IMU chip has pulldowns on board and if we use the internal pulldowns of the pi and are not adding more devices to the i2c bus it should be enough to keep the signal square…

Why are they necessary/useful?
If you have no pull-up or pull-down resistors attached to a port, its status is not clearly defined. It is “floating”. It is susceptible to random electromagnetic radiation from you, from any devices near or far and from the environment. Any wires attached to the GPIO ports act as antennae for this radiation (it’s mostly radio waves).

So imagine the situation where you have a motor with a propellor on it, or one which controls Granny’s stairlift, which is switched on or off by a GPIO port. If that port is susceptible to random changes of state, the propellor might spin when it shouldn’t and hurt someone. Or Granny might be sent back upstairs when she wants to be downstairs. It’s an out-of-control situation. You can easily get it under control by using pull-up or pull-down resistors, so that’s what we do.

What’s all this about internal ones then?
In our button circuit, we used resistors to pull down the voltage. This was to demonstrate the idea.

But, in fact, the Raspberry Pi has built-in pull-up and pull-down resistors which can be enabled in software. The i²c ports have permanent, hardware pull-ups), but all the rest can be set/unset in software. This means we can eliminate our pull-down resistors for the button – as long as we enable the internal ones. (The circuit will still function without either, but may be unreliable.)

duecedriver, 2015-12-16:
I think I meant pull ups.. since the chip is designed for a pullup.. 

Activating internal pull-ups and pull-downs with WiringPi
WiringPi enables you do that. Having initialised WiringPi and set the port(s) to input or output, as we did in part 1, <http://raspi.tv/2013/how-to-use-wiringpi2-for-python-on-the-raspberry-pi-in-raspbian> you need one of the following commands…

Activate pull-up
wiringpi.pullUpDnControl(pin_or_port_num, 2)
Activate pull-down
wiringpi.pullUpDnControl(pin_or_port_num, 1)
Deactivate both pull-up and pull-down
wiringpi.pullUpDnControl(pin_or_port_num, 0)

scubadogrick, 2015-12-16:
Looks like Ifly realizes there is an opportunity here. I received this a few minutes ago. 

Rich, 

I've upped this on our priority list.  With ForeFlight just cutting RasPi AHRS support, I want to jump on it and offer an option for users to come to.  

You would need to be on our latest beta found here.

skypuppy, 2015-12-16:
There is a specific code you can send to the specific pin for I2C to set
pull up/down directly within the stratux software during
initialization.  No external code needed.  I'd have to look up which
code (0x07?) and which address to use, assuming I2C is already activated
in user space in the kernel (and all of cyoung's images it is.)

Shane-iFlyGPS, 2016-01-11:
I've been working with Chris recently and we've added the new AHRS coding to our new beta v9.4.33b. It's very similar to another model we have, but we need a controlled test to be sure.  I have a couple Stratux builds in office, but I'm currently waiting for an AHRS to arrive to add and test.  

You can download the latest iFly GPS beta from http://iflygps.com/androidbeta/ and our app comes with a free 30-day trial, but thereafter would need a VFR or IFR subscription.  Weather, Traffic and GPS have been reported to work fine, we are just adding in the AHRS support.  

This new beta version has lots of new ADS-B improvements to Traffic display, Winds Aloft Flight Planning, Touch/Hold/Release Shortcuts, Stratux support in the Connected Devices and more.  

If someone with a Stratux AHRS model would like to perform a controlled test, that would be great.  Connect the Stratux with this latest version of iFly GPS.  Then touch Menu -> About -> Connected Devices -> and set the Wireless Devices to Stratux.  Once you see data start to populate in the window, place the stratux on a flat surface and follow the below steps.  

1. Touch Wireless Device and check "Enable ADS-B Logger".  Touch Close.
2. Slowly pitch Up 90 degrees, hold for 3 seconds, return to level. 
3. Pitch Down 90 degrees, hold for 3 seconds, return to level. 
4. Roll Left 90 degrees, hold for 3 second, return to level. 
5. Roll Right 90 degrees, hold 3 seconds, return to level. 
6. Touch Menu About Connected Devices and inside Wireless Device uncheck the Enable ADS-B Logger.  Touch Close. 

Now connect back to the internet and touch Menu -> About -> Send Bug Report -> Select Log and chose the AdsbRaw.bin file.  Touch Select and send with a note "Stratux AHRS Test" at the top.  With this test we should easily be able to confirm all is good for our production release.

Shane-iFlyGPS, 2016-01-13:
Just wanted to post an update.  The next production release of iFly GPS will support Stratux with 978, 1090, GPS and AHRS.  We've received several tests and the AHRS is working correctly.  That said, there are some underlying accuracy bugs in the AHRS coding, but these are being worked on by Chris.

scubadogrick, 2016-01-13:
Thanks guys I for one appreciate your support for the Stratux project. Do you have a timetable for this update release?

Thank you,
Rick Herbert

Shane-iFlyGPS, 2016-01-13:
Absolutely, sorry it was not sooner.  We’ll have a beta release out to support the Stratux and AHRS later this week.  The v9.4 production version on iFly, iOS and Android is targeted for the end of the month.

scubadogrick, 2016-01-13:
Looking forward to it and thanks again!!

Rick Herbert

skypuppy, 2016-01-13:
I'm only one customer but your company's efforts re stratux is why I
renewed this year.
But I do wonder why y'all didn't do the same thing when I asked as an
individual.

David Merchant

Shane-iFlyGPS, 2016-01-14:
Without to much detail.  We actually had confirmed support, just using another device setting, but were missing some aspects.  I suspect it was lost in translation and received no response, maybe on the forum, etc.  

This is an internal project that I have prioritized, monitoring and have stressed the importance of supporting the Stratux project to our development team...  

Please don't take offense, it's been in the work for some time, but I just poured some fuel to it.  Call it timing or an opportune time, if you will.

Nokomis449, 2016-01-14:
Skypuppy you weren't alone; I also pinged iFly a time or two to support Stratux and I bet there were others.  It's not an exact fit, but you could almost apply one of my favorite quotes here: "No single raindrop thinks it's responsible for the flood".

skypuppy, 2016-01-14:
I was referring to some very simple but very important questions related
to my AHRS efforts before I became aware of stratux.  I surmised it was
the iFly people trying to protect their investment in the ADS-B
equipment that they sell (but not produce.)  Looks like stratux has a
large enough market footprint now that several vendors have had to sit
up and take notice!  Yay!

Skypuppy

Ergonomicmike, 2016-03-05:
I submit that this issue can be closed now, since AHRS is seen by iFly now.

I just tried iFly 9.4.68 (the public release) with my new [LEADIY-D3 AHRS module](https://www.reddit.com/r/stratux/comments/479wy0/ry835ai_ahrs_alternative_for_sale_2499/) from LithuanianAmerican.  Ground testing for now. (Tablet app of iFly on Android.) When I call up the Instrument Panel and the AI, AHRS is working. (Albeit I have to orient the board's y axis along nose of aircraft. Don't know if that's typical or if a silkscreen problem.)

I put it in a bank and spun like an ice skater, trying to generate centripetal force. Hard to tell from hand holding. But it looks usable. We plan to fly Monday, so I should have video after that.

RobertGary1, 2016-03-05:
The ahrs isn't usable in flight at the moment so the issue you are seeing is not specific to iFly. 

-Robert

cyoung, 2016-03-05:
@Ergonomicmike - Thanks!

closed 2016-03-05 by cyoung