cyoung/stratux #97: WebUI menu highlight not correct

bradanlane, 2015-10-28:
From Reddit:

When selecting a particular UI page, (i.e. Status, Traffic, Weather, Settings, etc) after the first new selection is made, the highlighted color bar that indicates the current page, lags behind each new selection indicating the previously viewed page, with each new selection.

bradanlane, 2015-10-30:
It turns out this bug is specific to the iPad in landscape mode and will require either lots of trial and error changes or a Mac to debug (and I don't have a Mac).

cyoung, 2015-10-30:
I do, what/where should I look?

bradanlane, 2015-10-30:
on the iPad, "Settings" > "Safari" > "Advanced" > "Web Inspector"
connect the iPad to your Mac using a USB cable
configure Safari on the Mac for "Developer" in "Advanced Preferences"

open the Stratus WebUI on the iPad in landscape made
verify the bug by tapping "logs", "settings", and then "status" on the left menu. When you tap "status", the "settings" entry should turn blue.

Now the fun part: you will need to inspect what is happening to the CSS and HTML from Safari on the Mac and try to figure out which CSS classes are changing.

On my dev system, the "list-group-item" correctly changes to and from "list-group-item active". But something different is happening in Mobile Safari on the iPad. Either some JS is not doing cleanup, or there is another CSS clase participating and I dont' get it on my desktop.

bradanlane, 2015-10-30:
There is an alternative possibility for this bug - AngularJS may not be aware of the necessary data change and thus the sidebar markup is changed but isnt repainted until the subsequent page change. This would give the impression of the menu lagging the page by one refresh.

cyoung, 2018-01-17:
Either this issue was resolved with recent iOS/Safari updates or we fixed it a while ago. Not able to reproduce.

closed 2018-01-17 by cyoung