cyoung/stratux #99: Multiple Devices seen in ForeFlight

JohnOCFII, 2015-11-01:
I've been meaning to ask about this for a while.  In ForeFlight there are two devices that show up under the Settings --> Devices display. One is titled Stratux.  The other is FreeFlight.  If you drill into the Stratux item, there is a switch for Enabled or not, and nothing else.  The FreeFlight item shows all (most?) of the items I see when I'm using the Appareo Stratus 2.  Screen shots below.  Is this expected behavior?  Based on my experience with the Stratus, I'd expect only one device in the hierarchy.

[image]
[image]
[image]

cyoung, 2015-11-01:
That's correct. It will only show two devices when you have AHRS enabled. The "Stratux" device is sending AHRS as a separate device since the FreeFlight GDL90 implementation, as far as I know, doesn't provide for a way to send AHRS data. Also it keeps us away from using "proprietary" formats.

JohnOCFII, 2015-11-01:
Sounds good.  Thanks for the explanation.

closed 2015-11-01 by JohnOCFII