cyoung/stratux #100: 978MHz vs 1090MHz GDL90 traffic report format

cyoung, 2015-11-06:
There's a different report that both FF and WingX (probably others also) use to separate out the source of data. I'm assuming that most devices use a GDL90 0x14 "Traffic Report" but there is something that marks the source between the two of them.

Anyone have a FreeFlight device that they could capture output from?

CL604pilot, 2015-11-12:
Good Evening,

I am flying Sunday and if there is a updated (not published file) that has
fixes etc over v0.4r4 that needs to be tested, would someone please let me
know and send to me.   Dropbox is fine or something else.
I have the AHRS/GPS unit connected now and appears to be working fine.  I
will broadcast Sunday on periscope (twitter app)  between 14:30-14:45pm to
demo the system LIVE from the Challenger 604 @ 39,000 feet along with
dmurray14 antenna system and a proper dual dongle case murray made for me.

Periscope / Twitter user name is @owenstacey add me before the demo on
Sunday so I can accept your request.

Thanks,
Owen

loather, 2015-11-12:
A friend of mine has one of these and I'll see if I can borrow it from him for a few days to get some packet captures. I'll let you know either way.

bkwny, 2015-12-24:
Is this issue the reason why the ForeFlight 7.5.1 traffic callout balloon says "978 ADS-B" even though the reason I know about the traffic is that Stratux heard the 1090 ES squit?

cyoung, 2016-12-01:
@bkwny - yep. Still is.

cyoung, 2016-12-01:
Known issue with FF. Answer to the original issue is "it doesn't exist".

closed 2016-12-01 by cyoung