cyoung/stratux #102: dump1090 errors not logged

bradanlane, 2015-11-10:
If dump1090 generates any error messages that do not abort the process, they are lost.

This is not a problem under normal operations but users who may experience power related oddities (and we have seen them) will not get any evidence in the log files.

The only way to detect the errors are by killing off the stratux process and then manually starting dump1090 manually. This is difficult now that  dump1090 is started inside of the main gen_gdl90 code.

jpoirier, 2015-11-11:
We can grab stderr (and stdio ?) from the running process and pass it to the logger.

bradanlane, 2015-11-12:
I attempted this but not being completely fluent with golang, I wasn't able to get the proper results.

I'll leave this open in the hopes someone can make the necessary changes.

jpoirier, 2015-11-12:
Thanks for giving it a shot. I'll try to add this within the next few days.

closed 2015-11-12 by cyoung

jpoirier, 2015-11-12:
It's simpler (five or six lines of code) to just pipe the dump1090 stdio and stderr to the current stratux log. My last commit is included in the pending push.

commit 7f502428 (2015-11-12): Fixes #102.

     START,Thu Nov 12 19:25:38 +0000 UTC 2015,Thu Nov 12 19:25:38 +0000
UTC 2015
     UNPAUSE,71289772
     2739899291,Found 2 device(s):
     2740283977,0: Realtek, RTL2838UHIDIR, SN: stratux:978
     2740438248,1: Realtek, RTL2838UHIDIR, SN: 00000001 (currently
selected)
     2740605904,Found Rafael Micro R820T tuner
     2740683299,Max available gain is: 49.60
     2740807935,Setting gain to: 49.60
     2740880330,Exact sample rate is: 2000000.052982 Hz
     2740943976,Gain reported by device: 49.60