cyoung/stratux #103: bugs related to unplugging SDRs

bradanlane, 2015-11-10:
I'm noticing a few things with the new code.

If the UAT dongle is disconnected, the ReadSync look goes crazy logging as fast as it possible can, saturating the log.

Here is the "sdr unplug" problem. In (UAT) read()
```
    for {
        select {
        default:
            nRead, err := u.dev.ReadSync(buffer, rtl.DefaultBufLength)
            if err != nil {
                log.Printf("\tReadSync Failed - error: %s\n", err)
                break
            }
            // log.Printf("\tReadSync %d\n", nRead)
            if nRead > 0 {
                buf := buffer[:nRead]
                godump978.InChan <- buf
            }
        case <-uat_shutdown:
            log.Println("UAT read(): shutdown msg received...")
            return
        }
    }
```

The `break` will only break from the `select` and not from the outer loop. Changing the `break` to a `return` solves this.

In testing, weird things happen with the two SDRs when one is unplugged.
1. When I unplug the UAT SDR, I see a set of ES related messages.
2. Plugging the UAT SDR back in seems to attempt a restart of dump1090.
3. I end us with a bunch of defunct processes.

jpoirier, 2015-11-11:
The break should stay and and the log message should be commented out. The sdrWatcher will eventually signal the goroutine to exit when it figures out the dongle has been removed.

bradanlane, 2015-11-11:
How long before the signal? I let it run for at least 15 seconds in one test. I also reconnected the SDR within about 5-10 seconds in a subsequent test. I continually saw the error message and it never restored operations.

jpoirier, 2015-11-11:
I've not timed it but you shouldn't take more than a few seconds. Keep in mind that sdrWatcher has a 1 second heartbeat whereas the goroutine is open throttle.

bradanlane, 2015-11-11:
OK, we need to test.

I definitely tested much more than "several seconds" in both scenarios.

jpoirier, 2015-11-11:
Watch what the web interface is doing when dongle/s are inserted and extracted not the log; the logging is incidental and latent.

jpoirier, 2015-11-11:
There's a pull request to s/return/break, which is what you want to test with.

bradanlane, 2015-11-11:
I'm not going to be back at a computer for at least a day so this is an open request:

When I changed the "break" to a "return" and then tested unplugging the UAT SDR followed by plugging it back in, I saw dump1090 being restarted. I also saw multiple defunct dump1090 processes. 

In my tests, changing the "break" to a "return" was not a complete fix.

bradanlane, 2015-11-18:
it seems we have addressed this issue so I will close it. I would like us to plan a "test cycle" when @cyoung plans the next "release".

closed 2015-11-18 by bradanlane