cyoung/stratux #105: Write mode ID string to dongle

jpoirier, 2015-11-11:
I propose that we add functionality such that a user can request that a mode ID string be written to a dongle.For example:

The user has a single dongle inserted with just 978UAT mode enabled. The user presses a "write current mode to dongle" button in the web ui's setting and the "stratux:978" string gets written to the dongle's eeprom. Similar for s single dongle with just 1090ES mode enabled, the "stratux:1090" string gets written to the dongle's eeprom.

bradanlane, 2015-11-11:
nice!

Since you've added all the necessary restart code, the capability can easily be exposed through the WebUI, showing what the current identification is (if any) and letting them "mark" the SDRs. Since the backend code is assuming specific strings, there is not need for the user to do anything more than click a button and get confirmation.

jpoirier, 2015-11-11:
Okay, hadn't thought about that. Having that sort of fine grained control
and/or information at the web ui would be really nice!

bradanlane, 2015-11-11:
Within the page-context help for "Settings" I can explain,

To use a specific SDR for 978UAT, plug in just the desired SDR and use the **UAT** tagging button.
To use a specific SDR for 1090ES, plug in just the desired SDR and use the **ES** tagging button.

jpoirier, 2015-11-11:
But doesn't that already happen implicitly via the mode buttons.

E.g.

- if both modes are disabled and no dongles inserted, then I insert a
dongle and enable 978 mode, the dongle gets configured to 978UAT.
- if both modes are disabled and no dongles inserted, then I insert a
dongle and enable 1090 mode, the dongle gets configured to 1090ES.
- if both modes are disabled and no dongles inserted, then I insert a
dongle and enable 978 mode, the dongle gets configured to 978UAT, then I
insert a second dongle and enable 1090 mode, the dongle gets configured to
1090ES.
- if both modes are disabled and no dongles inserted, then I insert a
dongle and enable 1090 mode, the dongle gets configured to 1090ES, then I
insert a second dongle and enable 978UAT mode, the dongle gets configured
to 978UAT.

bradanlane, 2015-11-11:
Since we have a large number of Stratux assembled and "out in the wild", when the next release is posted, we will have one of two possible scenarios - the user has one SDR plugged in or the user has 2 SDRs plugged in.

When a user first starts Stratux after a new "release", the stratux.conf file has been initialized to its default with is 978UAT only. So we don't start with "both modes disabled".

If the user only has one SDR installed, then they may not choose to tag their SDR. In their mind, it does not add any value. They may not even go to the WebUI Settings page.

In the case of 2 SDRs inserted when they start the new release of Stratux. The user will [eventually] realize the absence of 1090ES, use the WebUI, and go to Settings to turn on the second SDR.

Since many users have tuned antennas or differing SDRs, only they will know which one they need to be UAT vs ES.

By giving them instruction on how they can control the tagging, we both insure the SDRs are correctly tagged and we give the user confidence that they control the tagging process.

jpoirier, 2015-11-11:
By tagging do mean write one of the defined mode strings to the dongle's
eeprom?

bradanlane, 2015-11-11:
yes. "tagging and SDR" is "writing mode string to eeprom".

jpoirier, 2015-11-11:
Oh, okay, now it makes sense.

bradanlane, 2015-11-11:
I've used several different terms / descriptors in this discussion which hasn't helped but I've been trying to work out the best "end user language".

jpoirier, 2015-11-11:
On Wed, Nov 11, 2015 at 11:53 AM, bradanlane <notifications@github.com>
wrote:

> Since we have a large number of Stratux assembled and "out in the wild",
> when the next release is posted, we will have one of two possible scenarios
> - the user has one SDR plugged in or the user has [...]
Yes. All of this. And a nice side benefit is that we ensure the proper ID
string is being written.

And for those following this discussion that already have dongles with
stratux IDs written to them, you're unaffected.

JohnOCFII, 2015-11-11:
As a person with separately tuned antennas and two dongles, one an older SDR that Chris has recommended I use for the less intensive 1090ES processing, and one newer version, I've been watching with great interest.  I would love to have a clear and definitive way to get my SDRs tagged properly to either 978 or 1090ES, and it sounds like this discussion has gotten us to a workable plan.

bradanlane, 2015-11-11:
Have no fear @JohnOCFII - I also have both an R820T and an R820T2 so having the definitive assignments will be helpful.

skypuppy, 2015-11-11:
Is the t2 that much better in real life?

cyoung, 2015-11-12:
@skypuppy - depends on your preferences. I haven't seen any tests definitively showing that the R820T has performance compared to the R820T2 that is worth saving the $10-15 for. If you want to run some tests that would be appreciated.

bradanlane, 2015-11-12:
Ibex done some broad spectrum tests and the R820T2 does perform better than the R820T. There are also some 3rd party tests to confirm. 

That said, you can now get the newer R820T2 for about $12 (and I just ordered 4 @ $6.78) so the price is no longer a factor.

skypuppy, 2015-11-12:
Can I ask where?  They're still $20 and up on Amazon.

bradanlane, 2015-11-12:
The $12 one seems to be out of stock. If you're not in a rush, there is a $10 (http://amzn.to/1MaieGF) deal but it will take about 4 weeks to arrive. I use these for multiple projects so I shop around and don't care if it's server all weeks for delivery.

closed 2018-01-17 by cyoung